espressif/esp-idf #6398: Data error when starting WiFi and MODBUS at the same time

leo-liuxo, 2021-01-14:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32]
- IDF version:
    v4.2-208-g21eac48ea
- Build System:         [Make]
- Compiler version:
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [Yes?]
    using vscode
- Power Supply:         [USB]

## Problem Description

When I turn on the functions of WIFI and MODBUS at the same time,and esp32 is used as the slave of Modbus.When the computer sends Modbus read command to the slave,esp32 always returns different register data.Before that,I did not use the write command to change the register data.

If I only turn on MODBUS function, and do not turn on WiFi, then the above error will not appear.

### Expected Behavior
| mater command           | esp32 returns data                     |
| ----------------------- | -------------------------------------- |
| CC 03 A0 28 00 04 F6 1C | CC 03 08 BE 1E 2A 4D 58 39 C1 C5 5F EB |
| CC 03 A0 28 00 04 F6 1C | CC 03 08 BE 1E 2A 4D 58 39 C1 C5 5F EB |
| CC 03 A0 28 00 04 F6 1C | CC 03 08 BE 1E 2A 4D 58 39 C1 C5 5F EB |
| CC 03 A0 28 00 04 F6 1C | CC 03 08 BE 1E 2A 4D 58 39 C1 C5 5F EB |
### Actual Behavior
| mater command           | esp32 returns data                     |
| ----------------------- | -------------------------------------- |
| CC 03 A0 28 00 04 F6 1C | CC 03 08 BB 8F 1E E4 98 12 5A 37 ED 2E |
| CC 03 A0 28 00 04 F6 1C | CC 03 08 C2 32 3C 5A 7E A2 35 11 FF 95 |
| CC 03 A0 28 00 04 F6 1C | CC 03 08 00 22 9D 7D 4F 2D 79 D8 85 2A |
| CC 03 A0 28 00 04 F6 1C | CC 03 08 63 13 AF 6C D2 82 03 89 E6 D4 |
### Steps to reproduce

1. Initialize WIFI and MODBUS
2. Connect the esp32 to the computer through P4 and P5 pins(P4:TXD  P5:RXD)
3. The computer sends the read command and monitors the data returned by esp32

## Debug Logs

```
I (22950) modbus: HOLDING READ (22335146 us), ADDR:41001, TYPE:2, INST_ADDR:0x3ffc3b96, SIZE:4
I (22950) modbus: fb df 4a 33 9f d0 fd 60 
I (23560) modbus: HOLDING READ (22951905 us), ADDR:41001, TYPE:2, INST_ADDR:0x3ffc3b96, SIZE:4
I (23560) modbus: 62 1e 53 49 c8 9d 8a 57 
I (23870) modbus: HOLDING READ (23262934 us), ADDR:41001, TYPE:2, INST_ADDR:0x3ffc3b96, SIZE:4
I (23870) modbus: 8f e4 80 a2 c5 26 35 86 
I (24270) modbus: HOLDING READ (23662729 us), ADDR:41001, TYPE:2, INST_ADDR:0x3ffc3b96, SIZE:4
I (24270) modbus: 7a ff 6c 2e c8 07 29 6a 
```

alisitsyn, 2021-01-14:
Hi @leo-liuxo,

Unfortunately this is known issue when you are using the WiFi and other tasks with priority higher than CONFIG_FMB_PORT_TASK_PRIO. The Modbus uses polling mode to get data 
from the UART driver and if its tasks are suspended often it causes processing delays. The CONFIG_FMB_TIMER_ISR_IN_IRAM=y can help to solve issue with interrupt latency when your app tasks use flash access which disables CPU cache. There some updates for modbus component but there is delay with merge of these fixes. I would propose you to apply the patch for v4.2 which could fix most known fixes. The errors are possible but their quantity will be decreased significantly and I propose to do retries of modbus transactions in this case. The merge of these fixes are pending...
The patch:
[backport_latest_fixes_v42.patch.log](https://github.com/espressif/esp-idf/files/5813158/backport_latest_fixes_v42.patch.log)

Please let me know if you have any questions or results. 
Thanks.

alisitsyn, 2021-02-09:
Hi @leo-liuxo,

Could you provide some update? Thanks.

Alvin1Zhang, 2021-07-19:
Thanks for reporting, would close due to short of feedback, feel free to reopen.

closed 2021-07-19 by Alvin1Zhang