espressif/esp-idf #6400: Problem with ESP32S2 OTG

wojciech-angelo, 2021-01-14:
Hello,

I have a problem with an espressif device (ESP32-S2). Specifically, I have on my desk the ESP32-S2-Soala-1_V1.2. I need to program the USB interface on the chip. The manufacturer claims that the chip has USB OTG. Unfortunately, the manufacturer's libraries and documentation do not show that it provides support at the software level.

It certainly provides support at the USB Device level, but what about USB Host? I am most interested in the USB Host functionality and more specifically in SCSI disk support. Does the manufacturer provide support for this type of architecture somewhere I haven't found it or will such support appear in the near future? If so, when? 

thank you in advance for your help

chegewara, 2021-01-14:
https://esp32.com/viewtopic.php?f=10&p=56361

USB device peripheral is supported more or less, USB host/OTG is not supported yet and as far as i know there is no timeline yet, but i may be wrong about it.

wojciech-angelo, 2021-01-14:
> https://esp32.com/viewtopic.php?f=10&p=56361
> 
> USB device peripheral is supported more or less, USB host/OTG is not supported yet and as far as i know there is no timeline yet, but i may be wrong about it.

Ohh, this no good news. Maybe next idf will be fast announced..

Anyway - dziękuję @chegewara

closed 2021-01-14 by wojciech-angelo