espressif/esp-idf #6402: SPP client server example with multiple server connections

TobiasBoeschMiele, 2021-01-14:
**Is your feature request related to a problem? Please describe.**

I would like to use more than one (currently only two) serial bluetooth LE (SPP) client connections handled by one esp32 to multiple serial bluetooth LE (SPP) servers running on other esp32. Each of the other esp32 is a serial bluetooth LE (SPP) server. This is for testing purposes, not production.

**Describe the solution you'd like**

I would like to know which implementation you suggest to achieve this. My idea is:

* multiple (two) SPP client examples running on client esp32 (UART handling?)

**Describe alternatives you've considered**

Alternatively one can use one SPP client example running on the client esp32 with multiple connections or application profiles. I don't know how to set this up. If you can give me a hint please. I will dive into the [GATT Client Multi-connection Example Walkthrough](https://github.com/espressif/esp-idf/blob/0b71a0a46d67cce681ec55973b020d950d8596bd/examples/bluetooth/bluedroid/ble/gattc_multi_connect/tutorial/Gatt_Client_Multi_Connection_Example_Walkthrough.md#gatt-client-multi-connection-example-walkthrough) in parallel to find out more.

**Additional context**

I would like to send text bidirectional from two esp32 to another esp32 (like one central and two peripherals) and for simplicity I don't want to use a specific profile.

Alvin1Zhang, 2021-01-15:
Thanks for raising this feature request.

TobiasBoeschMiele, 2021-01-18:
Is this something that can be done best using [Application Profiles](https://github.com/espressif/esp-idf/blob/0b71a0a46d67cce681ec55973b020d950d8596bd/examples/bluetooth/bluedroid/ble/gattc_multi_connect/tutorial/Gatt_Client_Multi_Connection_Example_Walkthrough.md#application-profiles)?

xiewenxiang, 2021-02-02:
The BLE SPP example is only for single connection, IF the application scenario is multi-connections, some modifications are needed do in client side, to identify which server the data is coming from and which server the data is going to.

This data source can be distinguished by MAC address. How to achieve this ultimately, the customer can decide for himself, this is just a suggestion.

venkateshgr-ayla, 2021-07-22:
Hi All,
I wanted to know if SPP can be supported with Nimble? currently I am not using Bluedroid. I want to port SPP to Nimble.
Is it technically possible and recommended? or does ESP have any plans to support SPP over nimble?

rahult-github, 2023-04-25:
The original query has been answered [here](https://github.com/espressif/esp-idf/issues/6402#issuecomment-771528927)

SPP is now available on Nimble and part of github.   

Closing issue.

closed 2023-04-25 by rahult-github