espressif/esp-idf #6404: Panic'ed reading file, using WiFi and BLE

spadespwnz, 2021-01-14:
I am experiencing a crash occasionally when Connecting to WiFi or Receiving BLE messages.
The error indicates it occurred from reading / opening a SPIFFS file, but only happens when connecting to WiFi or handling bluetooth messages.
To my knowledge, neither the WiFi Event callback nor the BLE event callback should be running in ISR context, which I verified by using the xPortInIsrContext() function, yet the error indicates that Core 0 was running in ISR context.

Is there some obvious reason why using WiFi or BLE would be causing a crash to occur when reading a file?
Thanks in advance!

Re-enable cpu cache.
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed).

Core  0 register dump:
PC      : 0x4012fc30  PS      : 0x00060134  A0      : 0x80087112  A1      : 0x3ffbe250
0x4012fc30: llcp_con_up_req_pre_handler at llc_llcp.c:?

A2      : 0x4012fc30  A3      : 0x00000000  A4      : 0x3ffe0cd2  A5      : 0x00000001
0x4012fc30: llcp_con_up_req_pre_handler at llc_llcp.c:?

A6      : 0x00000006  A7      : 0x3ffaf7bc  A8      : 0x80044715  A9      : 0x00000001
A10     : 0x00000000  A11     : 0x00000006  A12     : 0x00000001  A13     : 0x3ffe0cd2
A14     : 0x00000058  A15     : 0x01f40000  SAR     : 0x00000009  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
Core  0 was running in ISR context:
EPC1    : 0x4008dcc7  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4012fc30
0x4008dcc7: portENTER_CRITICAL_NESTED at C:/msys32/home/spade/swidget-homekit-update-2/esp-idf/components/freertos/xtensa/include/freertos/portmacro.h:147
 (inlined by) xPortInIsrContext at C:/msys32/home/spade/swidget-homekit-update-2/esp-idf/components/freertos/xtensa/port.c:349

0x4012fc30: llcp_con_up_req_pre_handler at llc_llcp.c:?

Backtrace:0x4012fc2d:0x3ffbe250 0x4008710f:0x3ffbe280 0x40085ea5:0x3ffbe2b0 0x40085751:0x3ffbe2d0 0x40087846:0x3ffbe2f0 0x400887af:0x3ffbe310 0x40081e42:0x3ffbe330 0x4009a78d:0x3ffdec50 0x40089863:0x3ffdec70 0x4014451d:0x3ffdecd0 0x4013a606:0x3ffded00 0x4013da5d:0x3ffded30 0x4013ba33:0x3ffded60 0x4013cf73:0x3ffdedb0 0x4013b078:0x3ffdedf0 0x4013a17c:0x3ffdee20 0x400d3d79:0x3ffdee70 0x401bcd85:0x3ffdee90 0x4010b5e9:0x3ffdeeb0 0x4010bb1b:0x3ffdeee0 0x4010bc7a:0x3ffdef00 0x40101132:0x3ffdef80 0x400f5a52:0x3ffdefb0 0x400f5fec:0x3ffdf000 0x400e0d32:0x3ffdf090 0x400e0d58:0x3ffdf0b0 0x400e1715:0x3ffdf0d0 0x400d86dd:0x3ffdf0f0 0x400d87b8:0x3ffdf110
0x4012fc2d: llcp_channel_map_req_pre_handler at llc_llcp.c:?

0x4008710f: r_lld_pdu_rx_handler at ??:?

0x40085ea5: r_lld_evt_end at ??:?

0x40085751: r_lld_evt_end_isr at ??:?

0x40087846: r_rwble_isr at ??:?

0x400887af: r_rwbtdm_isr_wrapper at intc.c:?

0x40081e42: _xt_lowint1 at C:/msys32/home/spade/swidget-homekit-update-2/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105

0x4009a78d: esp_rom_spiflash_read_data at C:/msys32/home/spade/swidget-homekit-update-2/esp-idf/components/spi_flash/esp32/spi_flash_rom_patch.c:270
 (inlined by) esp_rom_spiflash_read at C:/msys32/home/spade/swidget-homekit-update-2/esp-idf/components/spi_flash/esp32/spi_flash_rom_patch.c:585

0x40089863: spi_flash_read at C:/msys32/home/spade/swidget-homekit-update-2/esp-idf/components/spi_flash/flash_ops.c:627

0x4014451d: esp_partition_read at C:/msys32/home/spade/swidget-homekit-update-2/esp-idf/components/spi_flash/partition.c:349

0x4013a606: spiffs_api_read at C:/msys32/home/spade/swidget-homekit-update-2/esp-idf/components/spiffs/spiffs_api.c:36

0x4013da5d: spiffs_phys_rd at C:/msys32/home/spade/swidget-homekit-update-2/esp-idf/components/spiffs/spiffs/src/spiffs_cache.c:161

0x4013ba33: spiffs_obj_lu_find_entry_visitor at C:/msys32/home/spade/swidget-homekit-update-2/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c:166

0x4013cf73: spiffs_object_find_object_index_header_by_name at C:/msys32/home/spade/swidget-homekit-update-2/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c:1694

0x4013b078: SPIFFS_stat at C:/msys32/home/spade/swidget-homekit-update-2/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c:765 (discriminator 2)

0x4013a17c: vfs_spiffs_stat at C:/msys32/home/spade/swidget-homekit-update-2/esp-idf/components/spiffs/esp_spiffs.c:555

0x400d3d79: esp_vfs_stat at C:/msys32/home/spade/swidget-homekit-update-2/esp-idf/components/vfs/vfs.c:565 (discriminator 3)

0x401bcd85: stat at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/syscalls/sysstat.c:11

0x4010b5e9: VFSFileImpl::VFSFileImpl(VFSImpl*, char const*, char const*) at c:\msys32\home\spade\swidget-homekit-update-2\accessories\swidget\build/../components/arduino-esp32/libraries/FS/src/vfs_api.cpp:232

0x4010bb1b: std::shared_ptr<VFSFileImpl> std::make_shared<VFSFileImpl, VFSImpl*, char const*&, char const*&>(VFSImpl*&&, char const*&, char const*&) at c:\users\spade\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\ext/new_allocator.h:136
 (inlined by) ?? at c:\users\spade\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\bits/alloc_traits.h:475
 (inlined by) ?? at c:\users\spade\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\bits/shared_ptr_base.h:543
 (inlined by) ?? at c:\users\spade\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\bits/shared_ptr_base.h:656
 (inlined by) ?? at c:\users\spade\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\bits/shared_ptr_base.h:1322
 (inlined by) ?? at c:\users\spade\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\bits/shared_ptr.h:360
 (inlined by) ?? at c:\users\spade\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\bits/shared_ptr.h:707
 (inlined by) std::shared_ptr<VFSFileImpl> std::make_shared<VFSFileImpl, VFSImpl*, char const*&, char const*&>(VFSImpl*&&, char const*&, char const*&) at c:\users\spade\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\bits/shared_ptr.h:723

0x4010bc7a: VFSImpl::open(char const*, char const*) at c:\msys32\home\spade\swidget-homekit-update-2\accessories\swidget\build/../components/arduino-esp32/libraries/FS/src/vfs_api.cpp:61

0x40101132: fs::FS::open(char const*, char const*) at c:\msys32\home\spade\swidget-homekit-update-2\accessories\swidget\build/../components/arduino-esp32/libraries/FS/src/FS.cpp:191

spadespwnz, 2021-01-14:
This issue seems similar to the issue mentioned here:
https://github.com/espressif/esp-idf/issues/6068

xiewenxiang, 2021-08-06:
Hi @spadespwnz 

Please enable `SPI_FLASH_AUTO_SUSPEND` in menuconfig.

AxelLin, 2021-08-06:
@spadespwnz 
Which esp-idf version are you using? (git describe --tags)
Which h/w (module) are you using?

AxelLin, 2021-08-06:
> Hi @spadespwnz
> 
> Please enable `SPI_FLASH_AUTO_SUSPEND` in menuconfig.

Why it's related to SPI_FLASH_AUTO_SUSPEND?
SPI_FLASH_AUTO_SUSPEND depends on IDF_TARGET_ESP32C3 but above log shows xtensa-esp32-elf.