espressif/esp-idf #6405: Cannot create a file using FATFS + wear leveling

ChromaMaster, 2021-01-14:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  ESP32-WROOM-32
- IDF version: v3.3.4
- Build System: Make
- Compiler version: 1.22.0-96-g2852398
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

I'm not able to create a file using non-encrypted FATFS with wear levelling, secure boot and flash encryption.

I'm trying to use a code very similar to the one of the wear levelling example [here](https://github.com/espressif/esp-idf/blob/2bfdd036b2dbd07004c8e4f2ffc87c728819b737/examples/storage/wear_levelling/main/wear_levelling_example_main.c) but unfortunately, when I'm trying to open a file for writing and thus creating a file, I get an error from the `fopen` function.

```
wl_handle_t s_wl_handle = WL_INVALID_HANDLE;
const char *base_path = "/spiflash";

esp_vfs_fat_mount_config_t mount_config = {
  .format_if_mount_failed = true,
  .max_files = 5,
  .allocation_unit_size = CONFIG_WL_SECTOR_SIZE
};

esp_err_t err = esp_vfs_fat_spiflash_mount(base_path, "storage", &mount_config, &s_wl_handle);

if (err != ESP_OK) {
  ESP_LOGE(tag, "Failed to mount FATFS (%s)", esp_err_to_name(err));
  return;
}

char *test_file_name = "/spiflash/test_file.txt";
FILE *test_file = NULL;

// Open test file for writing
ESP_LOGI(tag, "Opening test file for writing");
test_file = fopen(test_file_name, "w");
if (test_file == NULL) {
  ESP_LOGE(tag, "Error opening the file %s for write: %s", test_file_name, strerror(errno));
  return;
}
```

Don't know exactly how to reproduce it but the device is newly flashed. Is there any known issue with the FATFS lib? Maybe the file system metadata got corrupted somehow?

### Expected Behavior

Mount the FAT file system and create, write and read files.

### Actual Behavior

I get an error from the first `fopen` function. The error code returned is 2 (No such file or directory). Digging into the virtual fs functions I find out that the problem comes from a function called `dir_alloc` which cannot find an empty block to allocate the file. 

```
I (1569) vfs_fat: vfs_fat_open: path="/test_file.txt", flags=601, mode=1b6
...
E (1969) vfs_fat: vfs_fat_open: fresult=7
E (1969) FATFS: Error opening the file /spiflash/test_file.txt for write: No such file or directory
```

The exact line that returns that result is: the `dir_alloc` function present in the `dir_register` function: [here](https://github.com/espressif/esp-idf/blob/2bfdd036b2dbd07004c8e4f2ffc87c728819b737/components/fatfs/src/ff.c#L2554)
Inside that function, the lib is looking for some empty block but eventually the `dir_next` returns errcode 4 which is `FR_NO_FILE`. [here](https://github.com/espressif/esp-idf/blob/2bfdd036b2dbd07004c8e4f2ffc87c728819b737/components/fatfs/src/ff.c#L1817)

igrr, 2021-01-15:
Hi @ChromaMaster, if you run the wear_levelling example, does the issue also occur?

ChromaMaster, 2021-01-15:
Hi @igrr.

Sorry for the name update, I fixed the typo and didn't realize that the tag was already there.

I've tried to use the wear levelling example and it does not fail. Although I've flashed it to a devkit which already has the secure bootloader and the flash encrypted (using the --encrypt flag for the esptool). Could be that the issue is caused because the first time the bootloader encrypts the flash it messes up with the FATFS partition or something?

igrr, 2021-01-15:
> Could be that the issue is caused because the first time the bootloader encrypts the flash it messes up with the FATFS partition or something?

I think the bootloader is not supposed to touch data partitions which are not marked as "encrypted" in the partition table; there is always a possibility of a bug that went undetected, though.

If you flash your original application to the same board, does it also work now, or does it still fail with the same symptoms?

ChromaMaster, 2021-01-15:
> If you flash your original application to the same board, does it also work now, or does it still fail with the same symptoms?

No, unfortunately, if I flash again the application the problem is still there. That's why I think it may be some corruption issue with the FAT file system.

igrr, 2021-01-15:
Do you see any log message like "Formatting FATFS partition, allocation unit size=..." when switching between wear_levelling example and your application? If no, it means that the same filesystem (that is not being formatted) works correctly with the wear_levelling example and doesn't work with your project. If you see this log, it means that the filesystem is being created from scratch, but it is not working correctly.

One more thing, could you describe the process you are using to flash the wear_levelling example after flash encryption and secure boot have been enabled?

ChromaMaster, 2021-01-15:
> Do you see any log message like "Formatting FATFS partition, allocation unit size=..." when switching between wear_levelling example and your application? If no, it means that the same filesystem [...]

No, the FAT FS partition is not formatted when I switch between app images. Both apps are using the same file system.
 
> One more thing, could you describe the process you are using to flash the wear_levelling example after flash encryption and secure boot have been enabled?

Yes of course. I first build the app using the `make` command. And the I flash it manually according to my partition table like so:

```
esptool.py --chip esp32 --port /dev/ttyUSB1 --baud 921600 write_flash --encrypt 0x20000 ./build/wear_levelling_example.bin
```

xcguang, 2021-02-01:
[image]
Please try enable long filename

ChromaMaster, 2021-02-01:
Hello @xcguang
Thanks for your response. I have that option enabled, here is the config that I'm using.

```
OEM Code Page (US (CP437))  --->
Long filename support (Long filename buffer on stack)  --->  
(32) Max long filename length
    API character encoding (API uses ANSI/OEM encoding)  --->
(0) Number of simultaneously open files protected by lock function
(10000) Timeout for acquiring a file lock, ms
[*] Use separate cache for each file
```

xcguang, 2021-02-01:
> (32) Max long filename length

Change this to 255?

ChromaMaster, 2021-02-01:
After dumping the storage partition I've been able inspect it using the `mdir` tool. Here is what I've got:

```
 Volume in drive : has no label
 Volume Serial Number is 0021-0001
Directory for ::/

Ù■ÉMSDOS 5.0         0 1980-01-31   0:00 
        1 file                    0 bytes
                             89 088 bytes free
```
It seems that the FS is not full at all (as expected since it's newly flashed). However, there is a file there (which also occupies 0B). But I don't really know how that file got there :S.

ChromaMaster, 2021-02-01:
> > (32) Max long filename length
> 
> Change this to 255?

Unfortunatelly, this doesn't fix it.

ChromaMaster, 2021-02-09:
I could not reproduce this anymore, the only way to reproduce it is by cutting the power off during the first FAT formatting process ... I'll close this and I'll assume that was the issue in the first place... Thanks for helping... If I'm able to reproduce it in any other way I'll open it again.

closed 2021-02-09 by ChromaMaster

noyelseth, 2021-10-16:
## Environment

- Module or chip used: ESP32-S2-WROOM-I
- IDF version: v4.2.1-141-g1e3638390
- Build System: idf.py
- Operating System: Windows10
- Using an IDE?: Eclipse
- Power Supply: USB

## Problem Description

I'm not able to read & write a file using non-encrypted FATFS with wear levelling, secure boot and flash encryption. I'm using wear levelling partition to store some states/errors/values into separate files. I'm using a maximum of 4 files and use file operation one at a time.
This problem is shown in some of the devices randomly likes 1 out of 10 devices.

I'm using the wrapper function to write and read files into the wear levelling partition.

# Write function
``` c
 esp_err_t write_into_file(const char *file_path, void *data, size_t data_len) {
	vTaskDelay(pdMS_TO_TICKS(100));
	ESP_LOGI(TAG, "Writing File: %s", file_path);
	FILE *fptr = fopen((const char *)file_path, "wb");
	if (fptr == NULL) {
		ESP_LOGE(TAG, "Failed to open file for writing: %s", file_path);
		return ESP_FAIL;
	}

	vTaskDelay(pdMS_TO_TICKS(100));
	size_t w_byte = fwrite(data, data_len, 1, fptr);

	vTaskDelay(pdMS_TO_TICKS(100));

	fclose(fptr);

	if (w_byte != 1) { // write 1 number of member into to file
		ESP_LOGE(TAG, "Failed to write data into the file: %s", file_path);
		return ESP_FAIL;
	}
	ESP_LOGI(TAG, "");
	return ESP_OK;
} 
```

# Read function

```c
/**
 *
 */
esp_err_t read_from_file(const char *file_path, void *data, size_t data_len) {
	vTaskDelay(pdMS_TO_TICKS(100));
	ESP_LOGI(TAG, "Reading File: %s", file_path);
	FILE *fptr = fopen((const char *)file_path, "rb");
	if (fptr == NULL) {
		ESP_LOGE(TAG, "Failed to open file for reading: %s", file_path);
		return ESP_FAIL;
	}

	vTaskDelay(pdMS_TO_TICKS(100));

	size_t r_byte = fread(data, data_len, 1, fptr);

	vTaskDelay(pdMS_TO_TICKS(100));

	fclose(fptr);

	if (r_byte != 1) {
		ESP_LOGE(TAG, "Failed to read data from the file: %s", file_path);
		return ESP_FAIL;
	}
	ESP_LOGI(TAG, "");
	vTaskDelay(pdMS_TO_TICKS(100));
	return ESP_OK;
}

```

# Functions Usages

``` c
uint8_t cust_Mac[6] = { 0 };

```