espressif/esp-idf #6406: 'idf.py build' appears to be single threaded

tpoppe, 2021-01-14:
When I run `idf.py build`, my machine is largely idle and only a single core is used.  For large builds, this unnecessarily extends the build time to the point of frustrating waste of time.

Please update idf.py to be multi-threaded or multi-processed so that more machine resources are utilized, reducing build time and frustration.

igrr, 2021-01-15:
Hi @tpoppe, idf.py uses CMake to generate the build script. The build process is done by Ninja (if available) or GNU Make. 

Ninja is multithreaded by default, it will detect the number of processors on the computer. 

When GNU Make is used, idf.py detects the number of processors and sets concurrence via `-j` argument:

https://github.com/espressif/esp-idf/blob/2bfdd036b2dbd07004c8e4f2ffc87c728819b737/tools/idf_py_actions/constants.py#L32

Some information requested by the issue template is missing from your report, could you please fill it in? Operating system, ESP-IDF version, and other information about your environment would be helpful to diagnose the problem.

dobairoland, 2021-01-15:
Hi @tpoppe. How these experiments of yours relate to https://github.com/espressif/esp-idf/issues/6407? Have you enabled Windows defender bypass in the installer or not?

dobairoland, 2021-02-28:
I'm closing this for inactivity. Please let us know if this is still an issue.

closed 2021-02-28 by dobairoland