espressif/esp-idf #6407: Windows Defender policy bypass rules are excessively persmissive

tpoppe, 2021-01-14:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- IDF version : v4.2
- Operating System:   Windows

## Problem Description

During installation, ESP-IDF executes a PowerShell command to allow `%userprofile%/.espressif/*.exe` to bypass Windows Defender policy. It is unclear why exactly this policy bypass is required and why *all* executables require this permission.

The rationale for the bypass policy needs to be well documented and the scope should be tightly controlled so as not to be excessive.

georgik, 2021-01-15:
@tpoppe Thank you for opening this topic.

Right now the installer offers only two options which are controlled by a checkbox on one installation screen with the title: `Register the ESP-IDF Tools executables as Windows Defender exclusions (improves compilation speed, requires elevation)`
- option 1 (checkbox==not selected) - do not add any exclusion
- option 2 (checbbox==selected) - add all *.exe files from IDF tools

In the past, there were just a few files covered by the exclusion. As ESP-IDF grew the more exe-files were automatically added to the exclusion list. There are now more than 100+ exclusions set to Windows Defender. The PowerShell script for setting up exclusions is here: https://github.com/espressif/esp-idf/blob/master/tools/windows/tool_setup/tools_WD_excl.ps1

The main reason for the exclusions of all *.exe was to speed up the build process, which was slowed down by Windows Defender about 3-5 times.

Meanwhile, Microsoft improved Windows Defender and Windows 20H2 seems to perform well even without exclusions.

The concept of the script requires some refinement and improvement.

An idea to solve the problem:
- create a list of most commonly executed binaries during the build process, measure impact on build time with/without Windows Defender, select binaries with a difference of more than 5 seconds of cumulative time
- add a tool to display the list of exclusions with an explanation in post-install phase and ask a user for consent
- apply changes if consent was granted
- the tool should provide also an option to revoke/manage existing exclusions

Would an implementation of the idea solve the problem? What's your opinion @tpoppe?

georgik, 2021-04-30:
@tpoppe We have an update regarding Windows Defender.

We made several measurements that indicate that Windows Defender increases build-time by about 30%. Some other antiviruses are increasing build time even more than 50%. 

The measurements also indicate that it's not sufficient to exclude just one tool like CMake or Ccache. It's necessary to exclude a majority of tools including toolchains.

We added support to our idf-env tool to add/remove the registration of extensions so that users can choose what to exclude and what won't be excluded. Right now, we have CLI version of the tool. We appreciate the feedback.

The quick way to exclude everything in .espressif/tools (PowerShell command):
```
Invoke-WebRequest 'https://dl.espressif.com/dl/idf-env/idf-env.exe' -OutFile .\idf-env.exe
.\idf-env.exe antivirus exclusion add --all
```
The command will produce the same behavior as PowerShell script in the previous installer.

Command to remove all exclusions:
```
.\idf-env.exe antivirus exclusion remove --all
```

It's possible to register a single tool from .espresif/tool by command (e.g. cmake):
```
.\idf-env.exe antivirus exclusion add --tool cmake
```

It's possible to register a single Path and Process exclusion by:
```
.\idf-env.exe antivirus exclusion add --path "C:\Users\esp\....\ar.exe"
```

The tool also contains support for measuring build time (requires IDF installed by Installer 2.7+):
```
idf-env idf build --repeat 10
```
The command repeats fullclean and build of blink example ten times. Results on build time became stable after the 2nd build.

Please, let us know whether the proposed solution helps to address the reported issue.
```

georgik, 2021-05-06:
@tpoppe We added support for Windows Defender exclusions to idf-env tool and we updated the installer https://dl.espressif.com/dl/esp-idf/ in version 2.8 the text with checkbox was changed to:
Register the ESP-IDF Tools executables as Windows Defender exclusions. The registration might improve compilation time by about 30%. The installer deployed the files on the operating system and antivirus software scanned them. The registration of exclusions requires the elevation of privileges. The exclusions can be added/removed by idf-env tool. More details: https://github.com/espressif/idf-env.

A user can choose not to allow the exclusion during the installation.

A user can change the configuration by using idf-env tool in both ways - add or remove exclusions. The scope can be all *.exe files in .espressif/tools or just specific tool.

Feel free to reopen the issue or suggest different solution.

closed 2021-06-08 by Alvin1Zhang