espressif/esp-idf #6408: Problem with USB OTG in ESP-32S2. Do you really support it?

wojciech-angelo, 2021-01-15:
Welcome

According to Your documentation:
https://www.espressif.com/sites/default/files/documentation/esp32-s2-wroom_esp32-s2-wroom-i_datasheet_en.pdf
page 3 section Hardware

> Hardware
• Interfaces: GPIO, SPI, LCD, UART, I2C, I2S,
Camera interface, IR, pulse counter, LED PWM,
TWAITM (compatible with ISO 11898-1), **USB
OTG 1.1**, ADC, DAC, touch sensor, temperature
sensor

and in more places in your documentation, there is mentioned that your ESP32-S2 embedded, Xtensa® single-core
32-bit LX7 microprocessor is equipment with USB OTG. 
Unfortunately I check your code on https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/get-started/index.html
and in section USB there is no information that you support USB OTG. As far as I know as a Embedded programmer and IT engineer your documentation includes only code for USB device. 
You told that your product has USB OTG. So include USB device but also USB HOST.
My question is why I don't see that you support this in your documentation.
Please tell me does your chip support USB OTG as you claim, or not?

I am waiting for your response. Please don't make me waiting too long...

igrr, 2021-01-15:
Hi @wojciech-angelo, the USB Host feature is in development. We have added the low level code of the driver in https://github.com/espressif/esp-idf/commit/602a747b310ce005295879586f1f94f9d6ccc77d and the next part (HAL layer) is going to be merged soon. We will update this issue when the USB Host driver is released.

wojciech-angelo, 2021-01-15:
> Hi @wojciech-angelo, the USB Host feature is in development. We have added the low level code of the driver in [602a747](https://github.com/espressif/esp-idf/commit/602a747b310ce005295879586f1f94f9d6ccc77d) and the next part (HAL layer) is going to be merged soon. We will update this issue when the USB Host driver is released.

Thank you for a reply. My employer and I are really interested in this feature. 
Please inform me if your work will be relased...

Please be in touch.
Greetings

T-vK, 2021-03-30:
@igrr Any updates on this?

Dazza0, 2021-06-08:
@T-vK As of commit 7f421048936c6007186db00ed382cd989c4b905e, the Host Controller Driver supports creating pipes of all four transfer types (Control, Bulk, Interrupt, and Isochronous). The HCD API provides enough to control the port and to create pipes to a connected device, but doesn't contain all the enumeration logic (this will be included in the USB Host Driver which is considered the public API).

There isn't much example code but the HCD test cases (located under `components/usb/test/hcd`) a minimal implementation of an MSC-SCSI class and a HID mouse class if you need something to refer to. There are also some internal documentation in `components/usb/maintainers.md`

chegewara, 2021-06-08:
> There isn't much example code

Few examples:
https://github.com/chegewara/esp32s2-usb-hid-host-example
https://github.com/chegewara/esp32-s2-usb-host-cdc
https://github.com/chegewara/esp32s2-usb-host-example

Dazza0, 2021-09-01:
Update: Initial USB Host Library support has been added as of commit accbaee57ceca241ca117a53f486b7f79ed528b9

Currently only asynchronous API is supported. Documentation, synchronous API, and more test to follow.

tore-espressif, 2021-12-14:
Update: USB Host CDC-ACM driver and example added as of commit https://github.com/espressif/esp-idf/commit/dd1b698075fce81f108ce2c211ce936e2ff79cd9

Mass Storage Class can be expected before end of the year.

Dazza0, 2021-12-24:
Update: USB Host MSC driver and example added as of commit  4f7c36d859abdec474646cfec709c0c2d273f500

closed 2022-01-06 by espressif-bot

chipweinberger, 2022-03-21:
Is OTG supported?

i.e selectively switching between Host and Device as per the USB OTG specification?

tore-espressif, 2022-03-25:
Hello @chipweinberger , thank you for your interest in USB Host.

Unfortunately at the moment we can't commit to any deadline about OTG support, even though it is included in our long-term plans. 

Would you mind sharing details of your usecase?

chipweinberger, 2022-03-25:
That’s too bad.

I need to support USB host i’m normal operation, but I want to use USB Jtag
if the device happens to be plugged into a computer.

Without OTG, i need to use some other means of switching between the two
modes., which is more complex for the user.

igrr, 2022-03-25:
Thanks for sharing this use case with us @chipweinberger.

tore-espressif, 2022-03-25:
What you are trying to do is possible only on ESP32-S3. This is a screen shot from ESP32S3 technical reference manual:
[image]

As you can see, USB-OTG and USB-Serial-JTAG are two separate peripherals. You can't switch between them using the On-The-Go. In other words: USB-Serial-JTAG is _not_ a device mode of USB-OTG peripheral.

You have to route the internal phy to either USB-OTG or USB-Serial-JTAG by software. You can detect the connected host/device by the ID pin of micro-USB connector.
[image]

However, we haven't considered this usecase, so we need to double check internally

chipweinberger, 2022-03-25:
Thanks for the useful info. The ID pin might solve it. 

I'd prefer to use USB Type-C port on the ESP32.

The full solution might then look like this?

1. USB-C port on GPIO 19/20 
2. CC pin pulled-down on PCB
3. ESP32-S3 boots, if it detects voltage current on CC pin, it goes into device mode. The host pulled CC high.
4. Otherwise, ESP32-S3 goes into host mode, because the Type-B cable cannot pull CC high.

Does this sound reasonable? Am I abusing USB spec too much?

chipweinberger, 2022-03-26:
I'm reading that I could also add an External USB Transceiver? Such as the **USB1T1103** or **USB1T1105A**? So I could support a computer connected with USB Serial/JTAG and an external USB device connected at the same time. They seem to go for ~$0.60, which is cheaper than a **CP2102N**, but also more complicated to enable.

tore-espressif, 2022-03-29:
Yes, what you're proposing is correct. 

> It would be better If I could do this without involving my app code, instead using fixed function, GPIOs, cheap external circuitry, or bootloader code.

At the momemt, you would have to handle this switching in your app code.

> USB in DFU mode by pulling GPIO 0 low on boot

if you are interested only in _flashing_ and not full debugging access, DFU is the easiest solution. No new code required in your app.

> I could also add an External USB Transceiver

Don't forget that when you want to use OTG and SerialJTAG at the same time (with external USB transceiver) you will need two USB connectors. 

Anyway, from what you are saying it seems to me that you want to cut down your BOM cost while maintaining debug access for _development phase_ of your project. If that's the case, keep in mind that the you don't have to place the usual CP210x chip on your board. You can find it on most of ESP's development boards, but that's just to provide plug-and-play experience with debugging. 

In production scenario, you'd use either external CP210x or native JTAG connection. The debugging circuitry is then not in your product, but rather in external debugger such as [ESP-Prog](https://docs.espressif.com/projects/espressif-esp-iot-solution/en/latest/hw-reference/ESP-Prog_guide.html)

chipweinberger, 2022-03-29:
Thanks for the info. That all matches my understanding.

> Anyway, from what you are saying it seems to me that you want to cut down your BOM cost while maintaining debug access for _development phase_ of your project. 

I wanted to consider adding USB debug capability in consumer devices. If a user encounters a problem, they could collect information easily over USB. It would just be "nice to have" which is why I don't want to add to BOM.

chipweinberger, 2022-04-22:
I found some relevant looking example code for switching these sort of things in software:

https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-tinyusb.c

```
static void usb_switch_to_cdc_jtag(){
    // Disable USB-OTG
    periph_module_reset(PERIPH_USB_MODULE);
    //periph_module_enable(PERIPH_USB_MODULE);
    periph_module_disable(PERIPH_USB_MODULE);

    // Switch to hardware CDC+JTAG
    CLEAR_PERI_REG_MASK(RTC_CNTL_USB_CONF_REG, (RTC_CNTL_SW_HW_USB_PHY_SEL|RTC_CNTL_SW_USB_PHY_SEL|RTC_CNTL_USB_PAD_ENABLE));

    // Do not use external PHY
    CLEAR_PERI_REG_MASK(USB_SERIAL_JTAG_CONF0_REG, USB_SERIAL_JTAG_PHY_SEL);

    // Release GPIO pins from  CDC+JTAG
    CLEAR_PERI_REG_MASK(USB_SERIAL_JTAG_CONF0_REG, USB_SERIAL_JTAG_USB_PAD_ENABLE);

    // Force the host to re-enumerate (BUS_RESET)
    pinMode(USBPHY_DM_NUM, OUTPUT_OPEN_DRAIN);
    pinMode(USBPHY_DP_NUM, OUTPUT_OPEN_DRAIN);
    digitalWrite(USBPHY_DM_NUM, LOW);
    digitalWrite(USBPHY_DP_NUM, LOW);

    // Initialize CDC+JTAG ISR to listen for BUS_RESET
    usb_serial_jtag_ll_disable_intr_mask(USB_SERIAL_JTAG_LL_INTR_MASK);
    usb_serial_jtag_ll_clr_intsts_mask(USB_SERIAL_JTAG_LL_INTR_MASK);
    usb_serial_jtag_ll_ena_intr_mask(USB_SERIAL_JTAG_INTR_BUS_RESET);
    intr_handle_t intr_handle = NULL;
    xSemaphoreHandle reset_sem = xSemaphoreCreateBinary();
    if(reset_sem){
        if(esp_intr_alloc(ETS_USB_SERIAL_JTAG_INTR_SOURCE, 0, hw_cdc_reset_handler, reset_sem, &intr_handle) != ESP_OK){
            vSemaphoreDelete(reset_sem);
            reset_sem = NULL;
            log_e("HW USB CDC failed to init interrupts");
        }
    } else {
        log_e("reset_sem init failed");
    }

    // Connect GPIOs to integrated CDC+JTAG
    SET_PERI_REG_MASK(USB_SERIAL_JTAG_CONF0_REG, USB_SERIAL_JTAG_USB_PAD_ENABLE);

    // Wait for BUS_RESET to give us back the semaphore
    if(reset_sem){
        if(xSemaphoreTake(reset_sem, 1000 / portTICK_PERIOD_MS) != pdPASS){
            log_e("reset_sem timeout");
        }
        usb_serial_jtag_ll_disable_intr_mask(USB_SERIAL_JTAG_LL_INTR_MASK);
        esp_intr_free(intr_handle);
        vSemaphoreDelete(reset_sem);
    }
}
#endif
```

chipweinberger, 2022-07-15:
I've never written ESP32 code like this before. Does this code look correct?

- I want USB Serial/JTAG to use the external PHY
- I want USB OTG to use the internal PHY
- I want to control with software (because I'm afraid to burn eFuses right now 🙂)

```
#include "soc/rtc_cntl_reg.h"

#if CONFIG_IDF_TARGET_ESP32S3
#include "hal/usb_serial_jtag_ll.h"
#endif

void enable_external_phy_for_usb_serial_jtag()
{
    // Enable Software Control of PHY Select (i.e. enables the next line)
    SET_PERI_REG_MASK(RTC_CNTL_USB_CONF_REG, RTC_CNTL_SW_HW_USB_PHY_SEL);

    // Software Control: Set PHY Select to 1
    SET_PERI_REG_MASK(RTC_CNTL_USB_CONF_REG, RTC_CNTL_SW_USB_PHY_SEL);

    // Use External PHY for USB Serial/JTAG
    SET_PERI_REG_MASK(USB_SERIAL_JTAG_CONF0_REG, USB_SERIAL_JTAG_PHY_SEL);
}

// called on PRO-cpu
void app_main() {
  enable_external_phy_for_usb_serial_jtag();
   ...
}
```

chipweinberger, 2022-07-16:
Or perhaps I just need to call code like this at startup?

```
#include "hal/usb_phy_types.h"
#include "esp_private/usb_phy.h"

    usb_phy_handle_t phy_hdl = NULL;

    usb_phy_gpio_conf_t gpio;
    gpio.vp_io_num = 42;   
    gpio.vm_io_num = 41;
    gpio.rcv_io_num = 47; // picked randomly...
    gpio.oen_io_num = 40; 
    gpio.vpo_io_num = 39;  
    gpio.vmo_io_num = 38;  

    //Initialize the internal USB PHY to connect to the USB OTG peripheral
    usb_phy_config_t phy_config = {
        .controller = USB_PHY_CTRL_OTG,
        .target = USB_PHY_TARGET_INT,
        .otg_mode = USB_OTG_MODE_HOST,
        .otg_speed = USB_PHY_SPEED_UNDEFINED,   //In Host mode, the speed is determined by the connected device
        .gpio_conf = &gpio,
    };
    esp_err_t err = usb_new_phy(&phy_config, &phy_hdl);
```

tore-espressif, 2022-07-18:
To everyone seeing this in the future: Please open new issues for new problems, if no other issue describes the problem you see, rather than replying here, we've locked this not because we don't care about issues but because we do care about them, and tracking a number of discussions in a long thread makes it hard for us to address them all.

chipweinberger, 2022-07-18:
Yes, got a bit side tracked. Fair enough!