espressif/esp-idf #6409: Bluetooth stack occasionally crashes/unresponsive when a client is disconnected during data transmission

BonifaceBassey, 2021-01-15:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v2
- Module or chip used:  ESP32-WROOM-32
- IDF version:  v4.0-203-gcfb6ac13c
- Build System:         Make
- Compiler version: xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Windows
- (Windows only) environment type: MSYS2 mingw32.
- Using an IDE?: Yes (VSCode)
- Power Supply:         external 5V

## Problem Description

I wanted to test the robustness of my Bluetooth classic application. So, I have a little application that only sends data out continuously every 50ms once a client is connected.

On the PC side, I have a script that only opens and closes a connection to my ESP32 board as fast as possible.

In the course of that, the Bluetooth stack occasionally crashes or becomes unresponsive when a client is disconnected during data transmission. If the stack becomes unresponsive, Connection/Disconnection is still possible but no data is sent to the client.
If the stack crashes, the watchdog task got triggered. And the only solution left would be to restart the device

### Expected Behavior
I want to be able to use my device without recycling power to it.

### Actual Behavior
The device is restarted before it can be reused.

### Code to reproduce this issue

```cpp
// On ESP32:

while(true) {
    vTaskDelay(50 / portTICK_PERIOD_MS);

    if(clientIsConnected) {
        while(congestionIsActive) ;  // will be set/reset on ESP_SPP_CONG_EVT
        while(sendingIsBusy) ;

        sendingIsBusy = true;  // will be reset on ESP_SPP_WRITE_EVT

        esp_err_t result = esp_spp_write(handle, len, data);
        if (result != ESP_OK()) 
            LOGE("result = %d", result);
    }
} 

// On PC (Python code):
while True:
    status = self.serial.open(self.portNumber)
    time.sleep(delay)
    
    self.serial.close()
    time.sleep(delay)
```

## Debug Logs

```
#When the stack becomes unresponsive, connection/disconnection still possible but no data is sent to the client, I get:

E (56079) BT_BTC: btc_spp_write unable to find RFCOMM slot!

#When the stack crashes and the watchdog got triggered, I get:

E (137173) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (137173) task_wdt:  - IDLE0 (CPU 0)
E (137173) task_wdt: Tasks currently running:
E (137173) task_wdt: CPU 0: BTC_TASK
E (137173) task_wdt: CPU 1: IDLE1
```

Alvin1Zhang, 2021-01-18:
Thanks for reporting and sorry for the inconvenience, we will look into.

boblane1, 2021-01-18:
Hi @BonifaceBassey,

On the PC side, when the connection is established successfully, it will not read the data from the ESP32, am I right? By the way, do you check the parameters `status` and `cong` of  event `ESP_SPP_WRITE_EVT`? Can you supply more code on ESP32?

Thanks!

BonifaceBassey, 2021-01-20:
I have resolved my problem:

When a client is disconnected during data transfer, _**ESP_SPP_CONG_EVT**_ will be triggered.
In my example, I was waiting on _**while(congestionIsActive)**_  forever.
The problem was resolved after I broke out of that loop after some timeout.

closed 2021-01-20 by BonifaceBassey