espressif/esp-idf #6414: Ethernet stop working with dm9051

fabiolino2416, 2021-01-16:
using example hyperf for a long time 600 seconds ethernet interface stops working (after about 100...300 seconds) no error or warning is detected in console only a restart solves the problem spi frequency 20mhz the test is run with two demo boards dm9051 demo_board and wroom32d kit I will run the test again on a production board to exclude any problems related to a demo hardware

MadsHHLund, 2021-01-19:
I have a similar problem.
I use a wiz850:
Connects nicely, gets IP.
But if I want to do "HTTP client connect", then it goes wrong every other time.

In the debug log ,,,
..
D (6134) spi_master: Allocate RX buffer for DMA
D (6134) spi_master: Allocate RX buffer for DMA
D (6134) spi_master: Allocate RX buffer for DMA
D (6134) spi_master: Allocate RX buffer for DMA
D (6134) spi_master: Allocate RX buffer for DMA
D (6134)

At the same time, I can no longer ping the device.

Have tried with all possible combinations of: Clock-speed, Queue-size.

It seems that the SPI communication is stopping. There will be no more interrupts
"emac_w5500_task"

Does anyone have a suggestion on what I should try

MadsHHLund, 2021-01-19:
I forgot to write ...

With a Wifi connection it runs perfectly.

fabiolino2416, 2021-01-23:
the problem seems to be in the interface of the missing flow control drivers esp_eth_mac_dm9051.c we are checking with idf v4.3 by performing dm9051 memory dumps (with the invaluable help of davicom great professionals) I also believe that the bug can intervene (IDFGH-4569) corrupting the flow but it is difficult to locate

MadsHHLund, 2021-01-23:
I think I have found the error in my system.

I have tried switching to another DNS server, after which it seems to run more stably.
Now I want to investigate why DNS does not provide consistent return responses.

MadsHHLund, 2021-01-25:
What a mystery ...

I try to do a "HTTP_CLIENT post"

On the wired network, in my private home office, wiz850io - w5500 runs perfectly, so does wifi.

At work, the wired network fails with DNS resolve error:
esp-tls: couldn't get hostname for xxx

After which - I can no longer "ping" the device.

At work, the same code runs perfectly on WIFI ...

Have tried switching DNS server, and much more.

Is there anyone who can give me, ideas to explore. ??????

MadsHHLund, 2021-01-27:
What a mystery ...

When event IP_EVENT_ETH_GOT_IP occurs,  I make a . esp_netif_set_dns_info () calls and sets secondary dns to 0.0.0.0, 

then everything runs perfectly.

david-cermak, 2021-08-20:
Hi @fabiolino2416 

Sorry for not replying sooner. It seems like the problem you reported is little related to the recent discussion, is it not?
Any update from your end? Do you still see the ethernet interface stop responding?

@MadsHHLund Are you using the wired and wireless configuration at the same time? lwip keeps an array of DNS servers common for all interfaces, couldn't the WiFi rewrite the DNS server somehow?

> When event IP_EVENT_ETH_GOT_IP occurs, I make a . esp_netif_set_dns_info () calls and sets secondary dns to 0.0.0.0,

Could you please check, what was the value before? and where it was set?

david-cermak, 2021-08-20:
@fabiolino2416 We've had a similar problem with missed GPIO interrupt on another device, could you please check if this change won't fix the issue?

```patch
diff --git a/components/esp_eth/src/esp_eth_mac_dm9051.c b/components/esp_eth/src/esp_eth_mac_dm9051.c
@@ -397,7 +397,7 @@ static void emac_dm9051_task(void *arg)
     uint32_t length = 0;
     while (1) {
         // block indefinitely until some task notifies me
-        ulTaskNotifyTake(pdTRUE, portMAX_DELAY);
+        ulTaskNotifyTake(pdTRUE, pdMS_TO_TICKS(1000));
         /* clear interrupt status */
         dm9051_register_read(emac, DM9051_ISR, &status);
         dm9051_register_write(emac, DM9051_ISR, status);
```

david-cermak, 2021-09-03:
Hi @fabiolino2416 

We were able to reproduce the issue after the stress testing. The patch above actually *works around* the problem, as it makes the Rx task to check periodically for any new Rx packets (it didn't address the root cause, though)

Here's a preliminary patch that fixes the issue:
[esp_eth-Fix-dm9051-Rx-interrupt-processing.patch.txt](https://github.com/espressif/esp-idf/files/7107069/esp_eth-Fix-dm9051-Rx-interrupt-processing.patch.txt)

closed 2021-09-10 by espressif-bot