espressif/esp-idf #6415: idf.py does not automatically find serial ports to flash

AxelLin, 2021-01-16:
## Environment

Module or chip used: ESP32-WROOM-32D
IDF version: v4.3-dev-2586-g526f682397a8
Build System: idf.py
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
Operating System: Linux
Power Supply: USB

## Problem Description

Notice below issue after pull recent esp-idf changes:
$ idf.py  encrypted-app-flash monitor
Executing action: encrypted-app-flash
No serial ports found. Connect a device, or use '-p PORT' option to set a specific port.

// Below works
$ idf.py  encrypted-app-flash monitor -p /dev/ttyUSB0

Note, it was working with esp-idf: v4.3-dev-2398-g2bfdd036b2db

dobairoland, 2021-01-16:
We are sorry for the inconvenience. We already fixed it and will be published soon.

jphx, 2021-01-19:
I ran into this also on a fresh install today (under Linux) and tracked it down to [this line](https://github.com/espressif/esp-idf/blob/master/tools/idf_py_actions/serial_ext.py#L23) in `serial_ext.py`:
```python
esp = esptool.get_default_connected_device(serial_list=ports, port=None, connect_attempts=3, initial_baud=args.baud)
```
It seems there is no function named `get_default_connected_device` in the `esptool` module.  It _does_ [exist in esptool.py](https://github.com/espressif/esptool/blob/HEAD/esptool.py#L97), but not in commit 8270199, which is what's being used currently.

AxelLin, 2021-02-02:
I update my tree to v4.3-dev-2940-g6e776946d01e
It's very easy to hit below issue.
(Also note, if I add -p /dev/ttyUSB0 it will always work)

$ idf.py monitor
Executing action: monitor
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____
/dev/ttyUSB0 failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header
No serial ports found. Connect a device, or use '-p PORT' option to set a specific port.

dobairoland, 2021-02-02:
I'm reopening this because there is another followup bug.

closed 2021-03-12 by espressif-bot

AxelLin, 2021-04-13:
Just to remind the v4.3 still needs this fix. (The fix is only in master now)

dobairoland, 2021-04-13:
Thank you @AxelLin for the reminder.

AxelLin, 2021-04-22:
v4.3-beta3 does not include the fix. Maybe need to re-open the issue in case if it get forgetton.

dobairoland, 2021-04-22:
I'm sorry @AxelLin for the inconvenience. It was merged internally a week ago but unfortunately it didn't make it into v4.3-beta3 which was created for testing earlier. Reopening the issue won't make this process faster. I'm sorry for that.