espressif/esp-idf #6417: BLE: Interrupt wdt timeout on CPU0

h2zero, 2021-01-16:
Currently using IDF v3.3.4 @commit cd59d107b. 

Occasionally while attempting to start a BLE scan or connection I get this crash:

```
10:44:59.429 ->  Disconnected: 30:ae:a4:4f:14:e6
10:44:59.529 -> vhci.c 132
10:44:59.866 -> Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
10:44:59.866 -> Core 0 register dump:
10:44:59.866 -> PC      : 0x4008d921  PS      : 0x00060e34  A0      : 0x8008dada  A1      : 0x3ffbe3b0  
10:44:59.866 -> A2      : 0x00000002  A3      : 0x3ffb4e28  A4      : 0x8008f91d  A5      : 0x3ffb9da0  
10:44:59.866 -> A6      : 0x00000000  A7      : 0x4008e3bc  A8      : 0x8008d921  A9      : 0x3ffbe320  
10:44:59.866 -> A10     : 0x0000000b  A11     : 0x0000000b  A12     : 0x0000000c  A13     : 0xffffffff  
10:44:59.866 -> A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
10:44:59.933 -> EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
10:44:59.933 -> Core 0 was running in ISR context:
10:44:59.933 -> EPC1    : 0x4008fa2f  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008d921
10:44:59.933 -> 
10:44:59.933 -> ELF file SHA256: 0000000000000000
10:44:59.933 -> 
10:44:59.933 -> Backtrace: 0x4008d921:0x3ffbe3b0 0x4008dad7:0x3ffbe3d0 0x4010f17f:0x3ffbe3f0 0x4008daf4:0x3ffbe410 0x40082c89:0x3ffbe430 0x40082ef3:0x3ffb5cf0 0x4008d8a9:0x3ffb5d10 0x4011d95d:0x3ffb5d30 0x4010f1d2:0x3ffb5d50 0x4001791d:0x3ffb5d70 0x4010f979:0x3ffb5d90 0x4010fa56:0x3ffb5dd0 0x400188f5:0x3ffb5e00 0x4010f40b:0x3ffb5e40 0x40044a63:0x3ffb5e60 0x40045983:0x3ffb5e80 0x40088352:0x3ffb5ea0 0x40019d11:0x3ffb5ed0 0x40055b4d:0x3ffb5ef0 0x4010ddf3:0x3ffb5f10 0x4010e405:0x3ffb5f30 0x4008f91a:0x3ffb5f60
10:44:59.966 -> 
10:44:59.966 -> Core 1 register dump:
10:44:59.966 -> PC      : 0x4012083a  PS      : 0x00060734  A0      : 0x800fde52  A1      : 0x3ffbc650  
10:44:59.966 -> A2      : 0x00000000  A3      : 0x00000000  A4      : 0x8008fb2c  A5      : 0x3ffc6200  
10:44:59.966 -> A6      : 0x00000000  A7      : 0x3ffbc340  A8      : 0x00060023  A9      : 0x3ffbc1c4  
10:44:59.999 -> A10     : 0x00000000  A11     : 0x00000000  A12     : 0x8008fb2c  A13     : 0x3ffc6200  
10:44:59.999 -> A14     : 0x00000000  A15     : 0x3ffbc340  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
10:44:59.999 -> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
10:44:59.999 -> 
10:44:59.999 -> ELF file SHA256: 0000000000000000
10:44:59.999 -> 
10:44:59.999 -> Backtrace: 0x4012083a:0x3ffbc650 0x400fde4f:0x3ffbc670 0x40090e05:0x3ffbc690 0x4008f91a:0x3ffbc6b0
10:44:59.999 -> 
10:44:59.999 -> Rebooting...

Core 0 decode:
0x40082ef3: cause_sw_intr_to_core_wrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bt.c line 800
0x4008f91a: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

Core 1 decode:
0x4012083a: esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c line 492
0x400fde4f: esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c line 63
0x40090e05: prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 3382
0x4008f91a: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

xulongzhuang, 2021-06-07:
Hi, @h2zero 
I used gatt_client on [cd59d10](https://github.com/espressif/esp-idf/commit/cd59d107b29b27077377d427a446b33012670a06) and failed to reproduce your problem. Could you provide more detailed steps to reproduce it? Compared to this v3.3.4 branch, I hope you switch to the latest release/v3.3 branch or a higher version of the release branch for development.

Alvin1Zhang, 2021-08-04:
@h2zero Thanks for reporting and sorry for slow turnaround, would you please help share if any updates for the issue? Thanks.

h2zero, 2021-08-04:
I was hoping I could reliably reproduce this with a minimal code example but it happened so infrequently I gave up and moved on to other things. FWIW I haven't seen this error in quite some time so I'll close this and reopen if it comes up again and I can gather more data.

closed 2021-08-04 by h2zero

Alvin1Zhang, 2021-08-05:
@h2zero Thanks for sharing the updates, feel free to reopen with more details.