espressif/esp-idf #6418: BT_BTM: No random address yet, please set random address and try

geza-pycom, 2021-01-17:
## Environment

- Module or chip used:  ESP32
- IDF version (run ``git describe --tags`` to find it): v4.1
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): crosstool-NG esp-2020r3 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

Configuring the Bluetooth stack to be secured with RANDOM address works when device booted for the first time but during any subsequent boot the following error is dropped by the Bluetooth library and the device does not advertise, if no proper deinitialization had happened before the previous restart:

**----> Initialization:**
Coex register schm btdm cb faild
D (2821) BT_SMP: SMP_Init
D (2821) BT_SMP: SMDBG l2c smp_l2cap_if_init
D (2861) BT_SMP: SMP_Register state=0
D (2874) BT_SMP: static passkey      0

**----> Calling esp_ble_gap_start_advertising() at this point**
E (2930) BT_BTM: No random address yet, please set random address and try
E (2931) BT_APPL: bta_dm_ble_set_adv_params_all(), fail to set ble adv params.

Also when the error messages are printed out the esp_ble_gap_start_advertising() returns value 0 and no event comes, thus the issue cannot be detected during runtime, only by enabling the internal error messages. There should be an event to be dropped stating there was a problem with the advertisement.

In the following cases the problem does not appear:
1. Very first time the Bluetooth stack is initialized
2. The address is set to PUBLIC and not to RANDOM
3. If the Bluetooth stack is deinitialized properly before initialized again

The probem is that if the deivce resets for any reasons before being able to deinitialize the Bluetooth stack, advertisement will not continue after it restarts again due to this issue.

This does not connect to this problem, but I would like to know why the "_Coex register schm btdm cb faild_" is printed out each time the Bluetooth stack is initialized.

Thanks in advance for the help !

Alvin1Zhang, 2021-01-18:
Thanks for reporting.

Weijian-Espressif, 2022-08-25:
@geza-pycom please check the own type  of adv parameters, if set to 1, you need set random address,  otherwise you will get ESP_GAP_BLE_ADV_START_COMPLETE_EVT with error status.

@blueMoodBHD  please have take a look  why the "Coex register schm btdm cb faild" is printed out each time the Bluetooth stack is initialized.

OverTree, 2023-01-29:
you should invoke this function: `esp_ble_gap_set_rand_addr()`

https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-reference/bluetooth/esp_gap_ble.html?highlight=esp_ble_gap_set_rand_addr#_CPPv425esp_ble_gap_set_rand_addr13esp_bd_addr_t

closed 2023-08-10 by Alvin1Zhang