espressif/esp-idf #6419: Cannot access environment variable from CMakeLists.txt

ammaree, 2021-01-17:
## Environment

- Development Kit:        [various]
- Kit version:                  [not applicable]
- Module or chip used: [various]
- IDF version:                 [v4.3-dev-2586-g526f68239]
- Build System:              [idf.py]
- Compiler version:        [xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0]
- Operating System:     [macOS]
- Using an IDE?:            [No]
- Power Supply:            [various]

## Background
We are trying to simplify our build process to automate the building of firmware for 4 (with more coming) different ESP32 based variants. In order to achieve this we started by breaking the CMake build process into 3 separate components:
1. Common initialization, setting up variables for and calling phase 2 
2. 5 different platform components, adding platform variables and calling phase 3
3. Common wrap up section eventually calling

In order to make the same set of build files work for the Eclipse IDE single build config as well as for the CLI initiated multiple build options we chose to use an environment variable ESP_IDF_MODEL=??????? as the mechanism to pass the selector for the specific platform selector to the 1st phase CMake file.

This process works perfectly when used inside Eclipse and generates the correct firmware based on the platform manually selected. To control the CLI initiated multiple build process we use 2 small scripts. The 1st script  (buildmany) simply cleans the environment, checks the parameter(s), creates the variable, and calls the second script, once for each parameter passed.
```bash
#!/bin/bash
[[ -f "sdkconfig" ]] && rm sdkconfig
[[ -d "build" ]] && rm -r build
while [[ ! -z $1 ]]; do
	if [[ -f "~/Dropbox/devs/ws/z-components/configs/"$1".txt" ]]; then
		echo "#################### $0 Starting build process for " $ESP_IDF_MODEL
		export ESP_IDF_MODEL=$1
#		declare -x ESP_IDF_MODEL=$1	
##### Subshell invocations tried #####
		buildone
#		. buildone
#		bash buildone && wait
		echo "#################### $0 Completed build process for " $ESP_IDF_MODEL
	else
		echo $1 "Invalid HW platform specified !!!!"
	fi
	shift
done
```

The second script (building) does the actual build by calling idf.py.
```bash
#!/bin/bash
echo "#################### $0 Starting build process for " $ESP_IDF_MODEL
idf.py build
# bash idf.py build
# idf.py set-target esp32 build
rm -r build.$ESP_IDF_MODEL
mv build build.$ESP_IDF_MODEL
mv sdkconfig build.$ESP_IDF_MODEL/sdkconfig
echo "#################### $0 Completed build process for " $ESP_IDF_MODEL
```

When we initiate the CLI build process "buildall PLATFORM1 PLATFORM2 PLATFORM3" the process runs correcly 3 times, the variable is found in the second script BUT when the idf.py build process runs, the ESP_IDF_MODEL variable (see code below used at end of 1st phase CMake file) is not found and the manually selected default is built.
```cmake
if( DEFINED $ENV{ESP_IDF_MODEL} )
	set (CONFIG_FILE $ENV{ESP_IDF_MODEL} )
	message( "Found ENVironment var : ${CONFIG_FILE}" )
else()
	set( CONFIG_FILE PLATFORM1 )
#	set ( CONFIG_FILE PLATFORM2 )
#	set ( CONFIG_FILE PLATFORM3 )
	message( "ENV variable ESP_IDF_MODEL not found, set to ${CONFIG_FILE}" )
endif()
include( ${PC}/configs/${CONFIG_FILE}.txt )
project( fake.value.to.avoid.build.warnings )
```

## Problem Description
When idf.py is called from a script the 1st (common init) component is unable to access an environment variable (ESP_IDF_MODEL) used to select the specific 2nd phase CMake file.

### Expected Behavior
The variable should be correctly access and used to select the specific 2nd phase CMake file.

### Actual Behavior
The environment variable is not accessed, alway found as not existing, and the default (hard-coded in the 1st phase CMake file) model selector is used.

### Steps to reproduce

### Code to reproduce this issue

## Debug Logs

## Other items if possible

ammaree, 2021-01-17:
If there is alternative mechanism for setting a variable of passing a parameter from the script file to idf.py and/or the CMake system to select the desired hardware platform we would be happy to consider using that.

igrr, 2021-01-18:
> If there is alternative mechanism for setting a variable of passing a parameter from the script file to idf.py and/or the CMake system to select the desired hardware platform we would be happy to consider using that.

If the `ESP_IDF_MODEL` model variable in your scripts represents different possible values of ESP-IDF compilation target (`esp32`, `esp32s2`, `esp32s3`, etc), then another option is to use `IDF_TARGET` build property in your component CMakeLists file, to do something differently depending on the target. For example,

```
idf_build_get_property(target IDF_TARGET)
message(STATUS "Building for ${target}")
```

I know it doesn't answer your original question "why is the environment variable set in the build script not visible in CMake". However it seems that you can use this simpler method of querying IDF_TARGET variable for the same end goal. Figuring out the reason why the environment variable is not set would involve doing some debugging with your build scripts, I'm afraid.

---

While not exactly part of your question, I'd like to mention that you can also simplify this code:
```
[[ -f "sdkconfig" ]] && rm sdkconfig
[[ -d "build" ]] && rm -r build
...
rm -r build.$ESP_IDF_MODEL
mv build build.$ESP_IDF_MODEL
mv sdkconfig build.$ESP_IDF_MODEL/sdkconfig
```

You can simplify it by leveraging a few idf.py features:
* `-B` options to set the build directory
* `-D` option to set values of CMake variables
* [Target-specific sdkconfig.defaults files](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/build-system.html#target-dependent-sdkconfig-defaults)

1. Create `sdkconfig.defaults` file with common configuration options for all targets, and `sdkconfig.defaults.esp32`, `sdkconfig.defaults.esp32s2`, etc. files for target-specific options.
2. When calling idf.py, specify the build directory, the name of the sdkconfig file to use, and the target:
   ```
   idf.py -B build.esp32 -DSDKCONFIG=sdkconfig.esp32 -DIDF_TARGET=esp32
   ```

As the result, the build system will
* use `esp32` as the target to build for (set using `IDF_TARGET` variable)
* take configuration options from `sdkconfig.defaults`
* append the values from `sdkconfig.defaults.esp32`
* set any configuration options which haven't been set explicitly to their default values
* save sdkconfig file to `sdkconfig.esp32` (the name has no special meaning, you can set `-DSDKCONFIG=anythingyoulike`)
* perform the build in `build.esp32` directory (again, no special meaning, you can set it to any directory).

To build for multiple targets, you could write (in Bash):
```bash
#!/bin/bash
set -eu
for target in esp32 esp32s2 esp32s3; do
    idf.py -B build.${target} -DSDKCONFIG=sdkconfig.${target} -DIDF_TARGET=${target} build
    cp sdkconfig.${target} build.${target}/sdkconfig  # extra step to match your script behavior
done
```

ammaree, 2021-01-19:
HI @igrr 

Thanks for your feedback, problem fixed with your help. Syntax changed from $VAR_NAME to ${VAR_NAME} and all works.

Re the other suggestions:
-DSDKCONFIG we already us to distinguish between platform using WROOM vs WROVER, PSRAM 0/4/8MB and different display options. 

-DIDF_TARGET not our solution, since the platform here is not the ESP vs ESPS2 but a more comprehensive spec incl MCU, sensors, actuators, display and combinations of software functionality.

-B build directory, aware of this but have to first change some CMakeLists.txt files where the build directory is assumed.

Once again thanks, you're a star !!!

André

closed 2021-01-19 by ammaree