espressif/esp-idf #6422: GPIO16 is unable to keep level while in sleep mode

poisonousfish, 2021-01-18:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32D
- IDF version :           4.1
- Build System:         idf.py
- Compiler version:  xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     Windows
- (Windows only) environment type: CMD.
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

当我尝试轻度睡眠并保持GPIO16输出高，我发现GPIO16无法在睡眠后保持高电平.
When I tried to sleep light and keep the GPIO16 output high, I found that GPIO16 couldn't keep the high level after sleep.

我尝试使用gpio_deep_sleep_hold_en（）和gpio_hold_en（16）用这两函数来保持电平
I tried to use gpio_deep_sleep_hold_en () and gpio_hold_en (16) to hold the level

但是esp32进去睡眠模式后GPIO16的电压都会变低
But when ESP32 goes into sleep mode, the voltage of GPIO16 goes down

我尝试使用其他的GPIO,就像GPIO23，GPIO23就可以正常工作
I tried to use another GPIO, like GPIO23, GPIO23 works fine

我不知道哪里出错还是bug，我似乎没有在数据手册中发现GPIO16特殊的地方
I don't know what's wrong or a bug, I don't seem to find anything special about GPIO16 in the data sheet

十分期待你们的回复，谢谢
Looking forward to your reply. Thank you very much

igrr, 2021-01-18:
Hi @poisonousfish. The datasheet doesn't make it obvious, but GPIO16 is in a different power domain (VDD_SDIO). You can find this information in the "IO MUX" table of the Appendix A of the datasheet. Unlike other power domains, VDD_SDIO is powered down in sleep modes.

poisonousfish, 2021-01-19:
> Hi @poisonousfish. The datasheet doesn't make it obvious, but GPIO16 is in a different power domain (VDD_SDIO). You can find this information in the "IO MUX" table of the Appendix A of the datasheet. Unlike other power domains, VDD_SDIO is powered down in sleep modes.

Hi @igrr  You are right.
I can keep VDD_SDIO running in sleep mode?

lennyJL, 2022-04-12:
Hi @EngineerWill. You can use `esp_sleep_pd_config(ESP_PD_DOMAIN_VDDSDIO, ESP_PD_OPTION_ON);` to keep `VDD_SDIO power domain` on before light sleep. Some GPIO are powered by `VDD_SDIO`. IDF have documents for it:
[Wakeup sources](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html#wakeup-sources)->`GPIO wakeup (light sleep only)`->`Warning`.