espressif/esp-idf #6423: Issue Related to the flash the code(PREVIEW TARGETS)

Romit188, 2021-01-18:
./vendors/espressif/esp-idf/tools/idf.py erase_flash flash monitor -p /dev/ttyUSB0 -B build
When is use this command to flash the code on to the ESP32 board.
I see this type of error.
Traceback (most recent call last):
  File "./vendors/espressif/esp-idf/tools/idf.py", line 788, in <module>
    main()
  File "./vendors/espressif/esp-idf/tools/idf.py", line 705, in main
    cli = init_cli(verbose_output=checks_output)
  File "./vendors/espressif/esp-idf/tools/idf.py", line 663, in init_cli
    extensions[name] = import_module(name)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/romit/esp/esp-idf/tools/idf_py_actions/core_ext.py", line 9, in <module>
    from idf_py_actions.constants import GENERATORS, SUPPORTED_TARGETS, PREVIEW_TARGETS
ImportError: cannot import name 'PREVIEW_TARGETS' from 'idf_py_actions.constants' (/home/romit/amazon-freertos/vendors/espressif/esp-idf/tools/idf_py_actions/constants.py)

How to solve this error?

dobairoland, 2021-01-18:
Hi @Romit188.

Please try `idf.py python-clean` to solve the issue. I suspect that you first run the code with latest master and then switched to an older IDF version.

If the above doesn't help then please provide more information as suggested in the issue template. (chip, board version, IDF version, ....)

Romit188, 2021-01-18:
Hi dobairoland.

Thank you for answering. But this command is not work for this error, So i'll try old version of IDF.

So can you help me how to get old version, I am not sure about this.

dobairoland, 2021-01-18:
@Romit188 I don't think switching to an older version would help you.

You have multiple copies of ESP-IDF (each of different version) which are calling each other: `/home/romit/esp/esp-idf` &
`/home/romit/amazon-freertos/vendors/espressif/esp-idf/`. Please examine the value of IDF_PATH by running `echo $IDF_PATH`.

If you want to switch between those IDF's then it is not enough to use `idf.py` from a different directory but you also have to change `IDF_PATH`.

Please read https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#step-2-get-esp-idf. Changing directory before running `install.sh` and `. export.sh` might do the trick.

Romit188, 2021-01-18:
Thank You for your support.
I set IDF_PATH variable and change the version also, But error is not solved yet... 
So if you have another solution pls provide me, As soon as possible...

dobairoland, 2021-01-18:
Please don't forget to rerun `idf.py python-clean` as well. And please post the new error message. It could not be the same.

Romit188, 2021-01-18:
Error is same, Nothing change in it.
But error is related to the PREVIEW_TARGETS environment variable , What is it? Any idea about this?

dobairoland, 2021-01-18:
It is a constant defined here: https://github.com/espressif/esp-idf/blob/master/tools/idf_py_actions/constants.py#L41

`core_ext.py` tries to import it and cannot find it.

Romit188, 2021-01-19:
./vendors/espressif/esp-idf/tools/idf.py erase_flash flash monitor -p /dev/ttyUSB0 -B build
Thank you, I solved the problem related to the preview targets, But new error is generated.
Usage: ./vendors/espressif/esp-idf/tools/idf.py monitor [OPTIONS]
Try './vendors/espressif/esp-idf/tools/idf.py monitor --help' for help.

Error: Invalid value for '--monitor-baud' / '-B': build is not a valid integer

Can you help for solving this error.

Romit188, 2021-01-19:
romit@admin-35:~/amazon-freertos/build$ make all -j4
-- Submodule update
Skipping submodule 'libraries/abstractions/backoff_algorithm/test/unit-test/Unity'
Skipping submodule 'libraries/abstractions/pkcs11/corePKCS11/source/dependency/3rdparty/mbedtls'
Skipping submodule 'libraries/abstractions/pkcs11/corePKCS11/test/cbmc/aws-build-accumulator'
Skipping submodule 'libraries/abstractions/pkcs11/corePKCS11/test/cbmc/aws-templates-for-cbmc-proofs'
Skipping submodule 'libraries/abstractions/pkcs11/corePKCS11/test/unit-test/CMock'
Skipping submodule 'libraries/coreHTTP/test/cbmc/aws-templates-for-cbmc-proofs'
Skipping submodule 'libraries/coreHTTP/test/cbmc/litani'
Skipping submodule 'libraries/coreHTTP/test/unit-test/CMock'
Skipping submodule 'libraries/coreJSON/test/cbmc/aws-templates-for-cbmc-proofs'
Skipping submodule 'libraries/coreJSON/test/cbmc/litani'
Skipping submodule 'libraries/coreJSON/test/unit-test/Unity'
Skipping submodule 'libraries/coreMQTT/test/cbmc/aws-templates-for-cbmc-proofs'
Skipping submodule 'libraries/coreMQTT/test/cbmc/litani'
Skipping submodule 'libraries/coreMQTT/test/unit-test/CMock'
Skipping submodule 'libraries/device_defender_for_aws/test/cbmc/aws-templates-for-cbmc-proofs'
Skipping submodule 'libraries/device_defender_for_aws/test/cbmc/litani'
Skipping submodule 'libraries/device_defender_for_aws/test/unit-test/Unity'
Skipping submodule 'libraries/device_shadow_for_aws/test/cbmc/aws-templates-for-cbmc-proofs'
Skipping submodule 'libraries/device_shadow_for_aws/test/cbmc/litani'
Skipping submodule 'libraries/device_shadow_for_aws/test/unit-test/CMock'
Skipping submodule 'libraries/freertos_plus/standard/freertos_plus_tcp/test/FreeRTOS-Kernel'
Skipping submodule 'libraries/freertos_plus/standard/freertos_plus_tcp/test/litani'
Skipping submodule 'libraries/freertos_plus/standard/freertos_plus_tcp/tools/CMock'
Skipping submodule 'libraries/jobs_for_aws/test/cbmc/aws-templates-for-cbmc-proofs'
Skipping submodule 'libraries/jobs_for_aws/test/cbmc/litani'
Skipping submodule 'libraries/jobs_for_aws/test/unit-test/Unity'
CMake Error at CMakeLists.txt:53 (message):
  Board is not supported: vendor.board

-- Configuring incomplete, errors occurred!
See also "/home/romit/amazon-freertos/build/CMakeFiles/CMakeOutput.log".
See also "/home/romit/amazon-freertos/build/CMakeFiles/CMakeError.log".
Makefile:6931: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1

And also i can't understand how to solve this type of error?

shubhamkulkarni97, 2021-01-19:
Hi @Romit188,

Version of ESP-IDF in Amazon-FreeRTOS is recently upgraded to v4.2. The documentation is not yet updated.

Please try following commands to build and flash the application:

* `export ESPPORT=<ESP32-serial-port>`
* `cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B build -GNinja` (Generates build files)
* `cd build && ninja all -j4` (builds the application)
* `ninja flash` (flash application on ESP32)

You can skip step 3 and 4 by running `./vendors/espressif/esp-idf/tools/idf.py build flash monitor`

Romit188, 2021-01-19:
Hi Subham
Thank you for answering.
I run this command.
ninja all -j4 , in the Build directory.

[483/1647] Performing configure step for 'bootloader'
FAILED: bootloader-prefix/src/bootloader-stamp/bootloader-configure 
cd /home/romit/amazon-freertos/Build/bootloader && /usr/local/bin/cmake -DSDKCONFIG=/home/romit/amazon-freertos/Build/sdkconfig -DIDF_PATH=/home/romit/amazon-freertos/vendors/espressif/esp-idf -DIDF_TARGET=esp32 -DPYTHON_DEPS_CHECKED=1 -DPYTHON=python -DEXTRA_COMPONENT_DIRS=/home/romit/amazon-freertos/vendors/espressif/esp-idf/components/bootloader -DLEGACY_INCLUDE_COMMON_HEADERS= -GNinja /home/romit/amazon-freertos/vendors/espressif/esp-idf/components/bootloader/subproject && /usr/local/bin/cmake -E touch /home/romit/amazon-freertos/Build/bootloader-prefix/src/bootloader-stamp/bootloader-configure
-- Found Git: /usr/bin/git (found version "2.7.4") 
-- Component directory /home/romit/amazon-freertos/vendors/espressif/esp-idf/components/mbedtls does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/romit/amazon-freertos/vendors/espressif/esp-idf/components/unity does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/romit/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/romit/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/romit/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- broken
CMake Error at /usr/local/share/cmake-3.15/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/home/romit/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/romit/amazon-freertos/Build/bootloader/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/local/bin/ninja cmTC_8e5aa && [1/2] Building C object CMakeFiles/cmTC_8e5aa.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_8e5aa
    FAILED: cmTC_8e5aa 
    : && /home/romit/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address   CMakeFiles/cmTC_8e5aa.dir/testCCompiler.c.obj  -o cmTC_8e5aa   && :
    /home/romit/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find crt1-sim.o: No such file or directory
    /home/romit/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find _vectors.o: No such file or directory
    /home/romit/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lsim
    /home/romit/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lhandlers-sim
    /home/romit/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lhal
    collect2: error: ld returned 1 exit status
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /home/romit/amazon-freertos/vendors/espressif/esp-idf/tools/cmake/project.cmake:260 (__project)
  CMakeLists.txt:27 (project)

-- Configuring incomplete, errors occurred!
See also "/home/romit/amazon-freertos/Build/bootloader/CMakeFiles/CMakeOutput.log".
See also "/home/romit/amazon-freertos/Build/bootloader/CMakeFiles/CMakeError.log".
[486/1647] Building C object esp-idf/app_update/CMakeFiles/__idf_app_update.dir/esp_ota_ops.c.obj
ninja: build stopped: subcommand failed.

But give me this type error , related to bootloader....
So can you help me ?

shubhamkulkarni97, 2021-01-19:
@Romit188, you are using older version of xtensa toolchain. ESP-IDF v4.2 is compatible with `xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0`.

To update the toolchain, please follow these steps:
* `cd vendors/espressif/esp-idf`
* `./install.sh` (installs required tools)
* `. ./export.sh` (adds tools to PATH)
* `unset IDF_PATH`

You should be able to run the applications after executing these commands

Romit188, 2021-01-19:
Thank you Shubham , it was work another type of error generated, How to solve this?

-c ../vendors/espressif/esp-idf/components/soc/src/hal/i2s_hal.c
In file included from ../vendors/espressif/esp-idf/components/soc/src/esp32/include/hal/i2s_ll.h:30:0,
                 from ../vendors/espressif/esp-idf/components/soc/include/hal/i2s_hal.h:28,
                 from ../vendors/espressif/esp-idf/components/soc/src/hal/i2s_hal.c:18:
../vendors/espressif/esp-idf/components/soc/include/hal/i2s_types.h:70:31: error: expected ',' or '}' before '__attribute__'
     I2S_COMM_FORMAT_I2S       __attribute__((deprecated)) = 0x01, /*!< I2S communication format I2S, correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
                               ^
../vendors/espressif/esp-idf/components/soc/src/hal/i2s_hal.c: In function 'i2s_hal_format_config':
../vendors/espressif/esp-idf/components/soc/src/hal/i2s_hal.c:117:9: error: case value '3' not in enumerated type 'i2s_comm_format_t {aka const enum <anonymous>}' [-Werror=switch]
         case I2S_COMM_FORMAT_STAND_MSB:
         ^
../vendors/espressif/esp-idf/components/soc/src/hal/i2s_hal.c:125:9: error: case value '4' not in enumerated type 'i2s_comm_format_t {aka const enum <anonymous>}' [-Werror=switch]
         case I2S_COMM_FORMAT_STAND_PCM_SHORT:
         ^
../vendors/espressif/esp-idf/components/soc/src/hal/i2s_hal.c:133:9: error: case value '12' not in enumerated type 'i2s_comm_format_t {aka const enum <anonymous>}' [-Werror=switch]
         case I2S_COMM_FORMAT_STAND_PCM_LONG:
         ^
../vendors/espressif/esp-idf/components/soc/src/hal/i2s_hal.c: At top level:

Romit188, 2021-01-19:
[101/101] Generating binary image from built executable
esptool.py v3.0
Generated /home/romit/amazon-freertos/Build/bootloader/bootloader.bin
ninja: build stopped: subcommand failed.

At this point error was generated.

Romit188, 2021-01-19:
I update the version also, and also build the directory and run "ninja all -j4" command , and it is work also, 
and i run the following command 
 ./vendors/espressif/esp-idf/tools/idf.py erase_flash flash monitor -p /dev/ttyUSB0 -B builde
but not flash the code and give me the error.
shown in below.
Usage: ./vendors/espressif/esp-idf/tools/idf.py monitor [OPTIONS]
Try './vendors/espressif/esp-idf/tools/idf.py monitor --help' for help.

Error: Invalid value for '--monitor-baud' / '-B': builde is not a valid integer

So what is problem i can't understand.

dobairoland, 2021-01-19:
> ./vendors/espressif/esp-idf/tools/idf.py erase_flash flash monitor -p /dev/ttyUSB0 -B builde

If you write it this way then "-p /dev/ttyUSB0 -B builde" is a local argument of "monitor". "-B" is the baudrate for "monitor". You want "-B" as build directory for "idf.py". Also the port should be set for all commands and not just "monitor". The correct order is:

> ./vendors/espressif/esp-idf/tools/idf.py -p /dev/ttyUSB0 -B builde erase_flash flash monitor

Romit188, 2021-01-19:
Executing action: erase_flash
Traceback (most recent call last):
  File "./vendors/espressif/esp-idf/tools/idf.py", line 788, in <module>
    main()
  File "./vendors/espressif/esp-idf/tools/idf.py", line 706, in main
    cli(sys.argv[1:], prog_name=PROG)
  File "/home/romit/.local/lib/python3.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/romit/.local/lib/python3.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/romit/.local/lib/python3.7/site-packages/click/core.py", line 1290, in invoke
    return _process_result(rv)
  File "/home/romit/.local/lib/python3.7/site-packages/click/core.py", line 1224, in _process_result
    value = ctx.invoke(self.result_callback, value, **ctx.params)
  File "/home/romit/.local/lib/python3.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "./vendors/espressif/esp-idf/tools/idf.py", line 622, in execute_tasks
    task(ctx, global_args, task.action_args)
  File "./vendors/espressif/esp-idf/tools/idf.py", line 211, in __call__
    self.callback(self.name, context, global_args, **action_args)
  File "/home/romit/amazon-freertos/vendors/espressif/esp-idf/tools/idf_py_actions/serial_ext.py", line 128, in erase_flash
    esptool_args = _get_esptool_args(args)
  File "/home/romit/amazon-freertos/vendors/espressif/esp-idf/tools/idf_py_actions/serial_ext.py", line 41, in _get_esptool_args
    with open(os.path.join(args.build_dir, "flasher_args.json")) as f:
FileNotFoundError: [Errno 2] No such file or directory: '/home/romit/amazon-freertos/builde/flasher_args.json'

Error is like this.... I can't understand

shubhamkulkarni97, 2021-01-19:
@Romit188 Please check if you have provided correct build directory name in idf.py command. Also check if directory with same name exists.

Romit188, 2021-01-19:
Yeah shubham,
It is correct Build directory.

Romit188, 2021-01-19:
If you have not problem so can you share zip file of the amazon-freertos?

Romit188, 2021-01-19:
Component directory /home/romit/amazon-freertos/vendors/espressif/esp-idf/components/mbedtls does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/romit/amazon-freertos/vendors/espressif/esp-idf/components/unity does not contain a CMakeLists.txt file. No component will be added
-- Checking Python dependencies...

How to solve this error?

shubhamkulkarni97, 2021-01-21:
@Romit188, You can download zip file from amazon-freertos GitHub portal.

> Component directory /home/romit/amazon-freertos/vendors/espressif/esp-idf/components/mbedtls does not contain a CMakeLists.txt file. No component will be added
> -- Component directory [...]

These are just warnings and build should work fine.

mahavirj, 2021-01-28:
Marking this as closed as it this is related to Amazon FreeRTO project. Please see my comment at https://github.com/espressif/esp-idf/issues/6472#issuecomment-768930866

closed 2021-01-28 by mahavirj