espressif/esp-idf #6424: esp_wifi_init returns ESP_ERR_INVALID_ARG when WIFI_INIT_CONFIG_DEFAULT() is used

Chylynsky, 2021-01-18:
Error happens when initializing WiFi:

        esp_err_t result = ESP_OK;
        wifi_init_config_t wifi_init_config = WIFI_INIT_CONFIG_DEFAULT();
        
        result = esp_wifi_init(&wifi_init_config);
        if (result != ESP_OK)
        {
            ESP_LOGE(TAG, "Wi-Fi initialization failed with error code %x [%s].", result, esp_err_to_name(result));
        }

Seems like regression came from 9ff1609a2064c6e964ad27729aa2fbb273c3165e commit.

igrr, 2021-01-18:
@Chylynsky just to check, did you run `git submodule update` after pulling the master branch? What does `git submodule status` report?

Chylynsky, 2021-01-19:
I didn't, that fixed the issue. Thank you for your help.

closed 2021-01-19 by Chylynsky