espressif/esp-idf #6425: strstr() and strnstr() stack overflow

JakubRakus, 2021-01-18:
## Environment

- Module or chip used:  ESP32-WROOM-32D
- IDF version : v4.2-47-g2532ddd9f
- Build System: idf.py
- Compiler version: esp-2020r3 8.4.0
- Operating System:  Linux
- Using an IDE?: Yes, Netbeans
- Power Supply: external 3.3V

## Problem Description

In my app there is an AT command parser which uses string manipulation functions a lot. I've found that `strstr()` or `strnstr()` functions slowly eats free space from task stack (in chunks from 8 to 192 words per call, always multiple of 8), up to the point where free space goes to zero and whole app crashes. Unfortunately it doesn't happen every time - sometimes once for 10 consecutive calls of `strstr`/`strnstr`, sometimes once for 1000 calls.

I'am checking this with simple code:

    ESP_LOGI(TAG,"task stack free: %d",uxTaskGetStackHighWaterMark(NULL));
    sp = strnstr(buffer,"EVENT",len); //search for event message
    ESP_LOGI(TAG,"task stack free: %d",uxTaskGetStackHighWaterMark(NULL));

And the result is for example:

    (5441) rf: task stack free: 308
    (5441) rf: task stack free: 292

### Expected Behavior

Free space on task stack not changed after strstr() or strnstr() calls.

### Code to reproduce this issue

Unfortunately I couldn't paste the whole code here, because I'm implementing proprietary protocol. Raw data are received from UART and then parsed. External ISM radio device sends AT-like commands.

igrr, 2021-01-18:
@JakubRakus thanks for an interesting issue report!
On the ESP32, the strstr function in newlib is built with "optimize for code size" option enabled, so the logic is very simple: https://github.com/espressif/newlib-esp32/blob/78c7e93abd86181307ba9a050f43a353f8853e62/newlib/libc/string/strstr.c#L45-L76. There is no recursion or data-dependent stack usage, as you can see. `strnstr` and `memmem` are similar, you can find the implementations in the same directory.

In the code snippet you have posted,
```c++
ESP_LOGI(TAG,"task stack free: %d",uxTaskGetStackHighWaterMark(NULL));
sp = strnstr(buffer,"EVENT",len); //search for event message
ESP_LOGI(TAG,"task stack free: %d",uxTaskGetStackHighWaterMark(NULL));
```
most of the extra stack usage would come for logging, not from strnstr. Logging (which uses printf internally) can use up to 1k stack from the point where a logging function is called. To check if this is the case, you can try modifying the test as follows. It is possible that "task stack free 1" and "task stack free 2" will report different values.
```c++
ESP_LOGI(TAG,"task stack free 1: %d",uxTaskGetStackHighWaterMark(NULL));
ESP_LOGI(TAG,"task stack free 2: %d",uxTaskGetStackHighWaterMark(NULL));
sp = strnstr(buffer,"EVENT",len); //search for event message
ESP_LOGI(TAG,"task stack free 3: %d",uxTaskGetStackHighWaterMark(NULL));
```

Note that it is also important to interpret the output of uxTaskGetStackHighWaterMark correctly. In your example, the result `308` returned from uxTaskGetStackHighWaterMark doesn't mean that the stack pointer is 308 bytes above the lowest address of the stack. Rather, it means that there are 308 bytes on the stack that have never been used. Therefore, the decrease of uxTaskGetStackHighWaterMark value after some leaf function got called does not necessarily mean that this leaf function used more stack than it used last time. It might mean that the stack pointer was at a lower compared to the previous call. This means, that some other function in the call stack used more stack space, and the leaf function was simply pushed further down the stack.

In general, if you see the size reported by uxTaskGetStackHighWaterMark gradually being reduced, you can try a couple of things to debug this:

1. Enable [CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/kconfig.html#config-freertos-watchpoint-end-of-stack) Kconfig option. This will set a data breakpoint (watchpoint) near the lowest address of the stack, so that when the task stack reaches that address, a debug exception will occur. This will produce a backtrace which you can analyze.
   A typical panic backtrace, decoded with IDF Monitor looks like this:

   ```
   Backtrace: 0x400e14ed:0x3ffb5030 0x400d0802:0x3ffb5050
   0x400e14ed: app_main at /Users/user/esp/example/main/main.cpp:36
   0x400d0802: main_task at /Users/user/esp/esp-idf/components/esp32/cpu_start.c:470
   ```
   The 2nd address in the pair (0x3ffb5030 and 0x3ffb5050 in this case) is the stack pointer value of each frame (frame pointer). In this case, you can subtract 0x3ffb5050-0x3ffb5030=0x20 and obtain that `main_task` function uses 32 bytes of stack. You can perform similar calculations for different functions in your app, and find the stack frame which uses unreasonable amount of stack.

   You may also enable [GDB Stub](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/fatal-errors.html#gdb-stub), it allows inspecting variables in different stack frames at the point when the exception happens.

2. Review the string handling code in your app and check that you aren't using variable-length arrays. Or, if you know you are using them, check that the length of the array is bounded, and is not growing.

JakubRakus, 2021-02-04:
@igrr  thank You very much for detailed explanations and suggestions how to debug the problem. I followed option 1 and from the panic backtrace I've found that bug sits in one of my string parsing functions which was invoked recursively. Rearranging the code to not use recursion solves the problem.

closed 2021-02-04 by JakubRakus