espressif/esp-idf #6427: WiFi auth not working on WROOM-32E (custom PCB) but working on the WROOM-32 (DevKitC)

MarcelKr, 2021-01-18:
## Environment

- Development Kit:     Custom PCB and ESP32-DevKitC clone
- Module or chip used:  ESP32-WROOM-32**E** 4 MB
- IDF version: v4.2-47-g2532ddd9f
- Build System:         CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         external 3.3V

## Problem Description

For some reason the minimal wifi station example is working on the DevKitC (ESP32-WROOM-32), however on my custom PCB with ESP32-WROOM-32**E** it is not. It's exactly the same software I'm flashing. On both boards, I first erased the flash via `idf.py erase_flash`.

### More details regarding the boards
The dev board's HW:

- DevKit 4 C clone
- Powered through USB (Thinkpad docking station)

The custom PCB:

- Powered through custom step down converter based on the LM7525. Can deliver up to 1A. I verified via scope: The Output voltage is less noisy than the USB powered DevKit supply.
- The ESP module is placed on the top right corner of the PCB. There's no PCB under the antenna area.
- Other functions I checked so far worked flawlessly. E.g. LED fading, TWAI, regular GPIO switching.

Attached find the full log of the devkit and the custom PCB variant.

### Expected Behavior
WiFi auth is successful and an IP is received.

### Actual Behavior
- On my custom PCB (WROOM-32E) the board is stuck in an init -> auth, auth -> init loop.
- On my DevKitC (WROOM-32), the connection is successful. 

Both times I flash the exact same software and use the same access point.

### Steps to reproduce

1. Adapt WiFi credentails and build
2. idf.py erase_flash
3. Flash the software
4. Observe the log

### Code to reproduce this iss

I use the WiFi station example. Just adapted the WiFi credentials directly in the C file.

## Debug Logs

The relevant part of the log given by the WROOM-32E board:

```
I (677) wifi_init: WiFi IRAM OP enabled
I (677) wifi_init: WiFi RX IRAM OP enabled
I (787) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (797) wifi:mode : sta (7c:9e:bd:d9:b6:50)
I (797) wifi station: wifi_init_sta finished.
I (1517) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (3467) wifi:state: init -> auth (b0)
I (4467) wifi:state: auth -> init (200)
I (4467) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (4467) wifi station: retry to connect to the AP
I (4467) wifi station: connect to the AP fail
I (6517) wifi station: retry to connect to the AP
I (6517) wifi station: connect to the AP fail
I (6637) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (6637) wifi:state: init -> auth (b0)
I (7637) wifi:state: auth -> init (200)
I (7637) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (7647) wifi station: retry to connect to the AP
```

Full logs attached:

- [WROOM-32--DevKitC.txt](https://github.com/espressif/esp-idf/files/5832196/WROOM-32--DevKitC.txt)
- [WROOM-32E--CustomPCB.txt](https://github.com/espressif/esp-idf/files/5832197/WROOM-32E--CustomPCB.txt)

## Other items if possible

- [sdkconfig.zip](https://github.com/espressif/esp-idf/files/5832283/sdkconfig.zip) -- Note: Wifi credentials are set in the source file itself.

MarcelKr, 2021-01-18:
Just for reference: A few days ago, I also [opened a post in the espressif forum](https://www.esp32.com/posting.php?mode=reply&f=13&t=19005&sid=ad275535b2f8d7acd3f7a42acf9f82ca). However it didn't lead to solving the problem.

Just got a hint that #6427 might be related. I'm not able to find any useful information in the discussion there however.

negativekelvin, 2021-01-18:
Suggest that you compare rssi values from wifi scan example and turn on wifi debugging in menuconfig

MarcelKr, 2021-01-19:
Thanks for the hint about the more verbose wifi output. Didn't know about that.

The signal strenght for both boards is between -55 and -60.

It seems that there's an auth timeout on the WROOM-32E custom PCB. Any Ideas how to debug this further?

```
I (12959) wifi:state: init -> auth (b0)
D (12969) wifi:start 1s AUTH timer
D (12969) wifi:clear scan ap list
D (12969) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (12979) wifi:set max rate: from <rate=130, phy=3, sig=0> to <rate=144, phy=3 sig=0>
D (12989) wifi:sig_b=0, sig_g=0, sig_n=0, max_b=22, max_g=108, max_n=144
D (13069) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (13179) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (13269) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (13369) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (13469) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (13579) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (13669) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (13769) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (13879) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (13969) wifi:auth timeout
I (13969) wifi:state: auth -> init (200)
D (13969) wifi:connect status 1 -> 4
D (13969) wifi:stop beacon/connect timer
D (13969) wifi:reason: auth expire(2)
D (13979) wifi:add bssid dc:39:6f:1e:0d:a7 to blacklist, cnt=0
D (13979) wifi:stop CSA timer
D (13979) wifi:remove dc:39:6f:1e:0d:a7 from rc list
```

Logs:
- [WROOM-32_DevKitC.txt](https://github.com/espressif/esp-idf/files/5834511/WROOM-32_DevKitC.txt)
- [WROOM-32e_CustomPCB.txt](https://github.com/espressif/esp-idf/files/5834512/WROOM-32e_CustomPCB.txt)

YouDONG-ESP, 2021-01-21:
@MarcelKr Thanks for reporting, did you use ADC2? It will cause problem to wifi. could you take a photo of your pcb?

MarcelKr, 2021-01-21:
Hey @YouDONG-ESP, no, I'm using the getting_startet/station example from the ESP-IDF examples. No other periphery was used during the testing.

Here's a photo of focused on the WROOM module. In case you need more, let me know.
[image]

YouDONG-ESP, 2021-01-21:
Is that possible for you to send this pcb to our shanghai base to test the RF functionality?

MarcelKr, 2021-01-21:
Uh sure, I guess that would be possible. Do you have any suspicions of the cause of the error? Can I continue to debug this in any way to avoid being blocked until the shipment arrives and analysis is done?

negativekelvin, 2021-01-21:
One suggestion would be test the module separately before installing to the board using "ESP32 test board small batch burn fixture" available to buy online

YouDONG-ESP, 2021-01-22:
@Marcelkr From the log it tells that the wifi has been actived and esp32 had recived beacon frame. It failed by authentication timeout. So we are not sure if it is a hardware problem.
Dose it only happens on your PCB or it happens in all WROOM-32E? If happens in all wroom-32E, it should be a software problem, please provide a code to reproduce this issue.

MarcelKr, 2021-01-22:
Just tried another one of my custom PCBs (shame on me, I don't know why I didn't do this before) and **it is able to connect successfuly**. I therefore suspect some kind of hardware issue.

@YouDONG-ESP [here's the the source code.](https://github.com/espressif/esp-idf/files/5854889/station.zip). Are you still interested in having the PCB sent to you?

@negativekelvin I don't have any reflow tools around but I'm thinking about getting a mini hotplate (MHP30) to do these kind of things. Thanks for the hint with the test fixture. Didn't know that this thing existed. It could be quite useful.

YouDONG-ESP, 2021-01-25:
@MarcelKr Seems like it's not a issue from ESP32E, you don't need to send it to us, we asked it just for debugging.

MarcelKr, 2021-01-25:
Any idea why it isn't working on one of the boards although all the other functions of the module are working?

YouDONG-ESP, 2021-02-02:
@MarcelKr the problem may comes from the welding? Or this boards is damaged?

ble0, 2022-06-06:
I think this issue still happens on ESP32-S3-WROOM1 DevKitC1. I have two routers Asus RX-AX3000 and UniFi UDM, when those routers perform automatic channel optimisation sometimes the dev boards will fail to connect with repeated logs and I'm using IDF4.4.1
```
I (167379885) wifi:state: init -> auth (b0)
I (167379885) wifi:state: auth -> assoc (0)
I (167379885) wifi:state: assoc -> run (10)
I (167383885) wifi:state: run -> init (2c0)
```
I don't think the issue is related to custom boards as it still happens in the DevKit without anything connected to it.

abhi-ratnman, 2023-02-22:
Hi, We are facing a similar issue. It works on some ESP32 but the same code is unable to connect on some another PCB with ESP32. I hope the team does not take this issue lightly as this may cause huge issues for mass manufacturing of products using ESP32

AxelLin, 2023-06-04:
> I think this issue still happens on ESP32-S3-WROOM1 DevKitC1. I have two routers Asus RX-AX3000 and UniFi UDM, when those routers perform automatic channel optimisation and sometimes the dev boards will fail to connect with repeated logs and I'm using IDF4.4.1. Until I hit the reset button then it may or may not re-connect.
> 
> ```
> I (167379885) wifi:state: init -> auth (b0)
> I (167379885) wifi:state: auth -> assoc (0)
> I (167379885) wifi:state: assoc -> run (10)
> I (167383885) wifi:state: run -> init (2c0)
> ```
> 
> I don't think the issue is related to custom boards as it still happens in the DevKit without anything connected to it.

@ble0 
Do you still hit the issue with recent esp-idf release? (for v4.4 branch, can you reproduce the issue with v4.4.4?)

ble0, 2023-06-07:
hi @AxelLin I do not see the issue since I'm not testing ESP32 anymore :(, hopefully other users won't see this issue moving forward

aureliendeb34, 2023-10-18:
Hi @AxelLin , I have the issue that you show on you last post. My situation is the following:
I have a custom board with ESP32. We've updated our firmware from ESP-IDF 4 to ESP-IDF 5.0. 
On production devices, we have about 10% of them that wasn't able to reconnect to AP.
On the log below, it seems the device is trying to connect to AP several times but it can't.

```
I (2096021) wifi: Enabling WiFi
I (2096031) wifi: WiFi connect: SSID='XXXXXXX' ...
I (2096041) wifi:mode : sta (94:b9:7e:ea:e2:60)
I (2096051) wifi:enable tsf
I (2096091) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2096101) wifi:state: init -> auth (b0)
I (2097111) wifi:state: auth -> init (200)
I (2097111) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2097121) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2097131) wifi:state: init -> auth (b0)
I (2097151) wifi:state: auth -> assoc (0)
I (2097181) wifi:state: assoc -> run (10)
I (2100321) wifi:state: run -> init (2c0)
I (2100331) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2100341) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2100351) wifi:state: init -> auth (b0)
I (2100371) wifi:state: auth -> assoc (0)
I (2100431) wifi:state: assoc -> run (10)
I (2103531) wifi:state: run -> init (2c0)
I (2103541) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
E (2125491) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (2125491) task_wdt:  - IDLE (CPU 1)
E (2125491) task_wdt: Tasks currently running:
E (2125491) task_wdt: CPU 0: IDLE
E (2125491) task_wdt: CPU 1: tiT
E (2125491) task_wdt: Print CPU 1 backtrace

Backtrace: 0x4008E41E:0x3FFC0E80 0x4008341D:0x3FFC0EA0 0x40168621:0x3FFD1200 0x4016868E:0x3FFD1220 0x4016A968:0x3FFD1240 0x4016AA2B:0x3FFD1260 0x40168791:0x3FFD1280 0x40160E85:0x3FFD12A0
0x4008e41e: esp_ptr_in_dram at /home/aurelien/esp/esp-idf/components/esp_hw_support/include/esp_memory_utils.h:48
 (inlined by) esp_flash_write at /home/aurelien/esp/esp-idf/components/spi_flash/esp_flash_api.c:881

0x4008341d: _xt_coproc_exc at /home/aurelien/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1012

0x40168621: udp_sendto_if at /home/aurelien/esp/esp-idf/components/lwip/lwip/src/core/udp.c:654

0x4016868e: udp_sendto_if at /home/aurelien/esp/esp-idf/components/lwip/lwip/src/core/udp.c:665

0x4016a968: icmp_input at /home/aurelien/esp/esp-idf/components/lwip/lwip/src/core/ipv4/icmp.c:162

0x4016aa2b: icmp_input at /home/aurelien/esp/esp-idf/components/lwip/lwip/src/core/ipv4/icmp.c:219

0x40168791: udp_sendto at /home/aurelien/esp/esp-idf/components/lwip/lwip/src/core/udp.c:583

0x40160e85: dns_send at /home/aurelien/esp/esp-idf/components/lwip/lwip/src/core/dns.c:864

E (2145491) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (2145491) task_wdt:  - IDLE (CPU 1)
E (2145491) task_wdt: Tasks currently running:
E (2145491) task_wdt: CPU 0: IDLE
E (2145491) task_wdt: CPU 1: tiT
E (2145491) task_wdt: Print CPU 1 

```

then the watchdog task is trigged in loop. I can't identify why sometimes on some AP I've got this issue. The issue is really difficult to reproduce because it appears only sometimes.
Please could you tell me which problem occurs when we have: `I (2103531) wifi:state: run -> init (2c0)`

If you need more information to help, ask me. (for example the sdkconfig file or something else) thanks for your help. I continue to investigate on my issue. BR

MaxwellAlan, 2023-12-05:
Hi @aureliendeb34 

> Please could you tell me which problem occurs when we have: I (2103531) wifi:state: run -> init (2c0)

(2c0) means STA receive AP's deauth frame, the reason code is 2 (auth expire), refer to [reason code](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html?highlight=reason%20code#wi-fi-reason-code)

We speculate that the issue may be related to abnormalities in the router. When reproducing the problem, try connecting to the router with another STA (Station) and see if the connection is normal. 

Additionally, please provide wireless packet capture during the process,  refer to [how to capture wireless pkts](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wireshark-user-guide.html) ?

Thanks.

MaxwellAlan, 2024-01-12:
Hi @aureliendeb34 

Since the source issue is different from the new issue, we will close this issue and feel free to open new issue if need.

Thanks.

closed 2024-01-12 by MaxwellAlan