espressif/esp-idf #6428: Trouble in PPPoS example - The method send_wait doesn't work when it is used before PPP

jonathandreyer, 2021-01-18:
## Environment

- Development Kit: none
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.1 ([upstream](https://github.com/jonathandreyer/esp-idf/tree/release/v4.1) version)
- Build System: PlatformIO (2.1.0)
- Operating System: macOS
- Using an IDE?: VSCode with PlatformIO
- Power Supply: USB or external 5V

## Problem Description

In the example [PPPoS](https://github.com/espressif/esp-idf/tree/master/examples/protocols/pppos_client), the SMS sending work fine (when it is enabled), but when the [code of SMS sending](https://github.com/espressif/esp-idf/blob/master/examples/protocols/pppos_client/main/pppos_client_main.c#L309-L313) is called after line [L285](https://github.com/espressif/esp-idf/blob/526f682397a8cfb74698c601fd2c5b30e1433837/examples/protocols/pppos_client/main/pppos_client_main.c#L285) (before the PPP example) the method `esp_modem_dte_send_wait` doesn't work properly.
This bug is maybe related to #5695.

### Expected Behavior

The code of sending SMS should work before or after the PPP example.

### Actual Behavior

The execution failed [here](https://github.com/espressif/esp-idf/blob/526f682397a8cfb74698c601fd2c5b30e1433837/examples/protocols/pppos_client/components/modem/src/esp_modem.c#L360) in the timeout.

### Steps to reproduce

Remove [codes](https://github.com/espressif/esp-idf/blob/master/examples/protocols/pppos_client/main/pppos_client_main.c#L309-L313) after line [L285](https://github.com/espressif/esp-idf/blob/526f682397a8cfb74698c601fd2c5b30e1433837/examples/protocols/pppos_client/main/pppos_client_main.c#L285) or remove all codes in examples related to Netif, PPP & MQTT.

lhespress, 2021-01-20:
@jonathandreyer Just want to check the IDF commit ID: Is it `8c6596c`?

lhespress, 2021-01-20:
@jonathandreyer Could you try use the patch as follows which apply on IDF commit ID is `8c6596c `:

[pppos_sms.patch.zip](https://github.com/espressif/esp-idf/files/5842210/pppos_sms.patch.zip)

jonathandreyer, 2021-01-20:
Thanks for your reply.

> @jonathandreyer Just want to check the IDF commit ID: Is it `8c6596c`?

To answer to your question, yes I have used this version of commit. I use platformIO and to use the upstream v4.1, I am forced to edit a bit the *ESP-IDF* repository. It may be important to specify that I am using the upstream versionof the PPPoS example (from the master branch).

I have tested your patch, but unfortunately, it doesn't fix the trouble. The program is stopped because it goes to timeout. The debug log with this new patch:
```
..
I (7115) pppos_example: Battery voltage: 3910 mV
D (7115) pppos_example: set message format ok
D (7115) pppos_example: set character set ok
E (127125) esp-modem: esp_modem_dte_send_cmd(306): process command timeout
E (127125) pppos_example: send command failed
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400848b4
file: "src/main.c" line 319
func: app_main
expression: example_send_message_text(dce, CONFIG_EXAMPLE_SEND_MSG_PEER_PHONE_NUMBER, message)

ELF file SHA256: fe32c866346ea157

Backtrace: 0x40084219:0x3ffb5960 0x400848b7:0x3ffb5980 0x400d2fd6:0x3ffb59a0 0x400db5e6:0x3ffb5ae0 0x40085b65:0x3ffb5b00
  #0  0x40084219:0x3ffb5960 in invoke_abort at /Users/jonathandreyer/.platformio/packages/framework-espidf@src-b0b0460958af2b530905d26a8e55e8c8/components/esp32/panic.c:157
  #1  0x400848b7:0x3ffb5980 in _esp_error_check_failed at /Users/jonathandreyer/.platformio/packages/framework-espidf@src-b0b0460958af2b530905d26a8e55e8c8/components/esp32/panic.c:759
  #2  0x400d2fd6:0x3ffb59a0 in app_main at src/main.c:319 (discriminator 1)
  #3  0x400db5e6:0x3ffb5ae0 in main_task at /Users/jonathandreyer/.platformio/packages/framework-espidf@src-b0b0460958af2b530905d26a8e55e8c8/components/esp32/cpu_start.c:588
  #4  0x40085b65:0x3ffb5b00 in vPortTaskWrapper at /Users/jonathandreyer/.platformio/packages/framework-espidf@src-b0b0460958af2b530905d26a8e55e8c8/components/freertos/port.c:143

Rebooting...
..
```

I have the feeling that the initial trouble (before your patch) is come from the huge work done by @david-cermak (in #6024) to fix troubles in the switching mode (between PPP to AT). Maybe, it is something a wrong initialization of the communication in the *esp_modem* with interrupt, pattern, etc.

lhespress, 2021-01-21:
@jonathandreyer Did you get the `Operator` information when output the LOG?

jonathandreyer, 2021-01-21:
@lhespress Yes, I have just checked. The `Operator` field is correct and well indicated in the console. Unfortunately, the method `example_handle_phone_number` is not called.

lhespress, 2021-01-22:
@jonathandreyer 
Please checkout the patch as follows:

[modem.patch.zip](https://github.com/espressif/esp-idf/files/5855820/modem.patch.zip)

jonathandreyer, 2021-01-22:
@lhespress Thanks for your new patch. I have tested this new version and I didn't have the trouble. Now, it is possible to send SMS before and after the PPP mode.

jonathandreyer, 2021-01-22:
@lhespress I have studied the patch and I have improved it. You can see it [here](https://github.com/jonathandreyer/esp32-sim7000/commit/cdc6ba57e6675ffaad20c48d548d6702c1b2247b). In brief, it is undoing the modification of timeout (10ms are OK), improve the check of result and fix a warning in console (by `dce->handle_line = NULL` in the main).

closed 2021-04-15 by espressif-bot