espressif/esp-idf #6430: Receiving from controller stops after a specific amount of packets

EspLover55, 2021-01-19:
## Environment

- Development Kit:  ESP32-Wrover-Kit
- Kit version: v4
- Module or chip used: ESP32-WROVER-B
- IDF version: 4.3-dev-2586-g526f68239
- Build System: idf.py
- Compiler version: (crosstool-NG esp-2020r3) 8.4.0
- Operating System: macOS
- IDE: CLion (but building from terminal with idf.py)
- Power Supply: USB

## Problem Description

Hi all,
I use the libbtm_app.a API functions: esp_vhci_host_check_send_available, esp_vhci_host_send_packet, esp_vhci_host_register_callback.

 I used the existing bluedroid registrations for the receive function, and shifted it to my code without getting up to the bluedroid stack.

I wrote a little POC code that sends an l2cap echo request, and prints the response.
The problem is that after 17 packets, my receiver doesn't receive anything anymore, although I can still send packets which arrive to the destination. I can see the destination sending my echo response packets, and my ESP32 just does not get them.

### Expected Behavior

The code runs forever and not stopping so fast.

### Actual Behavior

The hci does not pass me packets after 17 packets.

xiongweichao, 2021-01-19:
Hi, Can you give us more information?
Provide code to repeat the issue or indicate the interval at which echo_req was sent.

Wth-Esp, 2021-02-05:
@EspLover55  Any update?

closed 2021-03-12 by Alvin1Zhang