espressif/esp-idf #6431: Uart devkitc v4

quim8, 2021-01-19:
I have 3 devkits, one it is older with a esp32-wroom-32 (I don't know the model but problably a DOIT Esp32 DevKit v1) and then I have 2 DevKitC v4. 

I started my code with the bt_spp_acceptor example code and then I added some code to recive data through uart2, pins 16 and 17. And send it through bt. 

So with the DOIT Esp32 DevKit v1 the code works fine and I see the data on my computer BUT, with one DevKitC v4 I don't recive any data, it reads 0 bytes from the buffer, and with the other DevKitC v4 if I change the rx pin to 17 works ok, but sometimes losses data. The rx pin have a 1k resistor and 1nF capacitor filter for the bus and the tx only the 1k resistor, this bus consists on one line.
[image]

The uart initialization code is the following:
```
static void spp_uart_init(void)
{
	bool install;
    uart_config_t uart_config = {
        .baud_rate = 4800,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_ODD,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
        //.rx_flow_ctrl_thresh = 122,
        .source_clk = UART_SCLK_APB,
		//.source_clk = UART_SCLK_REF_TICK
    };
    //Set UART parameters
    uart_param_config(UART_NUM_2, &uart_config);
    //Set UART pins
    vTaskDelay(100);
    ESP_ERROR_CHECK(uart_set_pin(UART_NUM_2, 16, 17, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE)); //17 16 is the correct
    vTaskDelay(100);
    //ESP_ERROR_CHECK(uart_set_pin(UART_NUM_2, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE));
    //Install UART driver, and get the queue.
    //ESP_ERROR_CHECK(uart_driver_install(UART_NUM_2, 4096, 8192, 10, &spp_uart_queue, 0));
    ESP_ERROR_CHECK(uart_driver_install(UART_NUM_2, BUF_SIZE*2, BUF_SIZE*2, 20, &spp_uart_queue, 0));
    vTaskDelay(100);
    install = uart_is_driver_installed(UART_NUM_2);
    if(install == true){
    	ESP_LOGI(SPP_TAG, "Correct driver install"); //correct install
    }else{
    	ESP_LOGI(SPP_TAG, "Driver install failed");
    }
    if(!uart_task_handle){
       			xTaskCreate(rx_task, "rxTask", 4096, NULL, 1, &uart_task_handle);
       			//xTaskCreate(uart_task, "uartTask", 4096, NULL, 1, &uart_task_handle);
                if(!uart_task_handle){
                	ESP_LOGI(SPP_TAG, "Uart Task Start Failed!");
                	return;
                }
    }
    esp_task_wdt_init(WDT_TIMEOUT, 1); //enable panic so ESP32 restarts
    esp_task_wdt_add(uart_task_handle); //add current thread to WDT watch
}
```

Then the uart rx task:
```
void rx_task(void *pvParameter){
	const int uart_num = UART_NUM_2;
	uint8_t *data = (uint8_t *) malloc(BUF_SIZE);
	int length = 0;
	TickType_t xLastWakeTime;
    const TickType_t xFrequency = 10;
    // Initialise the xLastWakeTime variable with the current time.
    xLastWakeTime = xTaskGetTickCount ();
	for (;;) {
		esp_task_wdt_reset();
		//printf("rx_uart_task\n");
		ESP_ERROR_CHECK(uart_get_buffered_data_len(uart_num, (size_t*)&length));
		//length = uart_read_bytes(uart_num, data, length, 100/portTICK_RATE_MS);
		length = uart_read_bytes(uart_num, data, length, portMAX_DELAY);
		/*printf("Length: %d\n", length);
		printf("Data: ");
		for(int i = 0; i < length; i++) {
		       printf("%d ", data[i]);
		}
		printf("\n");*/
		if (length != 0 && _spp_client != 0){
            _spp_queue_packet((uint8_t *)data, length);
        }else{
        	//printf("No client or data\n");
        }
        vTaskDelayUntil(&xLastWakeTime, xFrequency);
	}
	vTaskDelete(NULL);
	uart_task_handle = NULL;
}

```
The res of the code I adapted from the Arduino BluetoothSerial library: BluetoothSerial.cpp [https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/src/BluetoothSerial.cpp](url)
```
static esp_err_t _spp_queue_packet(uint8_t *data, size_t len){
	//printf("Queue the packet\n");
    if(!data || !len){
    	ESP_LOGI(SPP_TAG, "No data provided");
        return ESP_OK;
    }
    spp_packet_t * packet = (spp_packet_t*)malloc(sizeof(spp_packet_t) + len);
    if(!packet){
        ESP_LOGI(SPP_TAG, "SPP TX Packet Malloc Failed!");
        return ESP_FAIL;
    }
    packet->len = len;
    memcpy(packet->data, data, len);
    if (xQueueSend(_spp_tx_queue, &packet, portMAX_DELAY) != pdPASS) {
        ESP_LOGI(SPP_TAG, "SPP TX Queue Send Failed!");
        free(packet);
        return ESP_FAIL;
    }
    return ESP_OK;
}

static uint8_t _spp_tx_buffer[SPP_TX_MAX];
static uint16_t _spp_tx_buffer_len = 0;

static bool _spp_send_buffer(){
    if((xEventGroupWaitBits(_spp_event_group, SPP_CONGESTED, pdFALSE, pdTRUE, portMAX_DELAY) & SPP_CONGESTED) != 0){
        esp_err_t err = esp_spp_write(_spp_client, _spp_tx_buffer_len, _spp_tx_buffer);
        if(err != ESP_OK){
            ESP_LOGI(SPP_TAG, "SPP Write Failed! [0x%X]", err);
            return false;
        }
        _spp_tx_buffer_len = 0;
        if(xSemaphoreTake(_spp_tx_done, portMAX_DELAY) != pdTRUE){
            ESP_LOGI(SPP_TAG, "SPP Ack Failed!");
            return false;
        }
        return true;
    }
    return false;
}

static void _spp_tx_task(void * arg){
    spp_packet_t *packet = NULL;
    size_t len = 0, to_send = 0;
    uint8_t * data = NULL;
    //const TickType_t xDelay = 20 / portTICK_PERIOD_MS;
    TickType_t xLastWakeTime;
    const TickType_t xFrequency = 20;
    // Initialise the xLastWakeTime variable with the current time.
    xLastWakeTime = xTaskGetTickCount();
    for (;;) {
    	//esp_task_wdt_reset();
    	//printf("spp_tx_task\n");
        if(_spp_tx_queue && xQueueReceive(_spp_tx_queue, &packet, portMAX_DELAY) == pdTRUE && packet){
        	//printf("spp_tx_task: There is data in the queue to send\n");
            if(packet->len <= (SPP_TX_MAX - _spp_tx_buffer_len)){
                memcpy(_spp_tx_buffer+_spp_tx_buffer_len, packet->data, packet->len);
                _spp_tx_buffer_len+=packet->len;
                free(packet);
                packet = NULL;
                if(SPP_TX_MAX == _spp_tx_buffer_len || uxQueueMessagesWaiting(_spp_tx_queue) == 0){
                    _spp_send_buffer();
                }
            }else {
                len = packet->len;
                data = packet->data;
                to_send = SPP_TX_MAX - _spp_tx_buffer_len;
                memcpy(_spp_tx_buffer+_spp_tx_buffer_len, data, to_send);
                _spp_tx_buffer_len = SPP_TX_MAX;
                data += to_send;
                len -= to_send;
                _spp_send_buffer();
                while(len >= SPP_TX_MAX){
                    memcpy(_spp_tx_buffer, data, SPP_TX_MAX);
                    _spp_tx_buffer_len = SPP_TX_MAX;
                    data += SPP_TX_MAX;
                    len -= SPP_TX_MAX;
                    _spp_send_buffer();
                }
                if(len){
                    memcpy(_spp_tx_buffer, data, len);
                    _spp_tx_buffer_len += len;
                    if(uxQueueMessagesWaiting(_spp_tx_queue) == 0){
                        _spp_send_buffer();
                    }
                }
                free(packet);
                packet = NULL;
            }
        }else{
            ESP_LOGI(SPP_TAG, "Something went horribly wrong");
        }
        //vTaskDelayUntil(&xLastWakeTime, xFrequency);

    }
    vTaskDelete(NULL);
    _spp_task_handle = NULL;
}
```

I also tried the Arduino SerialToSerialBT example with Arduino IDE and worked fine on all the kits...
```
void setup(){
  //Serial2.begin(4800);
  Serial.begin(4800, SERIAL_8O1, 16, 17); //set uart for EgoBus
  SerialBT.begin("ESP32_COM_BOX_x"); //Bluetooth device name ESP32_COM_BOX_x, match with com box number
  Serial.println("The device started, now you can pair it with bluetooth!");
   //create a task that will be executed in the Task1code() function, with priority 1 and executed on core 0
  xTaskCreate(
                    Task1code,   /* Task function. */
                    "Task1",     /* name of task. */
                    10000,       /* Stack size of task */
                    NULL,        /* parameter of the task */
                    1,           /* priority of the task */
                    &Task1);      /* Task handle to keep track of created task */                 
  delay(200);                  
  esp_task_wdt_init(WDT_TIMEOUT, true); //enable panic so ESP32 restarts
  esp_task_wdt_add(Task1); //add current thread to WDT watch
}
int last = millis();

void Task1code( void * pvParameters ){
  Serial.print("Task1 running on core ");
  Serial.println(xPortGetCoreID());
  for(;;){
    // resetting WDT every 2s, 5 times only
    if (millis() - last >= 2000) {
        //Serial.println("Resetting WDT...");
        esp_task_wdt_reset();
        last = millis();
    }
    if (Serial.available()) {
      SerialBT.write(Serial.read());;
    }
    /*if (SerialBT.available()) {
      Serial.write(SerialBT.read());
    }*/
    delay(1);
  }
}
```

Wth-Esp, 2021-01-20:
Hi @quim8 

Please provide the basic information to reproduce the problem as in other issues.

And what `DOIT` stands for in `DOIT Esp32 DevKit v1`? Is it a module designed by ESPRESSIF? If not, how can I get it?

Moreover, could you please provide the demo which can reproduce the problem? Which can save much more time, thanks.

Thanks

quim8, 2021-01-20:
Just run the code in the v4 kit and printing the uart data on the rx_task, the data is not what it should be or I don't recive any data. I checked the rx pin with an osciloscope and there are uart frames. And with the Arduino code works fine.

DOIT is the brand that made this kit. But I can confirm it because it is soldered to a board.

My Eclipse project is the following:
[ESP32_COM_BOX_BTClassic_Eclipse.zip](https://github.com/espressif/esp-idf/files/5840870/ESP32_COM_BOX_BTClassic_Eclipse.zip)

Wth-Esp, 2021-01-20:
@quim8  Thanks for the information you provided. We will test the demo ASAP.

Wth-Esp, 2021-01-21:
@quim8 The circuit appears to be fine and as you said, with Arduino code, everything works fine, so please check the code.

quim8, 2021-01-21:
I also think that, but how it is possible that with the DOIT Esp32 DevKit v1 (ESP32-WROOM-32) my code works fine but, with the DevKitC v4 (ESP32-WROOM-32D) I don't recive any data or wrong one?

quim8, 2021-01-25:
I changed the initialization of the uart and removed a queue I created and now works fine. I Really don't know what was the problem. This is the code:

```
void rx_task(void *pvParameter){
	const int uart_num = UART_NUM_2;
	static const char *RX_TASK_TAG = "RX_TASK";
	uint8_t *data = (uint8_t *) malloc(BUF_SIZE);
	int length = 0;
	TickType_t xLastWakeTime;
    const TickType_t xFrequency = 10;
    // Initialise the xLastWakeTime variable with the current time.
    xLastWakeTime = xTaskGetTickCount ();
	for (;;) {
		esp_task_wdt_reset();
		//printf("rx_uart_task\n");
		ESP_ERROR_CHECK(uart_get_buffered_data_len(uart_num, (size_t*)&length));
		length = uart_read_bytes(uart_num, data, length, 100/portTICK_RATE_MS);
		//length = uart_read_bytes(uart_num, data, BUF_SIZE, 100/portTICK_RATE_MS);
        if (length > 0) {
            //data[length] = 0;
            ESP_LOGI(RX_TASK_TAG, "Read %d bytes: '%s'", length, data);
            ESP_LOG_BUFFER_HEXDUMP(RX_TASK_TAG, data, length, ESP_LOG_INFO);
            if (_spp_client != 0){ //if there is someone connected
                   _spp_queue_packet((uint8_t *)data, length); //send through bluetooth
            }
        }else{
        	ESP_LOGI(RX_TASK_TAG, "No uart data");
        }
        vTaskDelayUntil(&xLastWakeTime, xFrequency);
	}
	vTaskDelete(NULL);
	uart_task_handle = NULL;
}

#define TXD_PIN (GPIO_NUM_17)
#define RXD_PIN (GPIO_NUM_16)

static void spp_uart_init(void)
{
	bool install;
    uart_config_t uart_config = {
        .baud_rate = 4800,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_ODD,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
        //.rx_flow_ctrl_thresh = 122,
        .source_clk = UART_SCLK_APB,
		//.source_clk = UART_SCLK_REF_TICK
    };
    //Set UART parameters
    //ESP_ERROR_CHECK(uart_driver_install(UART_NUM_2, BUF_SIZE*2, BUF_SIZE*2, 20, &spp_uart_queue, 0));
    ESP_ERROR_CHECK(uart_driver_install(UART_NUM_2, BUF_SIZE, BUF_SIZE, 0, NULL, 0));
    vTaskDelay(100);
    uart_param_config(UART_NUM_2, &uart_config);
    //Set UART pins
    vTaskDelay(100);
    ESP_ERROR_CHECK(uart_set_pin(UART_NUM_2, TXD_PIN, RXD_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE)); //17 16 is the correct
    vTaskDelay(100);
    //ESP_ERROR_CHECK(uart_set_pin(UART_NUM_2, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE));
    //Install UART driver, and get the queue.
    //ESP_ERROR_CHECK(uart_driver_install(UART_NUM_2, 4096, 8192, 10, &spp_uart_queue, 0));
    install = uart_is_driver_installed(UART_NUM_2);
    if(install == true){
    	ESP_LOGI(SPP_TAG, "Correct driver install"); //correct install
    }else{
    	ESP_LOGI(SPP_TAG, "Driver install failed");
    }
    if(!uart_task_handle){
       			xTaskCreate(rx_task, "rxTask", 4096, NULL, 1, &uart_task_handle);
       			//xTaskCreate(uart_task, "uartTask", 4096, NULL, 1, &uart_task_handle);
                if(!uart_task_handle){
                	ESP_LOGI(SPP_TAG, "Uart Task Start Failed!");
                	return;
                }
    }
    esp_task_wdt_init(WDT_TIMEOUT, 1); //enable panic so ESP32 restarts
    esp_task_wdt_add(uart_task_handle); //add current thread to WDT watch
}
```

Wth-Esp, 2021-01-29:
@quim8

Mhmm, we really don't know the details of your code. Since the issue is fixed, could you please close this issue, if any new issue, feel free to reopen it. 

Thanks.

quim8, 2021-01-29:
I think i know where the problem could be, the tx and rx pin with my configuration ara always at 3.3V. And the bus should be at high impedance i order to receive correct data. How can I do it to make it high impedance? And only activate the tx when needed?
I don't have hardware control. The rts and cts are not connected.

closed 2021-02-08 by quim8