espressif/esp-idf #6432: Feature request: toolchain for Nvidia Jetson Nano, Xavier, TX2  aarch64 hosts?

ArtofWar-dev, 2021-01-20:
Many of us are using the New Nvidia Boards (Linux ARMx64)  for Learning and Development with GPU Accelerated application and AI, and would like to be able to develop for ESP32 using this board. I know there are recent support for Raspberry PI OS, would love to have it available on this platform too.

Thanks.

daveismith, 2021-03-13:
Any estimate on when this will be integrated. The linked issue indicates Q1 and we're getting near the end of Q1. I saw that the latest 4.3 beta release claims support for the new macos M1 platform which I expect would require a new release of the toolchain. Will this platform make it in?

SebastianBar, 2022-03-13:
Any updates on this?

igrr, 2022-03-13:
Hi @SebastianBar, aarch64 toolchains (for Xtensa and RISC-V) are now provided and should be installed automatically when you run `install.sh`. This is available in `master` branch and in release v4.4.
We still don't have an aarch64 toolchain for ESP32 ULP coprocessor, so it won't be installed when running `install.sh` on aarch64. Aside from that, installation on aarch64 using `install.sh` should succeed. If you run into any issue, please share some details about your OS and the IDF version you are using.

RepRapid, 2022-03-14:
not sure but I installed this last night, on a windows system 64 bit Intel, running in docker ubuntu, with the build instructions provided as of last night. 3/13/2022 10:pm I woke today, and just ran the test-rust-toolchain.sh and ran into this error, I followed the install instructions , including the dependency requirements..

```
`error: failed to run custom build command for `esp-idf-sys v0.30.6`

Caused by:
  process didn't exit successfully: `/root/rust-build/rust-esp32-std-demo/target/debug/build/esp-idf-sys-a70d87110bf5a3f6/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=MCU
  PATH=/root/rust-build/rust-esp32-std-demo/.embuild/espressif/esp-idf-v4.3.2/tools:$PATH

  --- stderr
  Error: Could not install esp-idf

  Caused by:
      command '"python3" "/root/rust-build/rust-esp32-std-demo/.embuild/espressif/esp-idf-v4.3.2/tools/idf_tools.py" "--idf-path" "/root/rust-build/rust-esp32-std-demo/.embuild/espressif/esp-idf-v4.3.2" "--quiet" "--non-interactive" "install-python-env"' exited with non-zero status code 1`
```

igrr, 2022-03-14:
@RepRapid indeed, the aarch64 related changes aren't present in release/v4.3. We could probably patch this in esp-idf-sys. Perhaps you can leave a comment in https://github.com/esp-rs/esp-idf-sys/issues/14?

Jigar0369, 2022-03-28:
> @RepRapid indeed, the aarch64 related changes aren't present in release/v4.3. We could probably patch this in esp-idf-sys. Perhaps you can leave a comment in [esp-rs/esp-idf-sys#14](https://github.com/esp-rs/esp-idf-sys/issues/14)?

@igrr @JurajSadel As it is confirmed now that v4.4 should support aarch64, but toolchain is still not supporting!
So Can you please help me by providing the steps to get ESP-IDF set up on Jetson Nano?
"It don't require to install toolchain but it's still supports aarch64" sounds bit confusing to me.

clowrey, 2022-08-02:
> We still don't have an aarch64 toolchain for ESP32 ULP coprocessor, so it won't be installed when running `install.sh` on aarch64

Is there a timeline for the ULP coprocessor support on aarch64? This is preventing me from compiling code on the rpi4 for ESPhome.. I know there are workarounds like compiling on a different machine but this particular program (ESPhome) requires re-compilation on any settings change for the devices so it is inconvenient.

popy2k14, 2022-09-09:
Any news on this?

Same issue on my odroid n2+

```
Processing midesklamp1s (board: esp32doit-devkit-v1; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
Tool Manager: Installing platformio/toolchain-esp32ulp @ ~1.22851.0
INFO Installing platformio/toolchain-esp32ulp @ ~1.22851.0
Error: Could not find the package with 'platformio/toolchain-esp32ulp @ ~1.22851.0' requirements for your system 'linux_aarch64'
```

igrr, 2022-09-09:
We have the new pre-release of binutils for the ULP ready [here](https://github.com/espressif/binutils-gdb/releases/tag/esp32ulp-elf-v2.35_20220830), but we need some time to integrate it into ESP-IDF. Unfortunately it's not a drop-in replacement as some of the arguments have changed. We'll update this ticket once the changes are merged in IDF!

popy2k14, 2022-09-09:
@igrr thx a lot.
just let us know an we can test it.

closed 2022-09-15 by espressif-bot

igrr, 2022-09-15:
ULP toolchain has been updated in the master branch in https://github.com/espressif/esp-idf/commit/a9bd454529e0e3a7b9207b079d8d511e1ca76a07 and now supports aarch64 Linux.
This will be backported to release/v4.4 branch as well.

popy2k14, 2022-09-16:
@igrr 
thx a lot!
Any hints when the home assistant get's this update?
Currently running "ESPHome Current version: 2022.8.3"

thx

popy2k14, 2022-09-29:
tried today with the newest esphome 9.2 release an got the same error:

```
INFO Reading configuration /config/esphome/schreibtisch.yaml...
WARNING GPIO2 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO12 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing midesklamp1s (board: esp32doit-devkit-v1; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
Tool Manager: Installing platformio/toolchain-esp32ulp @ ~1.22851.0
INFO Installing platformio/toolchain-esp32ulp @ ~1.22851.0
Error: Could not find the package with 'platformio/toolchain-esp32ulp @ ~1.22851.0' requirements for your system 'linux_aarch64'
```

Jason2866, 2022-09-29:
@popy2k14 Since there is no official release for IDF 4.4.3 all the new stuff is not yet in Platformio. I dont use esphome nor have i skills how to setup anything there.
If you manage to use my Platformio fork you can use the new patch5 toolchains and ULP v2.35. The setup is
```
platform = https://github.com/Jason2866/platform-espressif32.git#new_ULP
```

popy2k14, 2022-09-29:
ok, thx. thought that it was released already.
i can wait until it's released :-)

thx

mkruiver, 2022-10-09:
> ULP toolchain has been updated in the master branch in [a9bd454](https://github.com/espressif/esp-idf/commit/a9bd454529e0e3a7b9207b079d8d511e1ca76a07) and now supports aarch64 Linux. This will be backported to release/v4.4 branch as well.

Any timeline available when this 4.4 including ULP toolchain for ESP32 on Aarch64 would be available?

mkruiver, 2022-10-09:
> @mkruiver it is available in release/v4.4 branch now, since the commit [a8e81f8](https://github.com/espressif/esp-idf/commit/a8e81f88f1b46212df709b4d27c3d534bfeb7c27)

Thansk for the quick response!

However, it seems there has not been a 4.4 release since this commit, so effectively not available yet?

The latest 4.4 is from more than 28 days ago, it seems?
https://github.com/espressif/esp-idf/releases/tag/v4.4.2

mkruiver, 2022-10-09:
> Of you are using IDF directly and if you need this functionality you can
> either git checkout the latest release/v4.4 branch or cherry-pick that
> commit.
>
> If you are using IDF via other frameworks (Arduino, Platformio, etc) then
> you'll have to wait until the next 4.4.x bugfix release is published. (I
> can't give an estimate of when that will happen, but you can subscribe to
> IDF release notifications here on Github.)

Clear, thanks! (I am using IDF via another framework, so will wait for the
next 4.4.x release) :)

Thanks a lot for the extensive replies on this thread!

popy2k14, 2022-10-15:
Any hint's i can get the 4.4 release branch to build on my arm 64 odroid n2+ with HA esphome addon:

What i am currently using is an rather old package, but OTA isnt working:

```
esphome:
  name: midesklamp1s
  platformio_options:
    platform_packages:
      - framework-arduinoespressif32 @ https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.6
    platform:
      - espressif32@1.11.0
  
esp32:
  board: esp32doit-devkit-v1
```

i want to use the newest 4.4 to check if OTA works.
Sure i can flash with serial the first time.
Sadly i cant compile it because of the ulp toolchain issue of current releases.

I also have the info, that the package i previously used (solo-no-mac-crc branch) should have the folloiwing config:

```
esphome:
  name: midesklamp1s

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
    advanced:
      ignore_efuse_mac_crc: true
```
Any hints how i can add the current 4.4 release where the toolchain is fixed? to the last xaml?

thx

Jason2866, 2022-10-15:
@popy2k14 What do you need now? Solo1 (`CONFIG_FREERTOS_UNICORE: y`) support or not?
If solo1 is needed idk if this is working.
```
esphome:
  name: midesklamp1s
  platformio_options:
    platform:
      - https://github.com/tasmota/platform-espressif32/releases/download/v2.0.5.1/platform-espressif32-2.0.5.1.zip

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
    advanced:
      ignore_efuse_mac_crc: true
```

popy2k14, 2022-10-15:
Thanks for trying to help. Sadly doesnt work:

```
Platform Manager: espressif32@2.0.5 has been installed!
INFO espressif32@2.0.5 has been installed!
Platform Manager: Removing espressif32 @ 2.0.5
INFO Removing espressif32 @ 2.0.5
Platform Manager: espressif32@2.0.5 has been removed!
INFO espressif32@2.0.5 has been removed!
Error: Development platform 'espressif32' is not compatible with PlatformIO Core v6.0.2 and depends on PlatformIO Core >=6.1.0.
```

Jason2866, 2022-10-15:
You have to upgrade your platformio version to 6.1 or newer

popy2k14, 2022-10-15:
SAure i would, but esphome is an addon in an supervisor which i dont have control over.

Jason2866, 2022-10-15:
See my changes above

Jason2866, 2022-10-15:
Try this one. I changed the platformio version requirement to 6.0.2
```
esphome:
  name: midesklamp1s
  platformio_options:
    platform:
      - https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.5-esp/platform-espressif32-2.0.5-esp.zip

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
    advanced:
      ignore_efuse_mac_crc: true
```

popy2k14, 2022-10-15:
thx a lot, now we are step further.
it builds it but fails with:

```
Compiling /data/midesklamp1s/.pioenvs/midesklamp1s/src/esphome/core/scheduler.o
Compiling /data/midesklamp1s/.pioenvs/midesklamp1s/src/esphome/core/util.o
Compiling /data/midesklamp1s/.pioenvs/midesklamp1s/src/main.o
*** [/data/midesklamp1s/.pioenvs/midesklamp1s/memory.ld] Source `/data/cache/platformio/packages/framework-espidf/components/esp_system/ld/esp32/memory.ld.in' not found, needed by target `/data/midesklamp1s/.pioenvs/midesklamp1s/memory.ld'.
========================= [FAILED] Took 136.13 seconds =========================
```

Jason2866, 2022-10-15:
Delete hidden folder `.platformio` Looks like old stuff is there. If this does not work esphome does changes "somewhere" which are not compatible to plain origi. Platformio

popy2k14, 2022-10-15:
Tried to rename the device, this way it uses /data/midesklamp1ss/ directory instead of /data/midesklamp1s/, so tehre should be no garbage from old builds. Sadly same issue.
Also uninstalled esphome addon and reinstalled to clear all garbage -> same issue.

Sorry to be OT here.

Just one last OT question.
Can you please post the needed tasmota config for the "1S" lamp.

Jason2866, 2022-10-15:
https://templates.blakadder.com/mi_desk_lamp.html

popy2k14, 2022-10-15:
HUGE THX!
Was looking a while ago for tasmota support without luck.
It seems as of 09:2022 the 1S is supported!

thx a lot!

cvjensen, 2022-11-07:
Hi @Jason2866 , 

I just came from this issue:
https://github.com/esphome/issues/issues/3093

Can you help me with this error I'm getting on my Pi 4 ARM when trying to compile in esphome:

`/bin/sh: 1: /data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc: not found
`

Jason2866, 2022-11-07:
@cvjensen  The arm toolchains are not yet official supported in platformio.
You can try my fork which does support.j
To use with platformio this entry is needed. 
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.5.2/platform-espressif32-2.0.5.2.zip
```
Idk if it is possible to use with HA. I am NOT using HA nor Esphome.

clowrey, 2022-11-07:
> It keeps giving me the error:
> 
> ```
> Run Build Command(s):/data/cache/platformio/packages/tool-ninja/ninja cmTC_6d7a8 && [1/2] Building C object CMakeFiles/cmTC_6d7a8.dir/testCCompiler.c.obj
> FAILED: CMakeFiles/cmTC_6d7a8.dir/testCCompiler.c.obj
> /data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc   -mlongcalls -o CMakeFiles/cmTC_6d7a8.dir/testCCompiler.c.obj   -c testCCompiler.c
> /bin/sh: 1: /data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc: not found
> ninja: build stopped: subcommand failed.
> ```
> 
> I just tried to log in to the docker container running esphome and it seems like the file is there:
> 
> ```
> root@5c53de3b-esphome:/# ls -l /data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-g++
> -rwxr-xr-x 1 root root 778104 Feb 11  2022 /data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-g++
> root@5c53de3b-esphome:/# ls -l /data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc
> -rwxr-xr-x 1 root root 774008 Feb 11  2022 /data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc
> ```
> 
> Any idea who I can contact to get help?

No one, wait until it is supported or install ESPhome / HA on X86 machine.

cvjensen, 2022-11-07:
> No one, wait until it is supported or install ESPhome / HA on X86 machine.

When will it be supported ?