espressif/esp-idf #6433: re-entrancy of httpd_ws_recv_frame

oclyke, 2021-01-20:
Thanks for the addition of websocket support in the IDF! Overall it is making my job far easier. The problem I am facing is that I am forced to use a local buffer with a fixed size to receive the packet payload.

I would like to use ```httpd_ws_recv_frame``` (or another high level idf function) to measure the message length of the websocket packet *before* allocating memory to use for the payload, thus allowing me to handle any size packet. For example:

``` c
httpd_ws_frame_t ws_pkt = {0};

// determine the length of the message
httpd_ws_recv_frame(req, &ws_pkt, 0);
ESP_LOGI(TAG, "frame recvd with length: %d", ws_pkt.len);

// allocate enough memory for the data
void* data = malloc(ws_pkt.len);
ws_pkt.payload = (uint8_t*)data;

// receive the data
esp_err_t ret = httpd_ws_recv_frame(req, &ws_pkt, ws_pkt.len);
if(ret != ESP_OK){
  ESP_LOGE(TAG, "could not receive frame data - error (%d)", ret);
}

```

However from experience I have deduced that one cannot call ```httpd_ws_recv_frame``` more than once for a particular httpd request. When I do I always get the error ```ESP_ERR_INVALID_SIZE``` ("WS Message too long")

An example of a similar api is that of ```snprintf``` which, when the output buffer is ```NULL``` will return the number of characters that would have been necessary to store the output.

(p.s. if there is an existing API to get this information I would be happy to hear about it. I have tried searching the IDF docs but info concerning the ```httpd_ws_recv_frame``` function seems sparse)

yuanjianmin, 2021-02-02:
[0001-esp_http_server-support-dynamic-payload-len-for-ws-s.zip](https://github.com/espressif/esp-idf/files/5909296/0001-esp_http_server-support-dynamic-payload-len-for-ws-s.zip)
@oclyke hi oclyke， Thank you for your suggestions, requirements and interface call ideas. I offer a patch to satisfy your requirements. Could you please help me to check this patch whether could meet your functions. 
Thanks！
Wait for your reply.

NikLeberg, 2021-02-18:
Hi @yuanjianmin
I'm not the original issue opener but I'm using your patch for a few days now. It works flawlessly!
Having this feature is very key for the app I'm building.
Thanks for the patch!
Nik

oclyke, 2021-02-18:
Hi @yuanjianmin, thanks for writing that patch! I have just reviewed it and agree with NikLeberg that it is quite useful. Hoping that Espressif will be able to accept the changes so that I can rely on them in my application.

yuanjianmin, 2021-03-01:
@NikLeberg @oclyke hi guys, the patch has been merged and will be synchronized to Github later. I will notify you as soon as synchronized to github. Thanks.

closed 2021-03-12 by espressif-bot