espressif/esp-idf #6434: TWAI Transmit corrupts on a full 8 bytes extended message

teekaytk, 2021-01-20:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32E]
- IDF version (run ``git describe --tags`` to find it): v4.3-dev-472-gcf056a7d0
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     [Linux/Docker]
- Using an IDE?: [vscode as editor]
- Power Supply:         [Battery/custom]

## Problem Description
I have a slave device on the bus and 120 R resistor. The messages from the slave device are acknowledged by ESP32 perfectly.

I am trying to transmit an 8 byte extended CAN frame using the TWAI drivers.
I was constantly getting a frame with id 030 or 130  and 0 payload, instead of my intended frame.
I check all setting multiple time and they were set as per the TWAI documentation and examples.
Then I went back to basics and started with standard frame and 0 bytes payload and that worked.
Next I made it extended frame with a small ID and that worked. 
Next I made it extended frame with long id (32 bits) and kept increasing the payload.

At 2 bytes payload it was working but at 4 bytes payload it broke.

I believe this to be a driver issue, not a user error.

This is what worked:
[image]
and this
[image]

and this is when it broke when I made the payload 4 bytes
[image]

### Expected Behavior
To be able to transmit 8 bytes extended frame

### Actual Behavior
 4 bytes and larger payload is turned into a small standard frame message instead of the intended message.

### Steps to reproduce

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

 ```
twai_message_t message;
 uint32_t rolling_number = 0;
    uint8_t len = 4;
for(int a= 0; a< 100; a++) {
                    memset(message.data, 0, sizeof(message.data));
                    memset(printBuffer, 0, sizeof(printBuffer));
                    message.data[0] = 0u;
                    message.data[1] = 0u;
                    message.data[2] = 0u;
                    message.data[3] = 0u;
                    message.data[4] = (rolling_number >> 24) & 0xFF;
                    message.data[5] = (rolling_number >> 16) & 0xFF;
                    message.data[6] = (rolling_number >> 8) & 0xFF;
                    message.data[7] = rolling_number & 0xFF;

                    rolling_number++;
                    message.rtr = 0;
                    message.ss = 0;
                    message.extd = 1;
                    message.identifier = 0x18FDD135;
                    message.data_length_code = len;
                  
                                     //Queue message for transmission
                    ret = twai_transmit(&message, pdMS_TO_TICKS(1000));
                    if(ret == ESP_OK)
                    {
                        printf("CAN TX DONE\n");
                        device->can.can_tx_error = ESP_OK;
                    }
                    else
                    {
                        printf("CAN ERROR = %d\n", ret);
                        device->can.can_tx_error = ESP_FAIL;
                    }
}
```

teekaytk, 2021-03-12:
Any updates for me?

Thanks

Dazza0, 2021-03-23:
@teekaytk I cannot reproduce your issue. I ran the following code below:

```c
#include <stdio.h>
#include <stdint.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/twai.h"

void app_main(void)
{
    twai_general_config_t g_config = TWAI_GENERAL_CONFIG_DEFAULT(21, 22, TWAI_MODE_NO_ACK);
    twai_timing_config_t t_config = TWAI_TIMING_CONFIG_1MBITS();
    twai_filter_config_t f_config = TWAI_FILTER_CONFIG_ACCEPT_ALL();

    twai_driver_install(&g_config, &t_config, &f_config);
    twai_start();

    twai_message_t message;
    uint32_t rolling_number = 0;
    uint8_t len = 4;
    esp_err_t ret;
    printf("sizeof(message.data) %d\n", sizeof(message.data));
    for(int a= 0; a< 100; a++) {
        memset(message.data, 0, sizeof(message.data));
        message.data[0] = 0u;
        message.data[1] = 0u;
        message.data[2] = 0u;
        message.data[3] = 0u;
        message.data[4] = (rolling_number >> 24) & 0xFF;
        message.data[5] = (rolling_number >> 16) & 0xFF;
        message.data[6] = (rolling_number >> 8) & 0xFF;
        message.data[7] = rolling_number & 0xFF;

        rolling_number++;
        message.rtr = 0;
        message.ss = 0;
        message.extd = 1;
        message.identifier = 0x18FDD135;
        message.data_length_code = len;
        
        //Queue message for transmission
        ret = twai_transmit(&message, pdMS_TO_TICKS(1000));
        if(ret == ESP_OK) {
            printf("CAN TX DONE\n");
        } else {
            printf("CAN ERROR = %d\n", ret);
        }
    }
}
```
It seems like when you set you try to send a length > 4,  standard frame is being instead of an extended frame (you can see that for the buggy frames, the IDE bit is not set in the buggy frames). Is there any possibility that the `message.extd` bit is being wiped somehow (maybe due to a stack overflow or something)?

teekaytk, 2021-03-26:
Hi @Dazza0  Thanks for testing this
I will replicate your test here and report back. Can you confirm that you have used a ESP32-WROOM32E
Just want to be sure we are using the same silicon.
Thanks

teekaytk, 2021-03-27:
I ran your code on a cheap dev board I had in my possession 
"node MCU ESP-32S"

It runs like this

[image]

only transmits 4 bytes

I'll replace one of these dev boards with ESP32-WROOM32E but I think the result may be the same?

Thanks

Dazza0, 2021-04-07:
@teekaytk Just to be clear. In the example I posted above, you tried chaning `uint8_t len` to another value and the example was the message still only contained 4 bytes?

Dazza0, 2021-09-07:
@teekaytk We've had a recent influx of TWAI issues, most of them were caused by the same compiler bug described in issue https://github.com/espressif/esp-idf/issues/5604. The issue was resolved on master as of commit 2e0ffbd543abc0226f0dbd2f725eb891235abe02. Could you try checking out 2e0ffbd543abc0226f0dbd2f725eb891235abe02 or later and see if your issue is resolved.

brainstorm, 2021-09-15:
With a ESP32S2 "Saola" board I'm seeing the following with [your snippet](https://github.com/espressif/esp-idf/issues/6434#issuecomment-804531753) (@Dazza0)... my next step was to add a [level converter](https://learn.sparkfun.com/tutorials/retired---using-the-logic-level-converter) and a [TJA1042] I had lying around, but I'm not sure where those `CAN ERROR`s come from, shouldn't the snippet just keep sending `CAN TX DONE`s when nothing else is attached to the dev board?:

```
--- idf_monitor on /dev/cu.usbserial-11210 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x17dc
load:0x4004c000,len:0xa48
load:0x40050000,len:0x2e68
entry 0x4004c204
I (21) boot: ESP-IDF v4.4-dev-3042-g220590d599 2nd stage bootloader
I (21) boot: compile time 16:47:00
I (21) boot: chip revision: 0
I (25) boot.esp32s2: SPI Speed      : 80MHz
I (30) boot.esp32s2: SPI Mode       : DIO
I (35) boot.esp32s2: SPI Flash Size : 2MB
I (40) boot: Enabling RNG early entropy source...
I (45) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (63) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00100000
I (78) boot: End of partition table
I (83) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=06dech ( 28140) map
I (97) esp_image: segment 1: paddr=00016e14 vaddr=3ffbcb60 size=01c58h (  7256) load
I (101) esp_image: segment 2: paddr=00018a74 vaddr=40022000 size=075a4h ( 30116) load
I (115) esp_image: segment 3: paddr=00020020 vaddr=40080020 size=132dch ( 78556) map
I (132) esp_image: segment 4: paddr=00033304 vaddr=400295a4 size=035b0h ( 13744) load
I (136) esp_image: segment 5: paddr=000368bc vaddr=50000000 size=00010h (    16) load
I (143) boot: Loaded app from partition at offset 0x10000
I (144) boot: Disabling RNG early entropy source...
I (160) ca〞f�~���^��ѥ���cache 	: size 8KB, 4Ways, cache line size 32Byte
I (161) cpu_start: Pro cpu up.
I (172) cpu_start: Pro cpu start user code
I (172) cpu_start: cpu freq: 160000000
I (172) cpu_start: Application information:
I (177) cpu_start: Project name:     esp-can-simple
I (183) cpu_start: App version:      1
I (187) cpu_start: Compile time:     Sep 15 2021 16:46:54
I (193) cpu_start: ELF file SHA256:  ce18242f9915aa41...
I (199) cpu_start: ESP-IDF:          v4.4-dev-3042-g220590d599
I (206) heap_init: Initializing. RAM available for dynamic allocation:
I (213) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (219) heap_init: At 3FFBF018 len 0003CFE8 (243 KiB): DRAM
I (225) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (232) spi_flash: detected chip: generic
I (236) spi_flash: flash io: dio
W (240) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (254) cpu_start: Starting scheduler on PRO CPU.
sizeof(message.data) 8
CAN TX DONE
CAN TX DONE
CAN TX DONE
CAN TX DONE
CAN TX DONE
CAN TX DONE
CAN ERROR = 263
CAN ERROR = 263
CAN ERROR = 263
(...)
```

Dazza0, 2021-09-15:
@brainstorm 

- The log `CAN TX DONE` simply means that `twai_transmit()` has returned `ESP_OK`. This simply means that the message has been enqueued for transmission. It doesn't necessarily that the transfer has actually completed. Use the `TWAI_ALERT_TX_SUCCESS` alert instead for confirmation of transmission completion.
- The `CAN_ERROR` you're getting means that `twai_transmit()` is returning `ESP_ERR_TIMEOUT`, meaning that the function timed out waiting for space on the TX queue.
- This probably means you're enqueueing messages for transmission, but none of them are transmitting thus the TX queue is filling up.
- Documentation for `twai_transmit()` can be found [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/twai.html#_CPPv413twai_transmitPK14twai_message_t10TickType_t)

I don't think it is related to this issue. Please feel free to open another GitHub issue.

Dazza0, 2022-02-08:
Closing due to lack of updates. If https://github.com/espressif/esp-idf/issues/5604 or https://github.com/espressif/esp-idf/commit/2e0ffbd543abc0226f0dbd2f725eb891235abe02 does not resolve the issue, please feel free to reopen.

closed 2022-02-08 by Dazza0