espressif/esp-idf #6437: Modbus TCP/IP vMBTCPPortMasterGetCurrInfo() Error

ProKampas, 2021-01-20:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): v4.3-dev-2586-g526f682397
- Build System:         [|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB]

## Problem Description

I am creating a Modbus TCP/IP application where I need to use the slave addresses [1, 2, 200]. However, for the slave address
200 the device resets and gets a problem in `port_tcp_master.c` file in line `895` specifically in the 
`if (pxInfo->xSockId > -1) {` line. 

I tried to debug this and found out that the function `vMBTCPPortMasterGetCurrInfo()`  returns a wrong item from the
`MbSlaveInfo_t **` table.  That happens because my slave address is 200. Therefore, it tries to get the 
` xMbPortConfig.pxMbSlaveInfo[xMbPortConfig.ucCurSlaveIndex - 1];` But the ucCurSlaveIndex is 200. 

This needs to change into ` xMbPortConfig.pxMbSlaveInfo[xMbPortConfig.usMbSlaveInfoCount - 1];` 

Now, even though I don't get an error, I get the correct `pxInfo` struct in the code.
### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. Create a descriptor table with a slave address 200. 
2. Connect to a Modbus TCP/IP device with its IP
3. Setup the communication as stated in the example tutorial.
4. Attempt to use `mbc_master_get_parameter()` to get the data from that slave address.

### Code to reproduce this issue

## Debug Logs

```
Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.

Core  0 register dump:
PC      : 0x4000c3f0  PS      : 0x00060130  A0      : 0x80058ac0  A1      : 0x3ffe5850  
A2      : 0x3ffe5beb  A3      : 0x400902e0  A4      : 0x00000002  A5      : 0x3ffdf738  
0x400902e0: prvCopyItemByteBuf at /home/pro_kampas/esp/esp-idf/components/esp_ringbuf/ringbuf.c:466

A6      : 0x00000075  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x400902e0  
0x400902e0: prvCopyItemByteBuf at /home/pro_kampas/esp/esp-idf/components/esp_ringbuf/ringbuf.c:466

A10     : 0x00000020  A11     : 0x3ffdf784  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x3ffbc058  SAR     : 0x00000016  EXCCAUSE: 0x00000003  
EXCVADDR: 0x400902e0  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
0x400902e0: prvCopyItemByteBuf at /home/pro_kampas/esp/esp-idf/components/esp_ringbuf/ringbuf.c:466

Backtrace:0x4000c3ed:0x3ffe5850 0x40058abd:0x3ffe5870 0x401c4d07:0x3ffe58a0 0x401ca2ae:0x3ffe58d0 0x401ca403:0x3ffe5be0 0x401c74cc:0x3ffe5d00 0x401bd1ed:0x3ffe6010 0x4013ee2a:0x3ffe6060 0x4013f3f5:0x3ffe6090 0x4013f0cb:0x3ffe60b0 0x4013db12:0x3ffe60f0 0x400941a5:0x3ffe6120
0x401c4d07: __sprint_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:433
 (inlined by) __sprint_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:403

0x401ca2ae: _vfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1773 (discriminator 1)

0x401ca403: __sbprintf at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:480

0x401c74cc: _vfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:868 (discriminator 5)

0x401bd1ed: printf at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/printf.c:56 (discriminator 5)

0x4013ee2a: xMBMasterTCPPortSendResponse at /home/pro_kampas/esp/esp-idf/components/freemodbus/tcp_master/port/port_tcp_master.c:896

0x4013f3f5: eMBMasterTCPSend at /home/pro_kampas/esp/esp-idf/components/freemodbus/modbus/tcp/mbtcp_m.c:145

0x4013f0cb: eMBMasterPoll at /home/pro_kampas/esp/esp-idf/components/freemodbus/modbus/mb_m.c:337 (discriminator 15)

0x4013db12: modbus_tcp_master_task at /home/pro_kampas/esp/esp-idf/components/freemodbus/tcp_master/modbus_controller/mbc_tcp_master.c:60

0x400941a5: vPortTaskWrapper at /home/pro_kampas/esp/esp-idf/components/freertos/port/xtensa/port.c:168

```

## Other items if possible

alisitsyn, 2021-01-21:
Hi @ProKampas,

Thank you for your issue. 
I can confirm this. When Modbus TCP stack was designed I made some constraints in its design. **The short addresses of the slaves in the Dictionary for Modbus parameters** (device_parameters) should not have any gaps in between and short address of slave (mb_parameter_descriptor_t::mb_slave_addr) should correspond to its IP address in the table which is referenced by comm_info.ip_addr.  This was described in the documentation and example. On start of stack the mbc_tcp_master_start() registers all slave addresses in master configuration structure and then connects to each slave in list. It supposed that we create the modbus segment and are able to configure the slave addresses accordingly. Because of gap in the slave address the slave information for slave with address 200 is saved in the xMbPortConfig.pxMbSlaveInfo[2] but vMBTCPPortMasterGetCurrInfo() tries to get information from xMbPortConfig.pxMbSlaveInfo[200] which cases issue. I think it worth to fix this and register each slave as a pair (slave_short_add, ip_address) in the configuration. Unfortunately your fix above will not work correctly in all cases. As soon as it is ready I will provide patch for you.

ProKampas, 2021-01-21:
Hi @alisitsyn,

Thank you for your response.

alisitsyn, 2021-01-26:
Hi @ProKampas,

Please check the patch attached. This patch fixes your issue with short address with gaps in the data dictionary.

[0001_fix_modbus_crash_when_short_address_gaps_in_data_dictionary.diff](https://github.com/espressif/esp-idf/files/5873044/0001_fix_modbus_crash_when_short_address_gaps_in_data_dictionary.diff.log)

Please try to use CONFIG_MB_MDNS_IP_RESOLVER, CONFIG_MB_SLAVE_IP_FROM_STDIN options in the menuconfig and let me know if you have any issues.
Thanks.

alisitsyn, 2021-02-09:
Hi @ProKampas,

Could you provide update for your project? Does the patch work for you?

alisitsyn, 2021-04-01:
@ProKampas,

Could you provide some update for this issue?

ProKampas, 2021-04-01:
Hi @alisitsyn,

I couldn't get the patch to work. I am using the solution I explained in the description and it's working fine.

alisitsyn, 2021-04-01:
Hi @ProKampas,
It is strange that the patch does not work on your side. Could you share the log for the application with applied patch?
I need to find a reason for your issues.

Thanks.

ProKampas, 2021-04-09:
I tried to apply the patch with `git apply` but the code was not building correctly. That was my problem.

alisitsyn, 2021-04-10:
Hi @ProKampas,
Could you answer to my questions below:
1. What is the version of IDF do you try to apply the patch for.
2. Do you see the errors when you apply the patch? 
3. Could you send the compilation log?

Please let me help you to solve this issue. The patch should work just fine, if not I can help you to fix.
Thank you.

ProKampas, 2021-04-20:
Hi @alisitsyn ,

I tested your patch. Sorry for the long wait. 
1. I used the latest release/4.3 for your patch.
2. I managed to apply the patch successfully.
3. The same error exists. I want to use slave addresses of 200, 1000, 1100 and this software doesn't allow that. This is my problem. I want to use the Modbus RTU over TCP/IP and that's why I had to make the modifications I explained in this issue and the other one I opened a while ago. 
The log from my attempt to start a communication is the following:

```
I (36822) MB_TCP_MASTER_PORT: TCP master stack initialized.
I (36822) MB_TCP_MASTER_PORT: Add slave IP: 192.168.1.2
I (36822) MB_TCP_MASTER_PORT: Connecting to slaves...
-I (36852) MB_TCP_MASTER_PORT: Connected 1 slaves, start polling...
I (36852) ALF: Modbus comm start ok
E (36932) MB_TCP_MASTER_PORT: Slave info for short address 200 not found.
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4013df6b  PS      : 0x00060930  A0      : 0x8013e4f8  A1      : 0x3fffbe70  
0x4013df6b: xMBMasterTCPPortSendResponse at /home/pro_kampas/esp/esp-idf/components/freemodbus/tcp_master/port/port_tcp_master.c:923

A2      : 0x3ffcca8e  A3      : 0x0000000c  A4      : 0x00000000  A5      : 0x00000010  
A6      : 0x3ffb5e40  A7      : 0x3ffb5dfc  A8      : 0x8013df68  A9      : 0x3fffbe50  
A10     : 0x00000000  A11     : 0x3ffc8d2c  A12     : 0x00000000  A13     : 0x3ffc0498  
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000012  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

Backtrace:0x4013df68:0x3fffbe70 0x4013e4f5:0x3fffbea0 0x4013e1c3:0x3fffbec0 0x4013cbe2:0x3fffbf00 0x4009583d:0x3fffbf30
0x4013df68: xMBMasterTCPPortSendResponse at /home/pro_kampas/esp/esp-idf/components/freemodbus/tcp_master/port/port_tcp_master.c:920

0x4013e4f5: eMBMasterTCPSend at /home/pro_kampas/esp/esp-idf/components/freemodbus/modbus/tcp/mbtcp_m.c:145

0x4013e1c3: eMBMasterPoll at /home/pro_kampas/esp/esp-idf/components/freemodbus/modbus/mb_m.c:337 (discriminator 15)

0x4013cbe2: modbus_tcp_master_task at /home/pro_kampas/esp/esp-idf/components/freemodbus/tcp_master/modbus_controller/mbc_tcp_master.c:113

0x4009583d: vPortTaskWrapper at /home/pro_kampas/esp/esp-idf/components/freertos/port/xtensa/port.c:168

```

alisitsyn, 2021-04-27:
@ProKampas ,

I can not check what is wrong on your side. As per log you did not configure the master and slave appropriately. The log says that master could find just one slave and I do not see the required log lines to control the resolving of the data dictionary table. Then master starts connection to one slave and trying to find slave with index 200. I can not control how correct the changes were applied to your project but can see that your code is incorrect and do not function properly. According to this please check the project here. The project requires configured and flashed TCP slaves with index 1 and index 200 in data dictionary (can be easily changed to any address). The project in fact is able to work. Please compile the project and check it. However we did not discuss about slave index > 255, but it is not a big issue. The log is below:
```
I (2587) esp_netif_handlers: example_connect: sta ip: 192.168.0.254, mask: 255.255.0.0, gw: 192.168.6.1
I (2587) example_connect: Got IPv4 event: Interface "example_connect: sta" address: 192.168.0.254
I (2597) example_connect: Connected to example_connect: sta
I (2597) example_connect: - IPv4 address: 192.168.0.254
I (2607) wifi:Set ps type: 0

I (2607) MASTER_TEST: Query PTR: _modbus._tcp.local
  PTR : mb_slave_tcp_01
  TXT : [3] mb_id=33221100; mac=A8032AEC08C8; board=esp32;
I (5687) MASTER_TEST: Resolved slave mb_slave_tcp_01[192.168.0.252]:502
I (5697) MASTER_TEST: Index: 0, sl_addr: 1, resolved to IP: [192.168.0.252]
I (5697) MASTER_TEST: Index: 1, sl_addr: 1, set to IP: [192.168.0.252]
I (5707) MASTER_TEST: Index: 2, sl_addr: 1, set to IP: [192.168.0.252]
  PTR : mb_slave_tcp_C8
  TXT : [3] mb_id=33221100; mac=A8032AEC0DFC; board=esp32;
I (5727) MASTER_TEST: Resolved slave mb_slave_tcp_C8[192.168.0.253]:502
I (5727) MASTER_TEST: Index: 3, sl_addr: 200, resolved to IP: [192.168.0.253]
I (5737) MASTER_TEST: Index: 4, sl_addr: 200, set to IP: [192.168.0.253]
I (5747) MASTER_TEST: Index: 5, sl_addr: 1, set to IP: [192.168.0.252]
I (5757) MASTER_TEST: Index: 6, sl_addr: 1, set to IP: [192.168.0.252]
I (5757) MASTER_TEST: Index: 7, sl_addr: 1, set to IP: [192.168.0.252]
I (5767) MASTER_TEST: Resolved 8 cids, with 2 IP addresses
I (5777) MASTER_TEST: Modbus master stack initialized...
I (5777) MB_TCP_MASTER_PORT: TCP master stack initialized.
I (5787) MB_TCP_MASTER_PORT: Add slave IP: 192.168.0.253
I (5797) MB_TCP_MASTER_PORT: Add slave IP: 192.168.0.252
I (5797) MB_TCP_MASTER_PORT: Connecting to slaves...
-W (5807) wifi:<ba-add>idx:0 (ifx:0, 64:d1:54:1a:23:5b), tid:0, ssn:5, winSize:64
-..--I (5867) MB_TCP_MASTER_PORT: Connected 2 slaves, start polling...
I (6017) MASTER_TEST: Start modbus test...
I (6077) MASTER_TEST: Characteristic #0 Data_channel_0 (Volts) value = 1.120000 (0x3f8f5c29) read successful.
I (6137) MASTER_TEST: Characteristic #1 Humidity_1 (%rH) value = 1.340000 (0x3fab851f) read successful.
I (6197) MASTER_TEST: Characteristic #2 Temperature_1 (C) value = 2.340000 (0x4015c28f) read successful.
I (6267) MASTER_TEST: Characteristic #3 Humidity_2 (%rH) value = 2.560000 (0x4023d70a) read successful.
I (6327) MASTER_TEST: Characteristic #4 Temperature_2 (C) value = 3.560000 (0x4063d70a) read successful.
I (6387) MASTER_TEST: Characteristic #5 Humidity_3 (%rH) value = 3.780000 (0x4071eb85) read successful.
I (6447) MASTER_TEST: Characteristic #6 RelayP1 (on/off) value = OFF (0x55) read successful.
I (6517) MASTER_TEST: Characteristic #7 RelayP2 (on/off) value = OFF (0xaa) read successful.
I (7077) MASTER_TEST: Characteristic #0 Data_channel_0 (Volts) value = 1.120000 (0x3f8f5c29) read successful.
I (7137) MASTER_TEST: Characteristic #1 Humidity_1 (%rH) value = 2.440000 (0x401c28f6) read successful.
I (7197) MASTER_TEST: Characteristic #2 Temperature_1 (C) value = 2.340000 (0x4015c28f) read successful.
I (7257) MASTER_TEST: Characteristic #3 Humidity_2 (%rH) value = 2.560000 (0x4023d70a) read successful.
I (7317) MASTER_TEST: Characteristic #4 Temperature_2 (C) value = 3.560000 (0x4063d70a) read successful.
I (7387) MASTER_TEST: Characteristic #5 Humidity_3 (%rH) value = 3.780000 (0x4071eb85) read successful.
I (7447) MASTER_TEST: Characteristic #6 RelayP1 (on/off) value = OFF (0x55) read successful.
I (7507) MASTER_TEST: Characteristic #7 RelayP2 (on/off) value = OFF (0xaa) read successful.
I (8067) MASTER_TEST: Characteristic #0 Data_channel_0 (Volts) value = 1.120000 (0x3f8f5c29) read successful.
I (8127) MASTER_TEST: Characteristic #1 Humidity_1 (%rH) value = 3.540000 (0x40628f5c) read successful.
I (8187) MASTER_TEST: Characteristic #2 Temperature_1 (C) value = 2.340000 (0x4015c28f) read successful.
I (8247) MASTER_TEST: Characteristic #3 Humidity_2 (%rH) value = 2.560000 (0x4023d70a) read successful.
I (8317) MASTER_TEST: Characteristic #4 Temperature_2 (C) value = 3.560000 (0x4063d70a) read successful.
I (8377) MASTER_TEST: Characteristic #5 Humidity_3 (%rH) value = 3.780000 (0x4071eb85) read successful.
I (8437) MASTER_TEST: Characteristic #6 RelayP1 (on/off) value = OFF (0x55) read successful.
I (8497) MASTER_TEST: Characteristic #7 RelayP2 (on/off) value = OFF (0xaa) read successful.
I (9057) MASTER_TEST: Characteristic #0 Data_channel_0 (Volts) value = 1.120000 (0x3f8f5c29) read successful.
I (9117) MASTER_TEST: Characteristic #1 Humidity_1 (%rH) value = 4.640000 (0x40947ae1) read successful.
I (9177) MASTER_TEST: Characteristic #2 Temperature_1 (C) value = 2.340000 (0x4015c28f) read successful.
[... cut ...]
I (12437) MASTER_TEST: Characteristic #7 RelayP2 (on/off) value = ON (0xff) read successful.
I (12937) MASTER_TEST: Alarm triggered by cid #7.
I (12937) MASTER_TEST: Destroy master...
I (13937) MASTER_TEST: Modbus master stack destroy...
```
Please find the project [Modbus TCP master slave projects with address gaps](https://github.com/alisitsyn/modbus_support/tree/mb_support/mb_m_s_support_address_gaps/mb_m_s_address_gaps). The project contains Modbus master and slave TCP examples).
The project is self-sufficient and overrides the existing Modbus component folder to patched one.

ProKampas, 2021-05-06:
Yes, you are right. This project is working fine.

alisitsyn, 2022-01-21:
Sorry, for the delay. This functionality has been merged into master with commit ID 2cad565781eb95ffbbe0840f801016acaca82e86 .
It needs some time to sync with github.

Alvin1Zhang, 2022-03-08:
Thanks for reporting, fix on master is available at https://github.com/espressif/esp-idf/commit/2cad565781eb95ffbbe0840f801016acaca82e86 and fix on release/4.3 is available at https://github.com/espressif/esp-idf/commit/5c78283f6eb9deea0b371851e318ea5f00675f14, feel free to reopen if the issue still happens, thanks.

closed 2022-03-08 by Alvin1Zhang

pr200sd, 2022-04-06:
> Please find the project [Modbus TCP master slave projects with address gaps](https://github.com/alisitsyn/modbus_support/tree/mb_support/mb_m_s_support_address_gaps/mb_m_s_address_gaps). The project [...]

Good afternoon. I need your help, I downloaded an updated library from your link, where is the issue with the answer for Modbus TCP resolved? the UNIT ID has not been transmitted before. The project worked successfully, but since I need constant polling via Modbus TCP, I excluded the possibility of stopping the driver from the project, and noticed that after some time from 2 to 10 hours, work fails until the esp is completely rebooted. I am attaching the logs from the exchange, I checked it on the simplest project from the "\examples\protocols\modbus\tcp\mb_tcp_master" folder, then I transferred the code to my more complex project, the behavior has not changed
[mb_tcp_master.log](https://github.com/espressif/esp-idf/files/8424439/mb_tcp_master.log)

I also saw errors in the logs MB_CONTROLLER_MASTER: mbc_master_send_request(94): Master send request failure error=(0x107) (ESP_ERR_TIMEOUT).
I increased the tcp timeout to 2 seconds, errors began to appear less often and the reboot time increased.
The last option worked for almost 10 hours and still the module rebooted, since there is a lot of functionality in my project, I also reduced the CONFIG_FMB_TCP_PORT_MAX_CONN parameter to 3
Errors associated with the socket disappeared and the program worked for ~ 10 hours, in difficult conditions, writing 64 registers with a period of 200 ms, Modbus Slave is used as a slave, I also checked it on a real device.

alisitsyn, 2022-04-06:
Hi @pr200sd,

The project is obsolete. The updated feature of Modbus TCP master has been merged to the master of ESP-IDF. Please try to compile your project using the latest master. Also, the `freemodbus` component in ESP-IDF will be removed.  
I would suggest you try using the separate [esp-modbus](https://components.espressif.com/component/espressif/esp-modbus) instead.  Here is the [repo](https://github.com/espressif/esp-modbus) for it. See the instructions on how to start using this component instead of `freemodbus` the line ```set(EXCLUDE_COMPONENTS freemodbus)``` needs to be added to your project `CMakeLists.txt` file. If you still see the errors please open the separate issue [here](https://github.com/espressif/esp-modbus/issues). 

Thank you.

pr200sd, 2022-04-06:
> Do you mean that the slave does not respond with its slave address in the UNITID field of MBAP?

Yes, the library from the attachment "mb_m_s_address_gaps solved" this problem.

alisitsyn, 2022-04-06:
Yes, the tcp slave does not set the UID to its short address (the slave currently does not support this). The UID field will is just 0 (does not support RTU over TCP now). The same processing is in the [my library](https://github.com/alisitsyn/modbus_support/blob/mb_support/mb_m_s_support_address_gaps/mb_m_s_address_gaps/mb_component/tcp_slave/port/port_tcp_slave.c#L712)
https://github.com/espressif/esp-idf/blob/master/components/freemodbus/tcp_slave/port/port_tcp_slave.c#L717

pr200sd, 2022-04-06:
Yes thanks, with UID everything is clear.

I tested with the new version, example
"\esp-modbus-master\test\tcp\mb_tcp_master" as with previous versions,
 at first everything goes well, but after a while errors appear and
the connection does not continue,
 in this case I did not wait for a reboot .

alisitsyn, 2022-04-07:
@pr200sd,

Just to confirm: did you try the esp-modbus library and original tcp_master example?
If yes, please provide the complete log again to confirm the issue. I need the log of output from `idf.py flash monitor command` which does backtracing of function calls in the output. For better result please send me complete project of your example including sdkconfig, and build folder with *.map and *.elf files. 
Thanks.

pr200sd, 2022-04-07:
Yes, I used the original project, leaving only 2 registers and 1 IP address.
The only moment, I could not add esp-modbus through the component manager,
there were errors in the logs, I placed the folder in components and
from there I launched the project for assembly.
I attached the project with the files, but I did not save the log of
the idf.py flash monitor command,
 now I tried to run it from eclipse and the command line,
 but in both options I could not enter the command "IP0: 192.168.31.153;"
I work through the Terminal program, everything is entered in it,
the log file attached in the last letter is from this program.
The log monitor.txt file contains information when loading the project.

alisitsyn, 2022-04-11:
Hi,

Sorry for the delay, 
> now I tried to run it from eclipse and the command line,
 but in both options I could not enter the command "IP0: 192.168.31.153;"
 
 > IP to the code, before that it rebooted with an error,
 
 The error happens because you entered the IP address manually in wrong format. The correct format to define the IP addresses manually in the terminal is: "IP0=192.168.31.153".
 
The project I sent you already has the dependency with the component manager. Should be able to be compiled on ESP-IDF v5.0. Please check if you have the component manager installed in your environment and your ESP_IDF version.

> I attached the project with the files, but I did not save the log of
the idf.py flash monitor command,

Unfortunately do not see the project. The project with your binaries (build folder) needed to backtrace the function calls before crash. Please reproduce the issue and attach the build folder. Also you can use [core dump](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/core_dump.html) functionality to save backtrace. This also will allow to save backtrace of functions on the moment of crash or WDT triggering. Just use the command the `idf.py menuconfig` then select `Print core dump to UART (UART)`. Refer to the documentation for core dump above.

Thanks.

pr200sd, 2022-04-12:
Thanks, I'll double-check with the IP address format and restart the test project again, I have ESP-IDF version 4.4, maybe this is the problem. Now I'm checking the work on the latest version of Modbus TCP master that you attached, I just copied it and registered it instead of freemodbus, I use it in my big project, failures still happen, 12-18 hours can work, then crashes and reboot.

alisitsyn, 2022-04-12:
Thank you for the information. 
Note: The approach is to keep only necessary changes in the project (Data Dictionary). Try to reproduce the issues without the changes related to your big project for now. Once it happens please keep the log and project `build` folder then please send it to me. I also strongly recommend setting up a coredump, it will help us to find the reason for the issue. It can be very useful for your project later as well.

alisitsyn, 2022-04-14:
> also attached a log from the monitor, but I can’t enter the IP in it,

I don't know why the regular `idf.py flash monitor` do not work for you. You can easily enter the IP address in notation: `IP0=192.68.1.2` + Enter key as example. Other variant is to use predefined IP as you did before or using the `CONFIG_MB_MDNS_IP_RESOLVER` to resolve slave IP automatically if it is ESP32 slave.

Thank you for artifacts. Still need your core_dump UART output file in BASE64 format once the issue is triggered.

pr200sd, 2022-04-14:
[20220413093143.log](https://github.com/espressif/esp-idf/files/8488568/20220413093143.log)
Everything worked before, but I practically didn’t use it, I called it from the eclipse program, I’ll check it from the command line, I think it will work there. Yesterday I tested all day, the reboot did not happen, there are only WD logs, if this can be of any help.

pr200sd, 2022-04-15:
[20220414172430.log](https://github.com/espressif/esp-idf/files/8495160/20220414172430.log)
Can't get esp to restart, but wd works, this is on the project I attached above.
There are also many messages about ESP_ERR_INVALID_RESPONSE as a slave computer with a Modbus slave.

alisitsyn, 2022-04-20:
@pr200sd, Please take a look at the newly created issue above.