espressif/esp-idf #6439: v4.2 new SPI feature

KaeLL, 2021-01-20:
[v4.2](https://github.com/espressif/esp-idf/releases/tag/v4.2) release note states:
`SPI: Supported sharing SPI bus among drivers`

What does that mean?
Consider the lack of an explanation, either through a link to a ticket or something in the docs, the bug report.

ginkgm, 2021-01-21:
It means that, you can use other drivers to control SPI1 if you wish, with the arbitration provided by the SPI bus lock.

Currently, there are two drivers can control SPI1 peripheral: esp_flash (new SPI Flash API) and SPI Master.

Please check the docs here

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html#notes-on-using-the-spi-master-driver-on-spi1-bus

KaeLL, 2021-01-25:
@ginkgm Just to confirm, the arbitration provided by spi_bus_lock is only for SPI1?

closed 2022-06-06 by KaeLL