espressif/esp-idf #6440: PVS-Studio: 71 issues

AndreyKarpov, 2021-01-20:
One of our readers recommended paying heed to the Espressif IoT Development Framework. He found an error in the project code and asked if the PVS-Studio static analyzer could find it. The analyzer can't detect this specific error so far, but it managed to spot many others: https://www.viva64.com/en/b/0790/

megazhuk, 2021-01-23:
How to get more attention to this issues?
I think this is important, since developers can fix 71 errors at once in 1 day.

projectgus, 2021-02-01:
Hi @AndreyKarpov,

Thanks for opening the issue to let us know about your analysis work and blog post. We're in the process of analyzing your results and dividing the tasks up between the various subsystems that are affected. This issue will be updated as soon as we start applying fixes.

In a few cases it seems like the analyser may have returned some false positives. For example, in the case of the static function `_mdns_dup_interface` "array index out of bounds", the caller has [always pre-checked the result of _mdns_get_other_if() before calling _mdns_dup_interface()](https://github.com/espressif/esp-idf/blob/master/components/mdns/mdns.c#L2237)).

In other cases your analysis has clearly found bugs, so thank you again for bringing them to our attention.

zikalino, 2022-03-08:
For some reason this issue was forgotten and all the problems should be reviewed properly.

igrr, 2022-03-18:
wpa_supplicant fixes applied in https://github.com/espressif/esp-idf/commit/6da52cf0d894a48450a92b13cde672f441c6023f (forgot to add a link back to this issue)

igrr, 2022-03-22:
The final fixes for driver, essl and hal have been merged and will be synced to Github soon.
Thanks again for finding and explaining these issues in detail!