espressif/esp-idf #6441: esp32ulp-elf, esp32s2ulp-elf, and openocd-esp32 tools not found

briantanabe, 2021-01-21:
----------------------------- Delete below -----------------------------

- [ √ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ √ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ √ ] I have searched the issue tracker for a similar issue and not found a similar issue.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROVER]
- IDF version (run ``git describe --tags`` to find it):
    // v4.2
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [Battery]

## Problem Description

I'm trying to install the idf on a Raspberry Pi 4 running Ubuntu 20.10 32bit for armhf. I've made it almost through the entire install process; however, running export.sh returns the following:
```
Adding ESP-IDF tools to PATH...
WARNING: directory for tool esp32ulp-elf version 2.28.51-esp-20191205 is present, but tool was not found
ERROR: tool esp32ulp-elf has no installed versions. Please run '/home/ubuntu/esp/esp-idf/install.sh' to install it.
WARNING: directory for tool esp32s2ulp-elf version 2.28.51-esp-20191205 is present, but tool was not found
ERROR: tool esp32s2ulp-elf has no installed versions. Please run '/home/ubuntu/esp/esp-idf/install.sh' to install it.
WARNING: directory for tool openocd-esp32 version v0.10.0-esp32-20200709 is present, but tool was not found
ERROR: tool openocd-esp32 has no installed versions. Please run '/home/ubuntu/esp/esp-idf/install.sh' to install it.
```

I've tried running install.sh repeatedly, and it runs without errors, but shows the following warnings:
```
WARNING: directory for tool esp32ulp-elf version 2.28.51-esp-20191205 is present, but tool was not found
Installing esp32ulp-elf@2.28.51-esp-20191205
file binutils-esp32ulp-linux-armel-2.28.51-esp-20191205.tar.gz is already downloaded
WARNING: destination path already exists, removing
Extracting /home/ubuntu/.espressif/dist/binutils-esp32ulp-linux-armel-2.28.51-esp-20191205.tar.gz to /home/ubuntu/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205
WARNING: directory for tool esp32s2ulp-elf version 2.28.51-esp-20191205 is present, but tool was not found
Installing esp32s2ulp-elf@2.28.51-esp-20191205
file binutils-esp32s2ulp-linux-armel-2.28.51-esp-20191205.tar.gz is already downloaded
WARNING: destination path already exists, removing
Extracting /home/ubuntu/.espressif/dist/binutils-esp32s2ulp-linux-armel-2.28.51-esp-20191205.tar.gz to /home/ubuntu/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205
WARNING: directory for tool openocd-esp32 version v0.10.0-esp32-20200709 is present, but tool was not found
Installing openocd-esp32@v0.10.0-esp32-20200709
file openocd-esp32-armel-0.10.0-esp32-20200709.tar.gz is already downloaded
WARNING: destination path already exists, removing
Extracting /home/ubuntu/.espressif/dist/openocd-esp32-armel-0.10.0-esp32-20200709.tar.gz to /home/ubuntu/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709
```

### Expected Behavior
It should not show the errors, and running install.sh should fix it?
### Actual Behavior
The export.sh script doesn't complete.
### Steps to reproduce

A brief list of commands I ran after installing Ubuntu and running sudo apt update && sudo apt upgrade

```
sudo update-alternatives --install /usr/bin/python python /usr/bin/python3 10
sudo apt-get install flex bison gperf gettext python3-setuptools cmake ninja-build ccache libssl-dev automake texinfo help2man libtool libtool-bin libbz2-dev libreadline-dev libsqlite3-dev llvm libncurses5-dev libncursesw5-dev tk-dev liblzma-dev
sudo apt-get install python3-pip
pip install --upgrade pip
mkdir ~/esp
cd ~/esp
git clone -b v4.2 --recursive https://github.com/espressif/esp-idf.git
cd ~/esp/esp-idf
./install.sh
echo 'export PATH="$PATH:$HOME/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin"' >> ~/.profile
echo '. $HOME/esp/esp-idf/export.sh' >> ~/.profile
source ~/.profile
```

dobairoland, 2021-01-21:
Hi @briantanabe. Please try to run the following commands and shere the output.

```bash
~/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin/esp32ulp-elf-as --version
~/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin/esp32s2ulp-elf-as --version
~/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin/openocd --version
```

briantanabe, 2021-01-21:
@dobairoland After posting this I started parsing through the Python, and explicitly added the tools to my path. Unfortunately:
```
ubuntu@bmo:~$ esp32ulp-elf-as
-bash: /home/ubuntu/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin/esp32ulp-elf-as: No such file or directory
ubuntu@bmo:~$ ~/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin/esp32ulp-elf-as --version
-bash: /home/ubuntu/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin/esp32ulp-elf-as: No such file or directory
ubuntu@bmo:~$ ~/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin/esp32s2ulp-elf-as --version
-bash: /home/ubuntu/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin/esp32s2ulp-elf-as: No such file or directory
ubuntu@bmo:~$ ~/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin/openocd --version
-bash: /home/ubuntu/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin/openocd: No such file or directory
```

I'm a bit out of my depth here, but I thought this might be relevant:
```
ubuntu@bmo:~$ file /sbin/init
/sbin/init: symbolic link to /lib/systemd/systemd
ubuntu@bmo:~$ file /lib/systemd/systemd
/lib/systemd/systemd: ELF 32-bit LSB shared object, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, BuildID[sha1]=30c275af2eab86ceb3dbf01bfa7975b30561dff2, for GNU/Linux 3.2.0, stripped
ubuntu@bmo:~$ file ~/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin/openocd
/home/ubuntu/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin/openocd: ELF 32-bit LSB shared object, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.3, for GNU/Linux 3.2.0, BuildID[sha1]=008620cc31894d6b8d34d5f678669b8dd4039112, stripped
```

dobairoland, 2021-01-22:
@briantanabe Do you have `libpython2.7` as well?

It is weird. I thought that running them with `--version` would print the missing libraries. Instead they behave as they were built for a completely different CPU architecture.

0xFEEDC0DE64, 2021-02-03:
I can confirm I have the same issues on a ODROID XU3, I'm missing the /lib/ld-linux.so.3 and don't know where to get them

dobairoland, 2021-02-03:
Which linux distribution and version are you using @0xFEEDC0DE64?

briantanabe, 2021-02-03:
@dobairoland I have libpython2.7 installed. 

I'm definitely in over my head so I'm not sure if this is useful, but I also discovered libusb-1.0.so.0 exists in /lib/arm-linux-gnueabihf and which is included in /etc/ld.so.conf.d/arm-linux-gnueabihf.conf.

briantanabe, 2021-02-03:
As a final update – my original goal was to get the ESP-IDF toolchain working on the Raspberry Pi. I followed a Youtube tutorial suggesting we use Ubuntu 20.10 32bit for armhf, but I now learned doing this on Raspbian works like a charm. I've wiped my old install so I won't be able to follow up on the original issue, but thank you for all the help!

dobairoland, 2021-02-04:
@briantanabe I'm happy that you resolved the issue. I'm sorry, if I knew that another distribution is acceptable to you then I would suggest to use Raspbian. We also use it for testing purposes.

Can this issue be closed now?

briantanabe, 2021-02-04:
Yep.

closed 2021-02-04 by briantanabe