espressif/esp-idf #6443: Smartconfig failed to send ACK message errno 0

djokba, 2021-01-21:
## Environment

- Development Kit: none
- Module or chip used: ESP32-D0WD-V3
- IDF version: v4.2-dirty
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Windows
- Environment type: PowerShell
- Using an IDE: No
- Power Supply: external 3.3V
- EspTouch app version: v1.1.1

## Problem Description

I have tried smartconfig example code using 3 different APs (TL-MR3420, TL-WR840N, and an unknown router model). The error can happen on any AP. AFAIK, rebooting the router will solve the issue but it will reoccur later without making any changes to the router.

As a side note, i've modified esp-idf source file `smartconfig_ack.c` and `Kconfig` so i can  add custom data to smartconfig ACK message, but the error can still occur even when i run unmodified files. So far, i haven't encountered this issue in esp-idf v4.1.

### Expected Behavior

Smartconfig successfully send the ACK message to the phone that broadcasting the AP info once it connected to the AP.

### Actual Behavior

Smartconfig failed to send ACK message and return errno 0. App will fail after timeout.

### Steps to reproduce

1. Build and flash smart_config example code

### Code to reproduce this issue

smart_config example code
[modified smartconfig_ack.c](https://gist.github.com/djokba/6f6072a4816ae8fe313ede0490b79183)
[modified Kconfig](https://gist.github.com/djokba/5d4cab86da9bcdc53b88d84ff649d25a)

## Debug Logs

```
I (709) wifi:mode : sta (7c:9e:bd:f8:92:60)
I (769) smartconfig: SC version: V2.9.0
I (4869) wifi:ic_enable_sniffer
I (4869) smartconfig: Start to find channel...
I (4869) smartconfig_example: Scan done
I (32199) smartconfig: TYPE: ESPTOUCH
I (32199) smartconfig: T|PHONE MAC:9e:f2:4b:a0:19:17
I (32199) smartconfig: T|AP MAC:98:da:c4:cb:96:18
I (32199) smartconfig: Found channel on 4-1. Start to get ssid and password...
I (32209) smartconfig_example: Found channel
I (53489) smartconfig: T|pswd: qciot01@2019
I (53489) smartconfig: T|ssid: QC-IoT
I (53489) smartconfig: T|bssid: 98:da:c4:cb:96:18
I (53489) wifi:ic_disable_sniffer
I (53499) smartconfig_example: Got SSID and password
I (53499) smartconfig_example: SSID:QC-IoT
I (53509) smartconfig_example: PASSWORD:qciot01@2019
I (53659) wifi:new:<4,1>, old:<4,1>, ap:<255,255>, sta:<4,1>, prof:1
I (53659) wifi:state: init -> auth (b0)
I (53659) wifi:state: auth -> assoc (0)
I (53669) wifi:state: assoc -> run (10)
I (53809) wifi:security: WPA2-PSK, phy: bgn, rssi: -70
I (53819) wifi:pm start, type: 1

I (53849) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (56009) esp_netif_handlers: sta ip: 192.168.0.117, mask: 255.255.255.0, gw: 192.168.0.1
I (56009) smartconfig_example: WiFi Connected to ap
E (56439) smartconfig: send failed, errno 0
```

YouDONG-ESP, 2021-01-22:
@djokba Confirm it can be reproduce in some routers. It's seems like a problem that router won't transfer ARP request sended by station.

YouDONG-ESP, 2021-01-25:
@djokba please try to comment out `goto_end` right after `ESP_LOGE(TAG, "send failed, errno %d", err);
`
This error log should show multiple times and then the phone will recieve response after ~5 seconds.

The problem is some router won't resply ARP that soon, so the packet sent by esp32 will be cached to a queque till arp reply. This queue only have 4 place, if the queue is full, it will return -1 and cause send failed. All we had to do is ignore this error return and wait the arp reply.

YouDONG-ESP, 2021-01-25:
I think it should solve your problem, we will give a more complete repair patch soon

closed 2021-02-10 by projectgus

djokba, 2021-02-15:
@YouDONG-ESP i've tried your solution and so far the error doesn't happen anymore, but it looks like the android app still hasn't received any ACK messages from ESP32 even in the log it shows `smartconfig over`. Are there any changes required in the app code to make this solution works?

Sorry for late reply and thank you for your help.

YouDONG-ESP, 2021-02-20:
Hi @adfik, this issue has been fixed in commit d21ae1faa8676c11610107513f5a85050d9217e7, please update your idf to see the change.

thanosPrio, 2022-05-05:
> I think it should solve your problem, we will release a repair patch soon

Thank you! Very helpful your suggestion