espressif/esp-idf #6446: SDK 4.2 assertion Error while trying to read current stack pointer address

Joelmaca9, 2021-01-21:
While trying to read current stack pointer error I get an assertion error:

````
assertion "esp_ptr_in_dram((const void *)get_sp())" failed: file "/home/joel/Documents/SDK_4/idf_4_0/esp-idf/components/spi_flash/cache_utils.c", line 118, function: spi_flash_disable_interrupts_caches_and_other_cpu

abort() was called at PC 0x401069f8 on core 1
0x401069f8: __assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

Backtrace:0x4009351e:0x3f803aa0 0x40093c15:0x3f803ac0 0x40097dfa:0x3f803ae0 0x401069f8:0x3f803b50 0x400885f1:0x3f803b80 0x400895e1:0x3f803ba0 0x400895ed:0x3f803bc0 0x40090dd1:0x3f803be0 0x40088d4b:0x3f803c00 0x4011d339:0x3f803c30 0x401f909f:0x3f803c50 0x401f9269:0x3f803c80 0x401fa96a:0x3f803cb0 0x401f8405:0x3f803cf0 0x401f74df:0x3f803d20 0x401f7e85:0x3f803d60 0x401a32f5:0x3f803e30 0x4019f434:0x3f803e60 0x401a095d:0x3f803fa0 0x401a0a1f:0x3f803fc0 0x40161e81:0x3f804060 0x40095e4d:0x3f8040e0
````

negativekelvin, 2021-01-21:
> Regarding stacks in PSRAM: For tasks not calling on code in ROM (flash) in any way, directly or indirectly, the menuconfig option :ref:`CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY` will eliminate the check in xTaskCreateStatic, allowing a task's stack to be in external RAM. Using this is not advised, however.

Joelmaca9, 2021-01-22:
HI!
Tanks for the repply, I already have that option enabled, and with IDF 4_0 didn´t wive me any problem.

igrr, 2021-01-22:
Hi @Joelmaca9, could you add a bit more information, please?

" Could you attach your sdkconfig file?
" Can the issue be reproduced by esp-idf spiffs example?
" As hinted by @negativekelvin, do you happen to be creating tasks with the stack in SPIRAM?
* Could you please decide the backtrace using IDF Monitor? https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#register-dump-and-backtrace

Joelmaca9, 2021-01-22:
Hi @igrr !!
Of course here you have my sdkconfig:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5855480/sdkconfig.txt)

" Can the issue be reproduced by esp-idf spiffs example?
Maybe if you try to access the SPIFFS form a static created task It will fail, but I haven't tried it

" As hinted by @negativekelvin, do you happen to be creating tasks with the stack in SPIRAM?

 Yes, im creating task with the stack allocated in the psram but I need to allocate them there because Im running out of RAM if I don do that. 

Thanks for your help!!

igrr, 2021-01-22:
Unfortunately doing flash operations from tasks in SPIRAM is not supported at the moment. Earlier versions of ESP-IDF didn't have such checks, which sometimes resulted in hard-to-detect stack data corruption. These checks have been added in IDF 4.2.
The only option at the moment is to only perform flash operations in some other task, which has the stack in the internall memory.

closed 2021-07-20 by igrr