espressif/esp-idf #6447: TLS error when trying to reconnect to MQTTS broker after loosing internet connection

FrGrQuim, 2021-01-21:
## Environment

- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it):
   4.0.1
- Build System:        idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    8.2.0
- Operating System:     Linux

- Using an IDE?: No
- Power Supply:        external 5V

## Problem Description

I use ESP to send data on broker via MQTTS. The internet connection is done via ethernet. The first connection on the broker (after a reboot) always work fine but it happens that internet connection is lost and when ESP is reconnected to internet, I have TLS error "The certificate is not correctly signed by the trusted CA" when it try to reconnect to the broker. 

### Expected Behavior

The esp is able to reconnect to the broker via MQTTS

### Actual Behavior

The esp fail to verify the certificate even if it has already verified it during the first connection to the broker

### Steps to reproduce

1. Start a mqtt connection to a broker with CA, CERT and PRIVATE KEY
2. Disconnect the ethernet cable
3. Reconnect the ethernet cable

## Debug Logs

E (64123) esp-tls: mbedtls_ssl_handshake returned -0x2700
I (64133) esp-tls: Failed to verify peer certificate!
I (64133) esp-tls: verification info:   ! The certificate is not correctly signed by the trusted CA

E (64143) esp-tls: Failed to open new connection
E (64143) TRANS_SSL: Failed to open a new connection
E (64163) MQTT_CLIENT: Error transport connect
E (64163) MQTT_MANAGER: MQTT_ERROR_TYPE_ESP_TLS: esp_tls_last_esp_err = 32784
E (64163) MQTT_MANAGER: MQTT_ERROR_TYPE_ESP_TLS: esp_tls_stack_err = 9984
E (64173) MQTT_MANAGER: MQTT_ERROR_TYPE_ESP_TLS: esp_tls_cert_verify_flags = 8
I (64183) MQTT_MANAGER: MQTT_EVENT_DISCONNECTED

## Other items if possible

    mqtt_cfg.client_cert_pem = (char *)mqtt_cert_crt;
    mqtt_cfg.client_key_pem = (char *)mqtt_cert_key;
    mqtt_cfg.cert_pem = (char *)mqtt_cert_ca;

david-cermak, 2021-01-22:
Hi @FrGrQuim 

Is this happening every time you unplug the ethernet and plug again?
Could you please share more verbose log for us to better understand the problem?
Please enable mbedTLS debugging in component config:
```
(Top) → Component config → mbedTLS
[*] Enable mbedTLS debugging
```
And post just the last failing handshake (note that some part of the console log may contain your private key)

Also, thinking of potential memory corruption of the referenced data, please double check that the pointers to the certificates and keys remain valid for entire application lifetime as mentioned here: https://github.com/espressif/esp-idf/blob/master/components/tcp_transport/include/esp_transport_ssl.h#L63

FrGrQuim, 2021-01-22:
This happens every time internet connection is lost even if the ethernet cable isn't unpluged (I unplug ethernet cable in order to simulate it). I will make a test in order to check if the pointer to cert files are OK. While waiting it a send you the logs a have with mbedtls debug enable.
[MQTTS_fail_logs.log](https://github.com/espressif/esp-idf/files/5855828/MQTTS_fail_logs.log)

FrGrQuim, 2021-02-01:
Can someone help me? I can give more information if required.

david-cermak, 2021-02-02:
@FrGrQuim The log above shows a different failure, but this time it failed to parse the private key. This seems like the original thought on corruption of the cert data sounds more plausible.

Could you please post the portion of your code where you allocate the cert/key data? Or a simple check could be to just print out the textual data to the console and check if it stays the same after reconnecting; you can use `BEFORE_CONNECT` event for that.

Alvin1Zhang, 2021-08-31:
@FrGrQuim Thanks for reporting. Would you please help share if any updates for this issue? Thanks.

closed 2021-09-02 by Alvin1Zhang