espressif/esp-idf #6449: GATTC background auto connection not working?

howroyd, 2021-01-22:
With the gattc open function ([here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_gattc.html#_CPPv418esp_ble_gattc_open13esp_gatt_if_t13esp_bd_addr_t19esp_ble_addr_type_tb)) the last parameter is a boolean to signal a background auto connection.

However, setting this to true, appears to not do anything.  I get no events firing pertaining to the server bda I have set when this device appears on GAP.  Setting to false works fine but obviously only works whilst the server is alive, which most battery powered sensors are not all the time.

Is this function working with the flag set to true or is this not yet implemented?  If the latter then updating the API reference and the comments would be good as it is misleading that it works.  If the former then I am doing something wrong, but I don't think I am since setting to false works fine and I have lots of experience with this.

IDF release/4v1

howroyd, 2021-03-17:
Any thoughts on this, please?

xiewenxiang, 2021-03-19:
Hei @howroyd 

Background auto connection is not currently supported.

xiewenxiang, 2021-03-19:
Does this feature very necessary for your product? If so, we can include this feature in the development plan.

howroyd, 2021-03-19:
Yes it is key for us to connect to a battery powered sensor which is not permanently on. If you can't include it that would be very helpful for our product.

Also, in the interim, updating the documents which imply this feature works, might help out others wondering why it's not working.

Thank you

Alvin1Zhang, 2021-06-08:
Thanks for reporting and sharing updates, the documentation update is available https://github.com/espressif/esp-idf/commit/bd4e34b15a44c76923dbaf1dff34634012a5f1a5.

closed 2021-06-08 by Alvin1Zhang