espressif/esp-idf #6451: SDK BROKEN - any version that update from master ior subtree

filzek, 2021-01-22:
error while trying to build

 from D:/esp-idf/components/lwip/lwip/src/core/tcp_in.c:44:
D:/esp-idf/components/lwip/lwip/src/core/tcp_in.c: In function 'tcp_receive':
D:/esp-idf/components/lwip/lwip/src/core/tcp_in.c:1580:25: error: 'TCP_OOSEQ_DEBUG' undeclared (first use in this function); did you mean 'TCP_QLEN_DEBUG'?
             LWIP_DEBUGF(TCP_OOSEQ_DEBUG,
                         ^~~~~~~~~~~~~~~

seens that the header is not include in tcp_in for somehow.

need to clarify how to solve.

Alvin1Zhang, 2021-01-25:
@filzek Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

Martin-Guenther, 2021-01-26:
@filzek Did you run `git submodule update --init --recursive` after checking out?
I got the same error after forgetting to checkout recursively.

closed 2021-03-05 by Alvin1Zhang