espressif/esp-idf #6452: error in Client key passing for SSL  Mutual Authentication

sowmimalu, 2021-01-22:
In my application , i have worked on  ssl mutual authentication i can able to connect to my server with three certificates.[ssl certificate, client certificate  and  client key ] , passed certificates in esp_mqtt_client_config_t its worked properly but when passing  .clientkey_password = "1234" , getting invalid name specifier but in mqtt_cfg structure defined  clientkey_password  but unable to pass the key. how to pass clientkey password ?

esp_mqtt_client_config_t mqtt_cfg = { .uri = "mqtt://12.15.17.78",

		.username = "", .password = "", .event_handle = mqtt_event_handler,

		.cert_pem = "-----BEGIN CERTIFICATE-----\n"
						"MIIC+jCCAeKgAwIBAgIQdzsvZg0v6KZD1vZJWhKapTANBgkqhkiG9w0BAQsFADA5\n"
						"klGCn/6DMMQjMDH4bcZ8iv9TrBeaduOXhVJs1iQjcSkgtHaPAo+32AwDfRQlZA==\n"
						"-----END CERTIFICATE-----\n",
		.client_cert_pem = "",
		.client_key_pem = "",
	//	.clientkey_password = "1234",

Alvin1Zhang, 2021-01-25:
@sowmimalu Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

david-cermak, 2021-01-25:
Hi @sowmimalu 

You have to supply the key length too:
```
    const esp_mqtt_client_config_t mqtt_cfg = {
            ...
            .clientkey_password = "1234",
            .clientkey_password_len = 4
    };
```

see https://github.com/espressif/esp-idf/blob/master/tools/test_apps/protocols/mqtt/publish_connect_test/main/connect_test.c#L96

sowmimalu, 2021-01-25:
i am currently using esp idf 4.0 but client id not defined , for this in which idf version supported . in 4.0 mutual authentication working but with passowrd protected not working.

david-cermak, 2021-01-25:
I'm sorry, this feature is not supported in `v4.0`.

The mqtt client supports this configuration only from `v4.1` as defined here

https://github.com/espressif/esp-mqtt/blob/master/include/mqtt_supported_features.h#L44

sowmimalu, 2021-01-25:
thank you , installing 4.1 but  while build program getting errors as below
Python requirements from C:/msys32/home/swmi_emb/esp/esp-idf_4.1/requirements.txt are satisfied.
WARNING: Missing submodule components/micro-ecc/micro-ecc...
Attempting 'git submodule update --init components/micro-ecc/micro-ecc' in esp-idf root directory...
error: pathspec 'components/micro-ecc/micro-ecc' did not match any file(s) known to git
make[1]: *** [/home/swmi_emb/esp/esp-idf_4.1/make/project.mk:707: /home/swmi_emb/esp/esp-idf_4.1/components/micro-ecc/micro-ecc/.git] Error 1
make: *** [/home/swmi_emb/esp/esp-idf_4.1/components/bootloader/Makefile.projbuild:41: /home/swmi_emb/esp/esp-idf_4.1/examples/protocols/mqtt/ssl_mutual_auth/build/bootloader/bootloader.bin] Error 2

sowmimalu, 2021-01-25:
[issue.txt](https://github.com/espressif/esp-idf/files/5865199/issue.txt)
please find the attached text file , errors has mentioned

sowmimalu, 2021-01-25:
installed  https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp-2020r2-win64.zip the same version , but getting the same error

david-cermak, 2021-02-02:
@sowmimalu Seems you have missed to export your variables.

Could you please close this issue, as you've been apparently able to successfully install `v4.1` (and raise the related https://github.com/espressif/esp-idf/issues/6465)

sowmimalu, 2021-02-03:
esp idf v4.1 installed thanks, i have missed toolchain got issue now successfully installed

closed 2021-02-03 by sowmimalu