espressif/esp-idf #6453: TRANS_SSL : Software caused connection abort Error

anubhav1, 2021-01-22:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.3-dev-2136-gb0150615d
- Build System:  idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:    WSL2 on windows 10
- (Windows only) environment type: ESP IDF  extension on visual code
- Using an IDE?:Visual code
- Power Supply:         USB

## Problem Description
Seems like esp32 cant reconnect with internet after internet connectivity is lost . my esp32 is publishing data using mqtt protocol to aws.

### Expected Behavior
Should be able to reconnect with internet automatically 
### Actual Behavior
Error : 
```
 TRANS_SSL: ssl_poll_read select error 113, errno = Software caused connection abort, fd = 55
 TRANS_SSL: ssl_poll_read select error 0, errno = Success, fd = 55
```

Logs:

```
(3876345) AWS Connect: Stack remaining for task 'aws_iot_task' is 140 bytes
I (3936345) AWS Connect: HTTP_EVENT_HEADER_SENT
I (3936415) AWS Connect: HTTP_EVENT_ON_HEADER, key=Date, value=Fri, 22 Jan 2021 17:37:38 GMT
I (3936415) AWS Connect: HTTP_EVENT_ON_HEADER, key=Content-Type, value=application/json; charset=utf-8
I (3936425) AWS Connect: HTTP_EVENT_ON_HEADER, key=Content-Length, value=2136
I (3936435) AWS Connect: HTTP_EVENT_ON_HEADER, key=Connection, value=keep-alive
I (3936445) AWS Connect: HTTP_EVENT_ON_HEADER, key=Vary, value=Accept-Encoding
I (3936445) AWS Connect: HTTP_EVENT_ON_HEADER, key=Tracking-ID, value=cfd0ba4b-f57c-44ba-9ef0-f1e204ef75f9
I (3936455) AWS Connect: HTTP_EVENT_ON_HEADER, key=Access-Control-Allow-Origin, value=*
I (3936465) AWS Connect: HTTP_EVENT_ON_HEADER, key=Expires, value=Fri, 22 Jan 2021 17:37:38 GMT
I (3936475) AWS Connect: HTTP_EVENT_ON_HEADER, key=Cache-Control, value=public, max-age=0
I (3936485) AWS Connect: HTTP_EVENT_ON_HEADER, key=Strict-Transport-Security, value=max-age=15724800; includeSubDomains
I (3936495) AWS Connect: HTTP_EVENT_ON_HEADER, key=Access-Control-Expose-Headers, value=Tracking-ID
I (3936505) AWS Connect: HTTP_EVENT_ON_DATA, 75
I (3936515) AWS Connect: HTTP_EVENT_ON_DATA, 512
I (3936515) AWS Connect: HTTP_EVENT_ON_DATA, 512
I (3936525) AWS Connect: HTTP_EVENT_ON_DATA, 512
I (3936525) AWS Connect: HTTP_EVENT_ON_DATA, 512
I (3936535) AWS Connect: HTTP_EVENT_ON_DATA, 13
I (3936535) AWS Connect: HTTP_EVENT_ON_FINISH
I (3936545) AWS Connect: Status = 200, content_length = 2136
I (3936555) AWS Connect:  Sending JSON Response to AWS : {"flowSegmentData":{"frc":"FRC0","currentSpeed":83,"freeFlowSpeed":83,"curre
ntTravelTime":30,"freeFlowTravelTime":30,"confidence":1,"roadClosure":false,"coordinates":{"coordinate":[{"latitude":48.7870878489149
86,"longitude":2.347790839653726},{"latitude":48.78720587687751,"longitude":2.3477353915120602},{"latitude":48.787543950744585,"longi
tude":2.347569572480296},{"latitude":48.78774835271241,"longitude":2.347482722057265},{"latitude":48.78804025702684,"longitude":2.347
3402259198792},{"latitude":48.78828976281337,"longitude":2.3472273999739457},{"latitude":48.78848364316665,"longitude":2.347108853394
7295},{"latitude":48.78934215171858,"longitude":2.3467004436268724},{"latitude":48.78971838967497,"longitude":2.346532193757554},{"la
titude":48.789812905401824,"longitude":2.3464885711354952},{"latitude":48.79005004312096,"longitude":2.346402142848291},{"latitude":4
8.790165362491535,"longitude":2.3463706986377133},{"latitude":48.79031453477112,"longitude":2.3463300208564135},{"latitude":48.790411
29050827,"longitude":2.3463071983134967},{"latitude":48.79043725999192,"longitude":2.346303380544498},{"latitude":48.79046908224158,"
longitude":2.346297591508744},{"latitude":48.79048730080616,"longitude":2.3462955287724867},{"latitude":48.79059841660334,"longitude"
:2.346291214516725},{"latitude":48.790716710990864,"longitude":2.3462864342947967},{"latitude":48.79080150172026,"longitude":2.346286
4342947967},{"latitude":48.79082427124021,"longitude":2.346286889801661},{"latitude":48.79083692305993,"longitude":2.34628839600191},
{"latitude":48.79097090689136,"longitude":2.346306877877879},{"latitude":48.79108307227994,"longitude":2.346326816276701},{"latitude"
:48.791171078245355,"longitude":2.346341008472507},{"latitude":48.791258593604006,"longitude":2.3463654387409747},{"latitude":48.7917
04355162985,"longitude":2.346471697787166},{"latitude":48.79198474835557,"longitude":2.3465401180430376},{"latitude":48.7922653394340
1,"longitude":2.3466069285518643},{"latitude":48.793024816906346,"longitude":2.346794432197356},{"latitude":48.79332339483452,"longit
ude":2.3468618575674896}]},"@version":"traffic-service 4.1.002"}}}
I (3936765) AWS Connect: Stack remaining for task 'aws_iot_task' is 140 bytes
I (3996765) AWS Connect: HTTP_EVENT_HEADER_SENT
I (3996905) AWS Connect: HTTP_EVENT_ON_HEADER, key=Date, value=Fri, 22 Jan 2021 17:38:39 GMT
I (3996905) AWS Connect: HTTP_EVENT_ON_HEADER, key=Content-Type, value=application/json; charset=utf-8
[... 57 lines cut ...]
I (5287245) AWS Connect: Stack remaining for task 'aws_iot_task' is 140 bytes
E (5347245) TRANS_SSL: ssl_poll_read select error 0, errno = Success, fd = 55
I (5347245) AWS Connect: Stack remaining for task 'aws_iot_task' is 140 bytes
E (5407245) TRANS_SSL: ssl_poll_read select error 0, errno = Success, fd = 55
I (5407245) AWS Connect: Stack remaining for task 'aws_iot_task' is 140 bytes
E (5467245) TRANS_SSL: ssl_poll_read select error 0, errno = Success, fd = 55
I (5467245) AWS Connect: Stack remaining for task 'aws_iot_task' is 140 bytes
E (5527245) TRANS_SSL: ssl_poll_read select error 0, errno = Success, fd = 55
I (5527245) AWS Connect: Stack remaining for task 'aws_iot_task' is 140 bytes
E (5587245) TRANS_SSL: ssl_poll_read select error 0, errno = Success, fd = 55
I (5587245) AWS Connect: Stack remaining for task 'aws_iot_task' is 140 bytes
E (5647245) TRANS_SSL: ssl_poll_read select error 0, errno = Success, fd = 55
I (5647245) AWS Connect: Stack remaining for task 'aws_iot_task' is 140 bytes
E (5707245) TRANS_SSL: ssl_poll_read select error 0, errno = Success, fd = 55
I (5707245) AWS Connect: Stack remaining for task 'aws_iot_task' is 140 bytes
E (5767245) TRANS_SSL: ssl_poll_read select error 0, errno = Success, fd = 55
I (5767245) AWS Connect: Stack remaining for task 'aws_iot_task' is 140 bytes
E (5827245) TRANS_SSL: ssl_poll_read select error 0, errno = Success, fd = 55
I (5827245) AWS Connect: Stack remaining for task 'aws_iot_task' is 140 bytes
E (5887245) TRANS_SSL: ssl_poll_read select error 0, errno = Success, fd = 55
I (5887245) AWS Connect: Stack remaining for task 'aws_iot_task' is 140 bytes
E (5947245) TRANS_SSL: ssl_poll_read select error 0, errno = Success, fd = 55
I (5947245) AWS Connect: Stack remaining for task 'aws_iot_task' is 140 bytes
E (6007245) TRANS_SSL: ssl_poll_read select error 0, errno = Success, fd = 55
```

Alvin1Zhang, 2021-01-25:
Thanks for reporting, we will look into.

yuanjianmin, 2021-02-05:
@anubhav1 It seems that the data packet sent by mqtt has not responded to the opposite end, which triggers tcp retransmission. After the number of retransmissions is reached, error 113 is reported. Can you help catch the air wifi packet and see if the packet is sent successfully and check if the server responds.

anubhav1, 2021-02-05:
are you talking about capturing packet Using Wireshark?

yuanjianmin, 2021-02-05:
> 
> 
> are you talking about capturing packet Using Wireshark?

Yes，Do you have an environment where you can capture packages.

anubhav1, 2021-02-05:
Unfortunately, not.

yuanjianmin, 2021-02-05:
Ok, Could you please open
`LWIP_DEBUG` to `#define LWIP_DEBUG     LWIP_DBG_ON` in `components/lwip/port/esp32/include/lwipopts.h` 
`TCP_DEBUG` to `#define TCP_DEBUG     LWIP_DBG_ON` in `components/lwip/port/esp32/include/lwipopts.h` 
`TCP_RTO_DEBUG` to `#define TCP_RTO_DEBUG    LWIP_DBG_ON` in `components/lwip/lwip/src/include/lwip/opt.h` ,
reproduce the issue and reply log to me. Or could you offer the method or demo that i can reproduce it.

anubhav1, 2021-02-05:
I have turned on the debugs as you mentioned. I will post logs once i get that error again.

yuanjianmin, 2021-02-25:
@anubhav1  Is this issue any updates. If yes, please let me know.

anubhav1, 2021-02-28:
For some unknown reasons I am not getting that error again. Closing this issue for now.

closed 2021-02-28 by anubhav1