espressif/esp-idf #6456: sdkconfig.defaults

chegewara, 2021-01-23:
Im not sure if this is real issue, but i would like to change flash size with sdconfig.defaults. 
Here is my setup:
```
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_ESPTOOLPY_FLASHSIZE_16MB=y
CONFIG_ESPTOOLPY_FLASHSIZE="16MB"
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions16.csv"
CONFIG_PARTITION_TABLE_FILENAME="partitions16.csv"
```
but all settings are used from sdkconfig.
It is possible that my issue is related to fact i am using arduino as component.

ESP-IDF: b0150615dff529662772a60dcb57d5b559f480e2

I have the same partitions16.csv set in sdkconfig, but flash size is 4MB and i am getting this error, which tells me that defaults are not used:
```
Partitions defined in '/home/chegewara/upwork/othernet/arduino/dreamcatcher_as_component/partitions16.csv' occupy 16.0MB of flash (16777216 bytes) which does not fit in configured flash size 4MB. Change the flash size in menuconfig under the 'Serial Flasher Config' menu.
```
and yes, my wrover module is with 16MB flash and all is fine when i set it in menuconfig.

dobairoland, 2021-01-23:
@chegewara  `sdkconfig.defaults` is used only when no `sdkconfig` file exists. You have to remove your `sdkconfig` file. This is the intended behaviour.

chegewara, 2021-01-23:
Ok, that explains my problems. Thanks.

closed 2021-01-23 by chegewara