espressif/esp-idf #6457: crosstool-NG: math & printf related errors

yskab, 2021-01-23:
## Environment

- Chip:  ESP32D2WDQ5 (revision 1)
- IDF version : framework-espidf 3.40001.200521 (4.0.1) 
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System: Linux
- IDE: PlatformIO
- Power Supply: USB

## Problem Description

The application is using MPU9250 sensor to capture accelerometer and magnetometer data. Both functions request data from the sensor via I2C interface.

While testing, everything seems fine and I am able to get expected outputs of heading and Accel_mag but sometimes after around 10mins, randomly I get Errors( as present in Debug logs) which on checking with GDB looks like functionality is missing from the compiler( not sure why functions worked before error). Backtrace in Debug logs.

### Expected Behavior
Expecting math functions and printf to produce output without errors.
### Actual Behavior
After 10mins, errors resulting in failures.

### Code snippet

```c
float get_mag_heading()
{
    //orientation is fixed
    get_MagData(&mx,&my,&mz,&mpu);
    float heading = (180 + (atan2(my, mz) * 180 / M_PI));
    return heading;
    
}

void detect_step(float ref_dir)
{
    //load buffer continuously, until full
    if(win_pos < WIN_LEN)
    {
        if(get_AccelData(&ax,&ay,&az,&mpu))
        { 
            float Accel_mag  = sqrt(ax*ax + ay*ay + az*az);
            low_pass_filter(Accel_mag,win_pos);  
            if(DEBUG)
                printf("win pos:%d Accel val %f\n",win_pos,Accel_Window[win_pos]);

            Mag_Window[win_pos] = get_mag_heading();
            
            win_pos++;
        }
}
                       
```

## Debug Logs

atan:
```
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x22620000  PS      : 0x00060530  A0      : 0x8012bf16  A1      : 0x3ffbc420  
A2      : 0x7334ed6c  A3      : 0xbf820002  A4      : 0xf4264984  A5      : 0x3e39a10d  
A6      : 0x3fef7141  A7      : 0x00000001  A8      : 0x22620000  A9      : 0x3ffbc400  
A10     : 0x151b2a49  A11     : 0x3dfb5096  A12     : 0xc54c206e  A13     : 0x3fb745cd  
A14     : 0x0b106000  A15     : 0x00000000  SAR     : 0x00000014  EXCCAUSE: 0x00000014  
EXCVADDR: 0x22620000  LBEG    : 0x400029ac  LEND    : 0x400029cb  LCOUNT  : 0x00000000  

ELF file SHA256: 7204e03b55c74eab

Backtrace: 0x2261fffd:0x3ffbc420 |<-CORRUPTED

Entering gdb stub now.
$T0b#e6

emote debugging using /dev/ttyUSB0
0x62620000 in ?? ()
(gdb) bt
#0  0x62620000 in ?? ()
#1  0x4012bf16 in __ieee754_atan2 (y=<optimized out>, x=-32.939697265625)
    at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libm/math/e_atan2.c:116
#2  0x4012bc89 in atan2 (y=32.36572265625, x=-32.939697265625)
    at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libm/math/w_atan2.c:77
#3  0x400d21f0 in get_mag_heading () at lib/motion_track/motion_track.c:90
#4  0x400d22a1 in detect_step (ref_dir=<optimized out>) at lib/motion_track/motion_track.c:112
#5  0x400d1902 in app_main () at src/main.c:321
#6  0x400d8ae9 in main_task (args=0x0) at /home/abhijeet/.platformio/packages/framework-espidf/components/esp32/cpu_start.c:553
#7  0x400931d4 in vPortTaskWrapper (pxCode=0x400d8a84 <main_task>, pvParameters=0x0)
    at /home/abhijeet/.platformio/packages/framework-espidf/components/freertos/port.c:143
(gdb) 
```
sqrt:

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4012c5c6  PS      : 0x00060330  A0      : 0x8012bc42  A1      : 0x3ffbc440  
A2      : 0x00200000  A3      : 0x00400000  A4      : 0x00000000  A5      : 0x3ffb397c  
A6      : 0x7ff00000  A7      : 0x004e6269  A8      : 0x004d3117  A9      : 0x00200000  
A10     : 0x00000000  A11     : 0xffffffff  A12     : 0x00000016  A13     : 0x00600000  
A14     : 0x3ffb397c  A15     : 0x3ffbc320  SAR     : 0x00000013  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x004d3117  LBEG    : 0x4012c5aa  LEND    : 0x4012c5c6  LCOUNT  : 0x00000000  

ELF file SHA256: fcc9c027558c6374

Backtrace: 0x4012c5c3:0x3ffbc440 0x4012bc3f:0x3ffbc460 0x400d2215:0x3ffbc4b0 0x400d18f3:0x3ffbc4d0 0x400d8a7e:0x3ffbc500 0x400931d1:0x3ffbc520

Entering gdb stub now.
$T0b#e6

emote debugging using /dev/ttyUSB0
0x4012c5c6 in __ieee754_sqrt (x=<optimized out>)
---Type <return> to continue, or q <return> to quit---
    at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libm/math/e_sqrt.c:155
155     /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libm/math/e_sqrt.c: No such file or directory.
(gdb) bt
#0  0x4012c5c6 in __ieee754_sqrt (x=<optimized out>)
    at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libm/math/e_sqrt.c:155
#1  0x4012bc42 in sqrt (x=0.85306060314178467)
    at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libm/math/w_sqrt.c:64
#2  0x400d2218 in detect_step (ref_dir=<optimized out>) at lib/motion_track/motion_track.c:107
#3  0x400d18f6 in app_main () at src/main.c:321
#4  0x400d8a81 in main_task (args=0x0) at /home/abhijeet/.platformio/packages/framework-espidf/components/esp32/cpu_start.c:553
#5  0x400931d4 in vPortTaskWrapper (pxCode=0x400d8a1c <main_task>, pvParameters=0x0)
    at /home/abhijeet/.platformio/packages/framework-espidf/components/freertos/port.c:143
(gdb) ^CQuit
```

vfprintf:

```
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4012479c: 61a2a421 0080042b 80000800
Core 0 register dump:
PC      : 0x401247a1  PS      : 0x00060530  A0      : 0x8011e514  A1      : 0x3ffbc150  
A2      : 0x3ffbc614  A3      : 0x3ffae920  A4      : 0x3f40047c  A5      : 0x3ffbc490  
A6      : 0x3ffbc470  A7      : 0x00000004  A8      : 0x3ffbc614  A9      : 0x3ffbc140  
A10     : 0x00000001  A11     : 0x3f4112a5  A12     : 0x000000ff  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: dce0e6783ab20ee9

Backtrace: 0x4012479e:0x3ffbc150 0x4011e511:0x3ffbc460 0x400d1f31:0x3ffbc4b0 0x400d17ef:0x3ffbc4d0 0x400d876e:0x3ffbc500 0x4009315d:0x3ffbc520

Entering gdb stub now.
$T04#b8

Remote debugging using /dev/ttyUSB2
_vfprintf_r (data=0x3ffbc614, fp=0x3ffae920, fmt0=0x3f40047c "win pos:%d Accel val %f\n", ap=...)
    at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:852
852     /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c: No such file or directory.
(gdb) bt
#0  _vfprintf_r (data=0x3ffbc614, fp=0x3ffae920, fmt0=0x3f40047c "win pos:%d Accel val %f\n", ap=...)
    at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:852
#1  0x4011e514 in printf (fmt=0x3f40047c "win pos:%d Accel val %f\n")
    at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/printf.c:56
#2  0x400d1f34 in detect_step (ref_dir=<optimized out>) at lib/motion_track/motion_track.c:109
#3  0x400d17f2 in app_main () at src/main.c:321
#4  0x400d8771 in main_task (args=0x0) at /home/abhijeet/.platformio/packages/framework-espidf/components/esp32/cpu_start.c:553
#5  0x40093160 in vPortTaskWrapper (pxCode=0x400d870c <main_task>, pvParameters=0x0)
    at /home/abhijeet/.platformio/packages/framework-espidf/components/freertos/port.c:143
(gdb) 
```

## What I Have Tried:

- Powering via a battery.
- Reinstalling esp-framework.
- Compiling from "toolchain-xtensa32", version: 2.80200.200827

Alvin1Zhang, 2021-01-25:
Thanks for reporting.

igrr, 2021-01-25:
@yskab could you please try replacing the USB cable? Given the rare occurrence, the exceptions look like they might be power supply glitches.

* atan: location corresponds to the call `atan(fabs(y/x))` in `__ieee754_atan2`, however the CPU jumps to an invalid address, 0x22620000. If this was a code generation issue, the CPU would go to this address every time, which means that you would see this issue sooner than after 10 minutes. However, such random CPU behavior can occur in case of a power supply glitch.
* sqrt: here an invalid pointer `0x004d3117` is dereferenced. It is hard to tell where the value originates without looking at the disassembly around the PC. This might be memory corruption (e.g. out of bounds write) or a power supply glitch.
* vfprintf: illegal instruction exception is generated at 0x401247a1, but the subsequent memory dump shows that the instruction is, in fact, valid: `048000` is `lsi f0, a0, 0`. It would be best to check the disassembly (if you still have the ELF file for that specific build) to confirm whether this instruction is indeed present. If this was a code generation issue, the CPU would generate illegal instruction exception on the first execution of this code, so a glitch (either in the CPU or in flash memory) is likely.

yskab, 2021-01-29:
Hi, @igrr Thanks for looking into this. Yes, this is a custom PCB with a Vendor provided ESP module. PCB is responsible for the power supply and programming interface.

As per your suggestion, I Tried testing with a different USB cable, errors appeared after 1hr. The power supply also looked stable, then I reached out to the vendor for this. They advised changing the SDIO flash voltage to 3.3V via espefuse. After performing this I have tested twice for 10hrs straight, no errors appeared. Looks like setting the correct flash voltage has resolved it.
Thanks for your assistance, I'll close this issue.

closed 2021-01-29 by yskab