espressif/esp-idf #6458: E (365) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107

kaibin1111, 2021-01-24:
I (0) cpu_start: App cpu up.
I (260) heap_init: Initializing. RAM available for dynamic allocation:
I (267) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (273) heap_init: At 3FFB2A98 len 0002D568 (181 KiB): DRAM
I (279) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (285) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (292) heap_init: At 4008A838 len 000157C8 (85 KiB): IRAM
I (298) cpu_start: Pro cpu start user code
I (316) spi_flash: detected chip: generic
I (317) spi_flash: flash io: dio
I (317) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (325) example: Initializing SD card
I (325) example: Using SDMMC peripheral
I (335) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
E (365) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (365) example: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.

Alvin1Zhang, 2021-01-25:
@kaibin1111 Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

kaibin1111, 2021-01-25:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6980
load:0x40078000,len:13072
load:0x40080400,len:3896
0x40080400: _init at ??:?

entry 0x40080688
I (29) boot: ESP-IDF v4.1-dirty 2nd stage bootloader
I (29) boot: compile time 10:58:50
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (96) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x09798 ( 38808) map
I (127) esp_image: segment 1: paddr=0x000197c0 vaddr=0x3ffb0000 size=0x02198 (  8600) load
I (130) esp_image: segment 2: paddr=0x0001b960 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:/Users/make/Desktop/esp-idf/components/freertos/xtensa_vectors.S:1778

I (134) esp_image: segment 3: paddr=0x0001bd6c vaddr=0x40080404 size=0x042ac ( 17068) load
I (150) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x1cb68 (117608) map
0x400d0020: _stext at ??:?

I (196) esp_image: segment 5: paddr=0x0003cb90 vaddr=0x400846b0 size=0x06188 ( 24968) load
0x400846b0: xt_unhandled_exception at C:/Users/make/Desktop/esp-idf/components/esp32/panic.c:361

I (213) boot: Loaded app from partition at offset 0x10000
I (213) boot: Disabling RNG early entropy source...
I (213) cpu_start: Pro cpu up.
I (217) cpu_start: Application information:
I (222) cpu_start: Project name:     sd_card
I (227) cpu_start: App version:      1
I (231) cpu_start: Compile time:     Jan 25 2021 10:58:38
I (237) cpu_start: ELF file SHA256:  6b10099a816da7f1...
I (243) cpu_start: ESP-IDF:          v4.1-dirty
I (248) cpu_start: Starting app cpu, entry point is 0x40081088
0x40081088: call_start_cpu1 at C:/Users/make/Desktop/esp-idf/components/esp32/cpu_start.c:271

I (0) cpu_start: App cpu up.
I (259) heap_init: Initializing. RAM available for dynamic allocation:
I (266) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (272) heap_init: At 3FFB2A98 len 0002D568 (181 KiB): DRAM
I (278) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (284) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (291) heap_init: At 4008A838 len 000157C8 (85 KiB): IRAM
I (297) cpu_start: Pro cpu start user code
I (315) spi_flash: detected chip: generic
I (316) spi_flash: flash io: dio
W (316) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (326) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (337) example: Initializing SD card
I (337) example: Using SDMMC peripheral
I (347) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
E (377) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (377) example: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.

closed 2021-09-01 by kaibin1111

MATTYGILO, 2022-03-09:
How did you fix this?