espressif/esp-idf #6460: Install issue

wrickout, 2021-01-24:
I got this error while installing:
install.bat
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, esp32s2ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache, dfu-util
Skipping xtensa-esp32-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s3-elf@esp-2020r3-8.4.0 (already installed)
Skipping riscv32-esp-elf@1.24.0.123_64eb9ff-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping cmake@3.16.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20200709 (already installed)
Skipping ninja@1.10.0 (already installed)
Skipping idf-exe@1.0.1 (already installed)
Skipping ccache@3.7 (already installed)
Skipping dfu-util@0.9 (already installed)
Setting up Python environment
Traceback (most recent call last):
  File "C:\Users\Sk3tCh\Documents\GitHub\esp-idf\tools\idf_tools.py", line 1622, in <module>
    main(sys.argv[1:])
  File "C:\Users\Sk3tCh\Documents\GitHub\esp-idf\tools\idf_tools.py", line 1618, in main
    action_func(args)
  File "C:\Users\Sk3tCh\Documents\GitHub\esp-idf\tools\idf_tools.py", line 1316, in action_install_python_env
    idf_python_env_path, _, virtualenv_python = get_python_env_path()
  File "C:\Users\Sk3tCh\Documents\GitHub\esp-idf\tools\idf_tools.py", line 959, in get_python_env_path
    idf_version_str = subprocess.check_output(['git', 'describe'],
  File "C:\Program Files\Python39\lib\subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Program Files\Python39\lib\subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Program Files\Python39\lib\subprocess.py", line 947, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Program Files\Python39\lib\subprocess.py", line 1416, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified

Kiznoh, 2021-01-29:
Hi, do you have git installed?

georgik, 2021-02-05:
@wrickout Thank you for reporting the issue. 

The problem might by caused by missing Git as mentioned by @Kiznoh . Please try to install it.

Other solution: Use new installer 2.5 which contains several corrections and checks for the Windows environment.
You can download one of following installers:
- Online installer (20 MB) : https://dl.espressif.com/dl/esp-idf-tools-setup-online-2.5.exe
- Offline installer - contains v4.2 and v4.1.1 (810 MB): https://dl.espressif.com/dl/esp-idf-tools-setup-offline-2.5.exe

dobairoland, 2021-03-04:
I'm closing this for lack of activity.

closed 2021-03-04 by dobairoland