espressif/esp-idf #6462: WIFI issue: failed to load RF calibration data (0xffffffff), falling back to full calibration at first when flashing the device

chaabanemalki, 2021-01-25:
## Environment
- Module or chip used:     Custom board ESP32D2WDQ5 (revision 1)
- IDF version :      v4.1-352-g5d258af93c
- Build System:     idf.py
- Compiler version:  xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows

## Problem Description

We are facing WIFI performance issues as well as WIFI connection stability and behaviors.
Our first step is to fix all warnings/ Errors regarding our WIFI code to setup WIFI.

We setup our WIFI to AP mode and stop dhcps to set a specific IP address then start WIFI. This works and I'm able to connect to my esp32 and do HTTP post/get commands, but I always get the following error:
` error: failed to load RF calibration data (0xffffffff), falling back to full calibration at first when flashing the device.`

This might be the reason for some bad WIFI performance in some devices

### Code to reproduce this issue
In SDK I have the following config
```cpp
#
# Wi-Fi
#
CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=10
CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=32
CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER=y
CONFIG_ESP32_WIFI_TX_BUFFER_TYPE=1
CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM=32
CONFIG_ESP32_WIFI_AMPDU_TX_ENABLED=y
CONFIG_ESP32_WIFI_TX_BA_WIN=6
CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED=y
CONFIG_ESP32_WIFI_RX_BA_WIN=6
CONFIG_ESP32_WIFI_NVS_ENABLED=y
CONFIG_ESP32_WIFI_TASK_PINNED_TO_CORE_0=y
CONFIG_ESP32_WIFI_SOFTAP_BEACON_MAX_LEN=752
CONFIG_ESP32_WIFI_MGMT_SBUF_NUM=32
CONFIG_ESP32_WIFI_IRAM_OPT=y
CONFIG_ESP32_WIFI_RX_IRAM_OPT=y
CONFIG_ESP32_WIFI_ENABLE_WPA3_SAE=y
# end of Wi-Fi
#
# PHY
#
CONFIG_ESP32_PHY_CALIBRATION_AND_DATA_STORAGE=y
CONFIG_ESP32_PHY_MAX_WIFI_TX_POWER=20
CONFIG_ESP32_PHY_MAX_TX_POWER=20
# end of PHY
```
```cpp
#include "esp_log.h"

void app_main()
{

    tcpip_adapter_init();
    esp_err_t res = esp_event_loop_create_default();

    esp_netif_t *wifi_ap = esp_netif_create_default_wifi_ap();
    res = esp_netif_dhcps_stop(wifi_ap);

    esp_netif_ip_info_t ip_info;

    IP4_ADDR(&ip_info.ip, 192, 168, 4, 1);
    IP4_ADDR(&ip_info.gw, 192, 168, 4, 1);
    IP4_ADDR(&ip_info.netmask, 255, 255, 255, 0);

    res = esp_netif_set_ip_info(wifi_ap, &ip_info);
    ESP_LOGI(TAG, "esp_netif_set_ip_info(): 0x%X", res);

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_init(&cfg));

    ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &event_handler, NULL));

    ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &event_handler, NULL));

    wifi_config_t wifi_config = {
        .ap = {
            .max_connection = WIFI_AP_MAX_CONN_DEVICES,
            .authmode = WIFI_AUTH_WPA_WPA2_PSK},
    };
    strcpy((char *)wifi_config.ap.ssid, _ssid);
    strcpy((char *)wifi_config.ap.password, _passwd);
    wifi_config.ap.ssid_len = strlen(_ssid);
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_set_mode(WIFI_MODE_AP));
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config));
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_set_protocol(ESP_IF_WIFI_AP, WIFI_PROTOCOL_11B));
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_start());
}

```

Can you please assist us on debugging this error ?

boarchuz, 2021-01-25:
When CONFIG_ESP32_WIFI_NVS_ENABLED or CONFIG_ESP32_PHY_CALIBRATION_AND_DATA_STORAGE, you need to initialise NVS (before esp_wifi_init):

https://github.com/espressif/esp-idf/blob/0b71a0a46d67cce681ec55973b020d950d8596bd/examples/wifi/getting_started/softAP/main/softap_example_main.c#L86-L92

Alvin1Zhang, 2021-01-26:
Thanks for reporting, we will look into,

YouDONG-ESP, 2021-02-02:
HI @chaabanemalki, dose @boarchuz's suggestion solve your problem？ If not, dose this log only shown once or multiple times? It's normal to have this log once when you power on a esp32.

chaabanemalki, 2021-02-02:
Hello,
I have already the piece of code that @boarchuz  suggested.

When I freshly compile the project I get the error. See logs below

```cpp
--- idf_monitor on COM13 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (6395) WIFI_AP: esp_event_loop_create_default(): 0
I (6402) WIFI_AP: tcpip_adapter_dhcps_stop(): 0
I (6406) WIFI_AP: tcpip_adapter_set_ip_info(): 0
I (6452) wifi:wifi firmware version: 04f00f3
I (6452) wifi:config NVS flash: enabled
I (6452) wifi:config nano formating: enabled
I (6453) wifi:Init data frame dynamic rx buffer num: 32
I (6457) wifi:Init management frame dynamic rx buffer num: 32
I (6463) wifi:Init management short buffer num: 32
I (6467) wifi:Init dynamic tx buffer num: 32
I (6472) wifi:Init static rx buffer size: 1600
I (6476) wifi:Init static rx buffer num: 10
I (6479) wifi:Init dynamic rx buffer num: 32
W (6484) event: handler already registered, overwriting
W (6497) phy_init: failed to load RF calibration data (0xffffffff), falling back to full calibration
I (6809) wifi:mode : softAP (b4:e6:2d:80:13:f6)
I (6813) wifi:Total power save buffer number: 16
I (6813) wifi:Init max length of beacon: 752/752
I (6814) wifi:Init max length of beacon: 752/752
I (6818) WIFI_AP: WiFi AP Started
I (6820) WIFI_AP: max uri handlers: 10
```
If I reset the esp32 I don't get this error. So I'm not what config is really used

YouDONG-ESP, 2021-02-02:
@chaabanemalki  That should not be an issue, it's normal to have it once when you power on esp32.

chaabanemalki, 2021-02-03:
@YouDONG-ESP so the warning message is miss leading !
how can I make sure that the config I chose is the one being used ?
What is the full calibration ?
For example I want the power to be 20dB ( CONFIG_ESP32_PHY_MAX_TX_POWER=20) how can I check if I am using this maximum value ?

YouDONG-ESP, 2021-02-03:
Hi, these apis could help you set and get power.
```
/**
  * @brief     Set maximum transmitting power after WiFi start.
  *
  * @attention 1. Maximum power before wifi startup is limited by PHY init data bin.
  * @attention 2. The value set by this API will be mapped to the max_tx_power of the structure wifi_country_t variable.
  * @attention 3. Mapping Table {Power, max_tx_power} = {{8,   2}, {20,  5}, {28,  7}, {34,  8}, {44, 11},
  *                                                      {52, 13}, {56, 14}, {60, 15}, {66, 16}, {72, 18}, {78, 20}}.
  * @attention 4. Param power unit is 0.25dBm, range is [8, 78] corresponding to 2dBm - 20dBm.
  * @attention 5. Relationship between set value and actual value. As follows:
  *              +------------+--------------+
  *              | set value  | actual value |
  *              +============+==============+
  *              |  [8,  19]  |      8       |
  *              +------------+--------------+
  *              |  [20, 27]  |      20      |
  *              +------------+--------------+
  *              |  [28, 33]  |      28      |
  *              +------------+--------------+
  *              |  [34, 43]  |      34      |
  *              +------------+--------------+
  *              |  [44, 51]  |      44      |
  *              +------------+--------------+
  *              |  [52, 55]  |      52      |
  *              +------------+--------------+
  *              |  [56, 59]  |      56      |
  *              +------------+--------------+
  *              |  [60, 65]  |      60      |
  *              +------------+--------------+
  *              |  [66, 71]  |      66      |
  *              +------------+--------------+
  *              |  [72, 77]  |      72      |
  *              +------------+--------------+
  *              |     78     |      78      |
  *              +------------+--------------+
  * @param     power  Maximum WiFi transmitting power.
  *
  * @return
  *    - ESP_OK: succeed
  *    - ESP_ERR_WIFI_NOT_INIT: WiFi is not initialized by esp_wifi_init
  *    - ESP_ERR_WIFI_NOT_START: WiFi is not started by esp_wifi_start
  *    - ESP_ERR_WIFI_ARG: invalid argument, e.g. parameter is out of range
  */
esp_err_t esp_wifi_set_max_tx_power(int8_t power);

/**
  * @brief     Get maximum transmiting power after WiFi start
  *
  * @param     power Maximum WiFi transmitting power, unit is 0.25dBm.
  *
  * @return
  *    - ESP_OK: succeed
  *    - ESP_ERR_WIFI_NOT_INIT: WiFi is not initialized by esp_wifi_init
  *    - ESP_ERR_WIFI_NOT_START: WiFi is not started by esp_wifi_start
  *    - ESP_ERR_WIFI_ARG: invalid argument
  */
esp_err_t esp_wifi_get_max_tx_power(int8_t *power);

```
The RF calibration  method can be found here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/RF_calibration.html?

chaabanemalki, 2021-02-03:
@YouDONG-ESP Thank you for these information, very helpful indeed.
Yet I still find the message miss leading, but this is another topic :)
I'll close the issue.

closed 2021-02-03 by chaabanemalki

chaabanemalki, 2021-02-03:
@YouDONG-ESP : reading in details the documentation, the message makes sense.
Thank you for your support, everything is clear now