espressif/esp-idf #6463: ESP32-S2 Software Serial __atomic_exchange_1 Not Implemented

kevin192291, 2021-01-25:
### Issue submission guidelines
See [detaled guidelines](http://crosstool-ng.github.io/support/). In short:
- Report crosstool-NG version not sure, But is a new install so whatever is latest.
- Report host OS and its version Linux Mint 20.1
- Attach `build.log` (unless crosstool-NG fails before the build starts)
This is what I have:
```
/home/kevin/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/xtensa-esp32s2-elf/xtensa-esp32s2-elf/include/c++/8.4.0/bits/atomic_base.h:413: undefined reference to `__atomic_exchange_1'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32S2 Dev Module.
```

igrr, 2021-01-25:
Have transferred to esp-idf since the missing functions should go into components/xtensa/stdatomic.c.

seackone, 2021-01-27:
Hi,
I have exactly the same issue as kevin:

```/Users/Name/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc6/tools/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: /Users/Name/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc6/tools/sdk/esp32s2/lib/librtc.a(rtc.o)(.text.rtc_pad_gpio_wakeup+0xa9): could not decode instruction; possible configuration mismatch
/Users/Name/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc6/tools/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: libraries/EspSoftwareSerial/SoftwareSerial.cpp.o:(.literal._ZN14SoftwareSerial6rxBitsEv+0xc): undefined reference to `__atomic_exchange_1'
/Users/Name/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc6/tools/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: libraries/EspSoftwareSerial/SoftwareSerial.cpp.o: in function `SoftwareSerial::rxBits()':
/Users/Name/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc6/tools/xtensa-esp32s2-elf/xtensa-esp32s2-elf/include/c++/8.4.0/bits/atomic_base.h:413: undefined reference to `__atomic_exchange_1'
collect2: error: ld returned 1 exit status
exit status 1
Fehler beim Kompilieren für das Board ESP32S2 Dev Module.
```

And here is the code:

```
#include <SoftwareSerial.h>

SoftwareSerial SoftSerial(32, 33);

void setup() {
}
void loop() {
}
```

Im using the Arduino IDE 1.8.13 as well on a Macbook Pro.

atanisoft, 2021-02-10:
@kevin192291 @seackone Here is a known working implementation that I wrote based on the existing versions for other atomic exchanges:

```
/// Disables interrupts and saves the interrupt enable flag in a register.
#define ACQ_LOCK()                                                             \
    uint32_t _pastlock;                                                        \
    __asm__ __volatile__ ("rsil %0, " XTSTR(XCHAL_EXCM_LEVEL) "\n"             \
                          : "=r"(_pastlock));

/// Restores the interrupte enable flag from a register.
#define REL_LOCK() __asm__ __volatile__ ("memw \n"                             \
                                         "wsr %0, ps\n"                        \
                                        :: "r"(_pastlock));

uint8_t __atomic_exchange_1(uint8_t *ptr, uint8_t val, int memorder)
{
    ACQ_LOCK();
    uint8_t ret = *ptr;
    *ptr = val;
    REL_LOCK();
    return ret;
}
```

@igrr feel free to modify this as needed for future integration.

brunohorta82, 2021-02-12:
@atanisoft you can tell me the file name where I need to put this code?

brunohorta82, 2021-02-12:
> @kevin192291 @seackone Here is a known working implementation that I wrote based on the existing versions for other atomic exchanges:
>
> ```
> /// Disables interrupts and saves the interrupt enable flag [...]
@atanisoft you can tell me the file name where I need to put this code?

atanisoft, 2021-02-12:
@brunohorta82 it can go in any of your code files.

brunohorta82, 2021-02-12:
> @brunohorta82 it can go in any of your code files.

I made this but the error continues.
`/Users/brunohorta/.platformio/packages/toolchain-xtensa32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: /Users/brunohorta/.platformio/packages/framework-arduinoespressif32@src-6f5c87011d02e174693ecf14a727008b/tools/sdk/esp32s2/lib/librtc.a(rtc.o)(.text.rtc_pad_gpio_wakeup+0xa9): could not decode instruction; possible configuration mismatch
/Users/brunohorta/.platformio/packages/toolchain-xtensa32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: .pio/build/esp32-s2/lib3c0/libEspSoftwareSerial.a(SoftwareSerial.cpp.o):(.literal._ZN14SoftwareSerial6rxBitsEv+0x8): undefined reference to `__atomic_exchange_1'
/Users/brunohorta/.platformio/packages/toolchain-xtensa32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: .pio/build/esp32-s2/lib3c0/libEspSoftwareSerial.a(SoftwareSerial.cpp.o): in function `SoftwareSerial::rxBits()':
/Users/brunohorta/.platformio/packages/toolchain-xtensa32s2/xtensa-esp32s2-elf/include/c++/8.4.0/bits/atomic_base.h:463: undefined reference to `__atomic_exchange_1'`

atanisoft, 2021-02-12:
Not sure what to suggest since you are using platformIO and it may be mangling things

closed 2021-04-27 by espressif-bot