espressif/esp-idf #6464: how to specify in a project under CMake to add a new component ?

edbek, 2021-01-25:
Please give me some advice on how to specify in a project under CMake to add a new component (e.g. HAL)  to it, which should be compiled and linked ?

igrr, 2021-01-25:
Hi @edbek, depending on where the additional component is located:

The simplest option is to place the new component under `components` subdirectory of the project. See [examples/peripherals/rmt/led_strip](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/rmt/led_strip), for instance.

You can also add a component located outside of the project directory by setting `EXTRA_COMPONENT_DIRS` variable in the project CMakeLists.txt file. This is used in many examples, e.g. https://github.com/espressif/esp-idf/blob/526f682397a8cfb74698c601fd2c5b30e1433837/examples/protocols/esp_http_client/CMakeLists.txt#L7. The documentation is [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#optional-project-variables).

When adding a new component, you will also likely need to specify dependencies between components. See [Component Requirements](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#component-requirements) section of the docs for details.

edbek, 2021-01-26:
Thanks for the answer, but it doesn't help yet. I'm a beginner and I don't understand the principles of building projects at all.

I am running some example ("..\esp-idf-v4.2\examples\protocols\http_server\restful_server" app). Next, I expand the functionality of this application, and I need to supplement it with existing components (from the "..\esp-idf-v4.2\components" directory), more precisely, libraries (for example, **libdriver** from "..\esp-idf-v4.2\components\driver").
How can I force CMake to compile the additional component I need and then link the application with the resulting libraries ?

First, it’s not clear what set of components the "restful_server" application picks up by default? Where can you see it?
Secondly, how to add a new component to the project not as source files, but so that the component is compiled separately and, as a result, its library is used?

igrr, 2021-01-26:
By default, the build system will look for components in the following locations ([doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#optional-project-variables)):

* IDF built-in components: ${IDF_PATH}/components
* Project components: {PROJECT_DIR}/components, 
* Extra directories specified by ${EXTRA_COMPONENT_DIRS}

So the "driver" component from ESP-IDF will be built by default.

The list of components is logged by the build system when the project is first built. You can run `idf.py reconfigure` to re-run CMake, the list of components will be part of the output. For example:
```
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_local_ctrl esp_netif esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs flash_openocd freemodbus freertos hal heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
```

> Secondly, how to add a new component to the project not as source files, but so that the component is compiled separately and, as a result, its library is used?

To link an externally-built static library, please refer to [this example](https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/import_prebuilt) and [the docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#using-prebuilt-libraries-with-components).

We don't provide facilities for compiling an ESP-IDF component "separately" at the moment (*). However, you can take a simple project (hello-world), include the component into that project, build the project, and find the static library under the `build` directory. You can also build a static library using some other build system, especially if this is a third-party library.

---
(*) The reason why there is no supported way to build a component into a static library "separately" from a project is that ESP-IDF components often rely on project configuration (sdkconfig.h). These configuration files are defined at project level. So the only supported build type is building a project.

edbek, 2021-01-30:
You write "However, you can take a simple project (hello-world), include the component into that project, build the project, and find the static library under the build directory."

1. How can I include a component in a project? Give a hint. For example, "driver".
2. As I understand it, there should be a library "libdriver.a" in the "build/esp-idf/driver" directory. But it also needs to be linked to the project. How to do it ?

igrr, 2021-01-30:
As mentioned previously, `driver` is a built-in ESP-IDF component, and as such it will be built and linked into the project by default.

edbek, 2021-01-31:
Yes, I see, thanks.

I still have a question: I want to take a project "esp-at" and expand it by linking it to components from "esp-idf" (for example, https-request). And I would like to fully work in Eclipse.

Way 1: I can take the project itself as a basis and start expanding it. But then I won't be able to work in Eclipse.
Way 2: I can take the "https-request" example-project from "esp-idf" as a basis and add the "libesp32s2_at_core.a" library from "esp-at" to it. Actually, I tried to do this, but when linking, errors appear: for some reason, the "libesp32s2_at_core.a" library does not see functions from component libraries from "esp-idf".

Could you give some advice on which way to solve my problem is correct or are there more solutions?
Thanks in advance.

igrr, 2021-02-01:
For option 1, I got it working in Eclipse after a bit of poking at the [esp-at build script](https://github.com/espressif/esp-at/blob/master/build.py):

* Clone the version of ESP-IDF [supported by esp-at](https://github.com/espressif/esp-at/blob/52d99727dc020e5d8123e65120cb63981c44b6be/module_config/module_esp32s2_default/IDF_VERSION#L2) into `esp-idf` subdirectory of `esp-at` project. This step you probably need to do outside of Eclipse.
* Import esp-at project into your Eclipse workspace if it is not there yet. If importing, choose to have the project in its existing location, without copying it.
* Fix a typo in esp-at (ESP_AT_PORJECT_PATH -> ESP_AT_PROJECT_PATH):
   ```diff
   diff --git a/main/Kconfig b/main/Kconfig
   index 954098e..64cb3ea 100644
   --- a/main/Kconfig
   +++ b/main/Kconfig
   @@ -9,10 +9,10 @@ choice AT_COMMUNICATION_METHOD
        default AT_BASE_ON_UART
        depends on AT_ENABLE
    
   -source "$ESP_AT_PORJECT_PATH/main/interface/uart/Kconfig"
   -source "$ESP_AT_PORJECT_PATH/main/interface/hspi/Kconfig"
   -source "$ESP_AT_PORJECT_PATH/main/interface/socket/Kconfig"
   -source "$ESP_AT_PORJECT_PATH/main/interface/sdio/Kconfig"
   +source "$ESP_AT_PROJECT_PATH/main/interface/uart/Kconfig"
   +source "$ESP_AT_PROJECT_PATH/main/interface/hspi/Kconfig"
   +source "$ESP_AT_PROJECT_PATH/main/interface/socket/Kconfig"
   +source "$ESP_AT_PROJECT_PATH/main/interface/sdio/Kconfig"
 
    endchoice
   ```

* Add a few environment variables in Eclipse workspace preferences (C++ > Build > Environment):
  - ESP_AT_PROJECT_PATH: <absolute path of the esp-at project>
  - ESP_AT_MODULE_NAME: WROOM
  - ESP_AT_PROJECT_PLATFORM: PLATFORM_ESP32S2

Once this is done, you should be able to build the project as any other project in Eclipse (set the launch target, etc.)

Note that I haven't tried these steps on Windows. ESP-AT project has only recently [fixed the issues with compiling on Winodws](https://github.com/espressif/esp-at/commit/be1e2172bfdc4fa5071a279b6c7b934148a6f9f5), so it is possible that some problems still remain.

Regarding the option 2, since libesp32s2_at_core.a depends on IDF components, you may need to specify the list of IDF components it depends on, along these lines:

https://github.com/espressif/esp-idf/blob/6e776946d01ec0d081d09000c36d23ec1d318c06/examples/build_system/cmake/import_prebuilt/main/CMakeLists.txt#L6-L7

If you get linking errors, check which component the function is defined in, and add that component to the list of requirements.

edbek, 2021-02-02:
For option 2 аn addition such as:

```
add_prebuilt_library(prebuilt "libesp32s2_at_core.a"
				 PRIV_REQUIRES wpa_supplicant esp32s2 driver lwip soc soc_esp32s2 smartconfig)
target_link_libraries(${COMPONENT_LIB} PRIVATE prebuilt)
```

gives an error message:

```
CMake Error at C:/Users/user/Desktop/esp-idf/tools/cmake/component.cmake:370 (__component_get_property):
  __component_get_property Function invoked with incorrect arguments for
  function named: __component_get_property
Call Stack (most recent call first):
  C:/Users/user/Desktop/esp-idf/tools/cmake/utilities.cmake:306 (idf_component_get_property)
  main/CMakeLists.txt:17 (add_prebuilt_library)

CMake Error at C:/Users/user/Desktop/esp-idf/tools/cmake/component.cmake:370 (__component_get_property):
  __component_get_property Function invoked with incorrect arguments for
  function named: __component_get_property
Call Stack (most recent call first):
  C:/Users/user/Desktop/esp-idf/tools/cmake/utilities.cmake:306 (idf_component_get_property)
  main/CMakeLists.txt:17 (add_prebuilt_library)

```

igrr, 2021-02-02:
This confusing error message probably means that one of the components listed in PRIV_REQUIRES hasn't been found in idf_component_get_property. Can you try removing `soc_esp32s2` from the list? It is sufficient to declare dependency on `soc` component, it will internally call `target_link_libraries` to add `libsoc_esp32s2.a`.

edbek, 2021-02-02:
For option 1:

work under Windows 10
I imported the esp-at project as "Espressif -> Existing IDF Project".
Injected all environment variables.
But unfortunately, the process does not go further because the error is thrown out:

```
20:21:33 **** Build of configuration Default for project esp-at ****
make all 
Cannot run program "make": Launching failed
Error: Program "make" not found in PATH
```

edbek, 2021-02-02:
Yes, option 2 worked: I had to remove two dependencies: (`soc_esp32s2` as you said) and another `smartconfig`.

The build went smoothly. But why these two dependencies had to be removed - it remains a mystery.

igrr, 2021-02-02:
Probably because neither of the two are IDF component names. Adding `esp_wifi` and `soc` as component dependencies also has the effect of linking all the underlying libraries, in this case `libsmartconfig.a` and `libsoc_esp32s2.a`.

projectgus, 2021-02-09:
Hi @edbek,

It looks like the issue you were experienced is resolved, is there anything you suggest we change or better document in ESP-IDF to better accommodate your use case?

edbek, 2021-02-09:
It remains a problem to synthesize several disparate projects (esp- *) into one project in Eclipse. Of course, this is difficult to do now, but perhaps it is worth thinking about it in the future.
Thanks for your hard work.

ESP-Marius, 2022-08-31:
Closing this one as the original issue has been solved. Feel free open a new ticket if there if you have any specific feature requests with regards to synthesize several disparate projects (esp- *) into one project in Eclipse.

closed 2022-08-31 by ESP-Marius