espressif/esp-idf #6465: CLIENT key Error in Mutual Authentication

sowmimalu, 2021-01-25:
hai ,
 I am using esp idf 4.1 trying to connetion connect my esp32 device with mutual authentication client key password protected.

uint8_t key_pass[4] = "abcd";
 const esp_mqtt_client_config_t mqtt_cfg = {
       
	   .uri =  "mqtts://xswrt.com:8885",  
        .event_handle = mqtt_event_handler,
	.cert_pem = (const char *)iot_eclipse_org_pem_start,
        .client_cert_pem = (const char *)client_cert_pem_start,
        .client_key_pem = (const char *)client_key_pem_start,
		.clientkey_password = (const char *)key_pass,
		.clientkey_password_len = 4
    };
. i am getting below error.
E (63830) esp-tls-mbedtls: mbedtls_pk_parse_keyfile returned -0x1f00
E (63840) esp-tls-mbedtls: Failed to set client pki context
E (63840) esp-tls-mbedtls: Failed to set client configurations
E (63850) esp-tls: create_ssl_handle failed

david-cermak, 2021-01-26:
@sowmimalu The error code of `0x1F00` means `MBEDTLS_ERR_PKCS12_FEATURE_UNAVAILABLE`, so that your key is signed with unsupported encryption scheme.

As for the supported ciphers (by default) you can check the esp-config of mbedTLS here:
https://github.com/espressif/esp-idf/blob/master/components/mbedtls/port/include/mbedtls/esp_config.h#L241

david-cermak, 2021-02-02:
@sowmimalu FYI: just to check the actual algorithm, you can run:
```
openssl rsa -in client_key.pem -text -noout
```

which tells you something like
```
Enter pass phrase for client_key.pem:
RSA Private-Key: (2048 bit, 2 primes)
modulus:
    00:c9:a7:52:a5:19......
    .....
....
```

mahavirj, 2021-02-22:
@sowmimalu Can we close this issue?

mahavirj, 2021-02-26:
Closing due to lack of update, please feel free to reopen if there are any updates.

closed 2021-02-26 by mahavirj