espressif/esp-idf #6466: Modbus master failed to initialize after called esp_restart() or program dump.

NguyenMinhTri, 2021-01-26:
## Environment

- Development Kit:      [ESP32-Wroom-32-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- IDF version: ESP-IDF v4.3-dev-2398-g2bfdd036b
- Compiler version : 8.4.0
- Operating System:     [macOS]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description
Modbus master failed to initialize after called esp_restart() or program dump.

### Steps to reproduce

1. Power plug for ESP32-WROOM
2. Modbus master is initialized.
3. Modbus master is running successfully.
4. Call esp_restart().
5. Modbus master failed to initialize as below.

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
    esp_err_t err = mbc_master_init(MB_PORT_SERIAL_MASTER, &master_handler);
    MASTER_CHECK((master_handler != NULL), ESP_ERR_INVALID_STATE,
                 "mb controller initialization fail.");
    MASTER_CHECK((err == ESP_OK), ESP_ERR_INVALID_STATE,
                 "mb controller initialization fail, returns(0x%x).",
                 (uint32_t)err);
    err = mbc_master_setup((void *)&comm);
    MASTER_CHECK((err == ESP_OK), ESP_ERR_INVALID_STATE,
                 "mb controller setup fail, returns(0x%x).",
                 (uint32_t)err);

```

## Debug Logs

```
D (11331) MB_PORT_COMMON: vMBPortSetMode: Port enter critical.
D (11331) MB_PORT_COMMON: vMBPortSetMode: Port exit critical
D (11331) MB_PORT_COMMON: eMBMasterRTUInit: Port enter critical.
I (11341) uart: queue free spaces: 20
D (11341) intr_alloc: Connected src 36 to int 3 (cpu 1)
D (11351) MB_PORT_COMMON: xMBMasterPortSerialInit Init serial.
/Users/nguy1nt5i/eiptrsa-ldc: Cmponcnts/fre 1tos/ int 4 (cpu .c:[31 (D (e1t61)upBePBiTs)- aON:reMfailed!R

abort() was called at PC 0x4008b77f on core 0
0x4008b77f: xEventGroupSetBits at /Users/nguyentri/esp/esp-idf/components/freertos/event_groups.c:531 (discriminator 1)

Backtrace:0x40088c17:0x3ffc7c60 0x40089421:0x3ffc7c80 0x400904e2:0x3ffc7ca0 0x4008b77f:0x3ffc7d10 0x4008b859:0x3ffc7d30 0x4008bb79:0x3ffc7d50 0x4008bc87:0x3ffc7d90 0x4008bd89:0x3ffc7dc0
0x40088c17: panic_abort at /Users/nguyentri/esp/esp-idf/components/esp_system/panic.c:356

0x40089421: esp_system_abort at /Users/nguyentri/esp/esp-idf/components/esp_system/system_api.c:108

0x400904e2: abort at /Users/nguyentri/esp/esp-idf/components/newlib/abort.c:46

0x4008b77f: xEventGroupSetBits at /Users/nguyentri/esp/esp-idf/components/freertos/event_groups.c:531 (discriminator 1)

0x4008b859: vEventGroupSetBitsCallback at /Users/nguyentri/esp/esp-idf/components/freertos/event_groups.c:661

0x4008bb79: prvProcessReceivedCommands at /Users/nguyentri/esp/esp-idf/components/freertos/timers.c:731 (discriminator 2)

0x4008bc87: prvTimerTask at /Users/nguyentri/esp/esp-idf/components/freertos/timers.c:558 (discriminator 1)

0x4008bd89: vPortTaskWrapper at /Users/nguyentri/esp/esp-idf/components/freertos/port/xtensa/port.c:168

```

alisitsyn, 2021-01-26:
Hello @NguyenMinhTri,

Thanks for issue. I was trying to reproduce it on my system on your commit 2bfdd036b2dbd07004c8e4f2ffc87c728819b737. I used modified modbus serial master example to reproduce the issue:
``` cpp
void app_main(void)
{
    // Initialization of device peripheral and objects
    ESP_ERROR_CHECK(master_init());
    vTaskDelay(10);
    
    xTaskCreate(&master_operation_func, "modbus_task", 16384, NULL, 5, NULL);
    vTaskDelay(300);
    esp_restart();

    master_operation_func(NULL);
}
```
Am I reproduce your conditions correctly? As you can see in the log below the modbus initialized, started communication then esp_restart() called and restarts processor and it is able to start properly again and read and write parameters. 
[esp_restart_after_normal_start_modbus.log](https://github.com/espressif/esp-idf/files/5873877/esp_restart_after_normal_start_modbus.log)

According to this I have some questions:
1. Which modbus slave device you are using with your master application?
2. Could you send me the example which can easily reproduce your issue?
3. Could you describe more conditions that would help to reproduce and then fix the issue?
Thanks.

NguyenMinhTri, 2021-01-27:
Hi @alisitsyn, sorry for missing information. I have updated new code as below.
- The master_init() function is initialized after the SYSTEM_EVENT_STA_GOT_IP event is activated.  
### Code to reproduce this issue

```cpp
static esp_err_t wifi_event_handler(void *ctx, system_event_t *event)
{
    switch (event->event_id) {
    case SYSTEM_EVENT_STA_START:
        esp_wifi_connect();
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
        xEventGroupSetBits(wifi_event_group, CONNECTED_BIT);
        // Initialization of device peripheral and objects
        ESP_ERROR_CHECK(master_init());
        xTaskCreate(&master_operation_func, "modbus_task", 16384, NULL, 5, NULL);
        vTaskDelay(300);
        esp_restart();
        vTaskDelay(10);
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        esp_wifi_connect();
        xEventGroupClearBits(wifi_event_group, CONNECTED_BIT);
        break;
    default:
        break;
    }
    return ESP_OK;
}

static void wifi_init(void)
{
    tcpip_adapter_init();
    wifi_event_group = xEventGroupCreate();
    ESP_ERROR_CHECK(esp_event_loop_init(wifi_event_handler, NULL));
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = CONFIG_ESP_WIFI_SSID,
            .password = CONFIG_ESP_WIFI_PASSWORD,
        },
    };
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config));
    ESP_LOGI(TAG, "start the WIFI SSID:[%s]", CONFIG_ESP_WIFI_SSID);
    ESP_ERROR_CHECK(esp_wifi_start());
    ESP_LOGI(TAG, "Waiting for wifi");
    xEventGroupWaitBits(wifi_event_group, CONNECTED_BIT, false, true, portMAX_DELAY);
}

void app_main(void)
{
        //Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);
    ESP_LOGI(TAG, "ESP_WIFI_MODE_STA");
    wifi_init();

}

```

NguyenMinhTri, 2021-01-27:
Hi @alisitsyn,  I have tried with your code. Sometime, this issues is found as below. Please check debug log.
## Environment

- Development Kit:      [ESP32-Wroom-32-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- IDF version: ESP-IDF v4.3-dev-2398-g2bfdd036b
- Compiler version : 8.4.0
- Operating System:     [macOS]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description
Modbus master failed to initialize after called esp_restart() or program dump.

### Code to reproduce this issue
sdkconfig.h
```cpp
/*
 * Automatically generated file. DO NOT EDIT.
 * Espressif IoT Development Framework (ESP-IDF) Configuration Header
 */
#pragma once
#define CONFIG_IDF_CMAKE 1
#define CONFIG_IDF_TARGET_ARCH_XTENSA 1
#define CONFIG_IDF_TARGET "esp32"
#define CONFIG_IDF_TARGET_ESP32 1
#define CONFIG_IDF_FIRMWARE_CHIP_ID 0x0000
#define CONFIG_SDK_TOOLPREFIX "xtensa-esp32-elf-"
#define CONFIG_APP_BUILD_TYPE_APP_2NDBOOT 1
#define CONFIG_APP_BUILD_GENERATE_BINARIES 1
#define CONFIG_APP_BUILD_BOOTLOADER 1
#define CONFIG_APP_BUILD_USE_FLASH_SECTIONS 1
#define CONFIG_APP_COMPILE_TIME_DATE 1
#define CONFIG_APP_RETRIEVE_LEN_ELF_SHA 16
#define CONFIG_BOOTLOADER_OFFSET_IN_FLASH 0x1000
#define CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE 1
#define CONFIG_BOOTLOADER_LOG_LEVEL_INFO 1
#define CONFIG_BOOTLOADER_LOG_LEVEL 3
#define CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V 1
#define CONFIG_BOOTLOADER_WDT_ENABLE 1
#define CONFIG_BOOTLOADER_WDT_TIME_MS 9000
#define CONFIG_BOOTLOADER_RESERVE_RTC_SIZE 0x0
#define CONFIG_ESPTOOLPY_BAUD_OTHER_VAL 115200
#define CONFIG_ESPTOOLPY_FLASHMODE_DIO 1
#define CONFIG_ESPTOOLPY_FLASHMODE "dio"
#define CONFIG_ESPTOOLPY_FLASHFREQ_40M 1
#define CONFIG_ESPTOOLPY_FLASHFREQ "40m"
#define CONFIG_ESPTOOLPY_FLASHSIZE_4MB 1
#define CONFIG_ESPTOOLPY_FLASHSIZE "4MB"
#define CONFIG_ESPTOOLPY_FLASHSIZE_DETECT 1
#define CONFIG_ESPTOOLPY_BEFORE_RESET 1
#define CONFIG_ESPTOOLPY_BEFORE "default_reset"
#define CONFIG_ESPTOOLPY_AFTER_RESET 1
#define CONFIG_ESPTOOLPY_AFTER "hard_reset"
#define CONFIG_ESPTOOLPY_MONITOR_BAUD_115200B 1
#define CONFIG_ESPTOOLPY_MONITOR_BAUD_OTHER_VAL 115200
#define CONFIG_ESPTOOLPY_MONITOR_BAUD 115200
#define CONFIG_PARTITION_TABLE_TWO_OTA 1
#define CONFIG_PARTITION_TABLE_CUSTOM_FILENAME "partitions.csv"
#define CONFIG_PARTITION_TABLE_FILENAME "partitions_two_ota.csv"
#define CONFIG_PARTITION_TABLE_OFFSET 0x8000
#define CONFIG_PARTITION_TABLE_MD5 1
#define CONFIG_ESP_WIFI_SSID "myssid"
#define CONFIG_ESP_WIFI_PASSWORD "mypassword"
#define CONFIG_OUTPUT_GPIO 27
#define CONFIG_MB_UART_PORT_NUM 2
#define CONFIG_MB_UART_BAUD_RATE 9600
#define CONFIG_MB_UART_RXD 5
#define CONFIG_MB_UART_TXD 4
#define CONFIG_MB_UART_RTS 16
[... 2152 lines cut ...]
I (14515) MASTER_TEST: Characteristic #2 Temperature_1 (C) value = 0.000000 (0x0) read successful.
I (14595) MASTER_TEST: Characteristic #3 Humidity_2 (%rH) value = 0.000000 (0x0) read successful.
I (14675) MASTER_TEST: Characteristic #4 Temperature_2 (C) value = 0.000000 (0x0) read successful.
I (14705) wifi:state: run -> init (0)
I (14705) wifi:pm stop, total sleep time: 10002274 us / 10658783 us

I (14705) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (14735) wifi:flush txq
I (14735) wifi:stop sw txq
I (14735) wifi:lmac stop hw txq
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6972
load:0x40078000,len:14332
ho 0 tail 12 room 4
load:0x40080400,len:3672
0x40080400: _init at ??:?

entry 0x40080678
```

alisitsyn, 2021-01-28:
Hi @NguyenMinhTri, 

Thank you for detailed report. I will let you know if found something new.
Thanks.

alisitsyn, 2021-01-28:
@NguyenMinhTri,

Thank you again for this issue. I was able to reproduce it and find the solution. Here are some information suggestions:

1. In general, calling esp_restart should work both in task or even wifi callback. However it is not recommended doing complicated stuff in the wifi callback functions like Modbus initialization, start and so on. If the stack sufficient you can do more complex things there and it is not the reason for the issue but it is still not good approach.

I would propose to do it as in the code below:
```
void app_main(void)
{
    //Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);
    ESP_LOGI(MASTER_TAG, "ESP_WIFI_MODE_STA");
    wifi_init();
    ESP_ERROR_CHECK(master_init());

    xTaskCreate(&master_operation_func, "modbus_task", 16384, NULL, 5, operation_task_handle);
    vTaskDelay(300);
    esp_restart();
```
2. The Modbus uses timer groups to measure some related timeouts. I think due to a bug in the esp_restart_noos() this function does not reset the group timer peripheral correctly and causes the issue after restart.

Please apply the patch below to try to fix this issue. 

[0001_fix_esp_restart_does_not_reset_timer_groups_periph.diff](https://github.com/espressif/esp-idf/files/5888351/0001_fix_esp_restart_does_not_reset_timer_groups_periph.diff.log)

Please let me know if it works for your project. 
Thanks.

NguyenMinhTri, 2021-01-29:
Hi @alisitsyn, thank you for supporting.

alisitsyn, 2021-02-08:
Hi @NguyenMinhTri,

Thanks for update. The MR with this fix is under review. The issue will be closed once the MR is merged.

AxelLin, 2022-02-17:
v4.3: e48f87468e290c8ec485d134614e3d77639e9cb5

Alvin1Zhang, 2022-02-17:
Thanks for sharing the updates, feel free to reopen if the issue happens again. Thanks.

closed 2022-02-17 by Alvin1Zhang