espressif/esp-idf #6467: Forced to use static wifi buffers together with SRAM even though PSRAM couldn't be detected

persan666, 2021-01-26:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROVER-B]
- IDF version (run ``git describe --tags`` to find it): 
    // v4.2-238-g8cd16b60f
- Build System:         [CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [Yes (Eclipse)]
- Power Supply:         [Battery]

## Problem Description

I am not able to set dynamic Wifi buffers when PSRAM is enabled.
I using two different ESP modules (ESP32-WROOM-32 and ESP32-WROVER-B) with the same application and has enabled SPIRAM_IGNORE_NOTFOUND.
The problem arrives when using the ESP32-WROOM-32 and when the PSRAM is not detected. This causes static allocation of the Wifi buffers and may application runs out of heap.
From the documentation it tells me that static buffer is recommended but not mandatory when using PSRAM. But the option for choosing dynamic buffers is not available. 

Eclipse Espressif plugin versions:

* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system
  - 1.10.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

### Expected Behavior
Possibility to choose dynamic Wifi buffers when PSRAM is enabled.
### Actual Behavior
Unable to choose dynamic Wifi buffers when PSRAM is enabled.
### Steps to reproduce
In SDK Configurator:
1. ESP32-Specific - Support for external, SPI-connected RAM.
2. Wi-Fi - Type of WiFi TX buffers - the only option is Static 

[image]

YouDONG-ESP, 2021-01-27:
Hi @persan666, In PSRAM mode, the CONFIG_ESP32_WIFI_TX_BUFFER is forced to be static.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html?#using-psram

negativekelvin, 2021-01-27:
@YouDONG-ESP it conflicts with SPIRAM_USE_MALLOC but why is this a conflict since it should be allocating from internal reserved pool using malloc_internal_wrapper?

 https://github.com/espressif/esp-idf/blob/526f682397a8cfb74698c601fd2c5b30e1433837/components/esp_wifi/Kconfig#L68-L70

YouDONG-ESP, 2021-01-27:
@negativekelvin Since the TX buffer must be DMA buffer, so when PSRAM is enabled, the TX buffer must be static.

negativekelvin, 2021-01-27:
@YouDONG-ESP that's what malloc_internal_wrapper and SPIRAM_MALLOC_RESERVE_INTERNAL is for

persan666, 2021-01-27:
Is there a way to get around this when no PSRAM is detected. Otherwise I will have to have two different applications for the different modules. If I remember correctly esp-idf 3.3 didn't have this limitation.

YouDONG-ESP, 2021-01-27:
@negativekelvin yeah you are right, didn't notice that. We considered PSRAM will provide enough memory for static buffer, so we don't need to make it dynamic.

persan666, 2021-01-28:
For me this is not a problem any more. I have changed the 'SPI RAM access method' to 'Make RAM allocatable using heap_caps_malloc(,,MALLOC_CAP_SPIRAM)'. But I still think it should be possible to select Dynamic buffer in case of that the SPIRAM could not be detected.

closed 2021-01-28 by persan666