espressif/esp-idf #6470: I2S to Internal DAC is Broken in 4.1/4.2

ducalex, 2021-01-27:
## Environment

- Development Kit: None
- Kit version:  None
- Module or chip used:  ESP32-WROVER-IB
- IDF version: v4.1
- Build System: idf.py (cmake)
- Compiler version: (crosstool-NG esp-2020r2) 8.2.0
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt|Plain Command Prompt
- Using an IDE?: No
- Power Supply: USB

## Problem Description

I2S to internal DAC produces no audio. 
- Code is working on 3.1, 3.2, 3.3.4, and 4.0.2. 
- Code is not working on 3.3.0, 4.0, 4.1 and 4.2. 
- Code works on 4.1 if I backport the i2s.c from 4.0.2.

### Expected Behavior

Audio to be playing through GPIO 25 and 26 or at least an error.

### Actual Behavior

No audio, no error.

### Steps to reproduce

The issue was fixed in 3.3.1 and 4.0.1 I believe, but the fix never made its way to 4.1 (and possibly 4.2).

Please refer to these two forums issues that describe the problem and provide a code example to reproduce:
- https://www.esp32.com/viewtopic.php?t=14309
- https://www.esp32.com/viewtopic.php?t=14029

## Debug Logs

None, there is no special error or warning or debug.

## Other items if possible

cfint, 2022-02-03:
I traced the breaking change to this commit 8c76a3c10dd0e40cbfb0becf83a6b84506279edb. With the following change applied to `v4.1`, the A2DP audio sink with internal DAC works again.

```
diff --git a/components/soc/src/hal/i2s_hal.c b/components/soc/src/hal/i2s_hal.c
index 6639525ea8..e21856bd34 100644
--- a/components/soc/src/hal/i2s_hal.c
+++ b/components/soc/src/hal/i2s_hal.c
@@ -194,7 +194,7 @@ void i2s_hal_config_param(i2s_hal_context_t *hal, const i2s_config_t *i2s_config
     i2s_ll_set_rx_pdm_en(hal->dev, 0);
     i2s_ll_set_tx_pdm_en(hal->dev, 0);
 #endif
-    if (i2s_config->communication_format & I2S_COMM_FORMAT_I2S || i2s_config->communication_format & I2S_COMM_FORMAT_I2S_MSB || i2s_config->communication_format & I2S_COMM_FORMAT_I2S_LSB) {
+    if (i2s_config->communication_format & I2S_COMM_FORMAT_I2S|| i2s_config->communication_format & I2S_COMM_FORMAT_I2S_LSB) {
         i2s_ll_set_tx_short_sync(hal->dev, 0);
         i2s_ll_set_rx_short_sync(hal->dev, 0);
         i2s_ll_set_tx_msb_shift(hal->dev, 1);
```

karawin, 2022-02-21:
Not playing on 4.4
A pity!

L-KAYA, 2022-03-30:
> Not playing on 4.4 A pity!

@karawin On v4.4, setting communication mode to `I2S_COMM_FORMAT_STAND_MSB` instead of `I2S_COMM_FORMAT_STAND_I2S` might help.

For the failure on v4.1, there will be a fix soon.