espressif/esp-idf #6472: Related to Amazon Freertos

Romit188, 2021-01-28:
In file included from /home/ar-cpu-35/amazon-freertos/vendors/espressif/boards/ports/ble/iot_ble_hal_common_gap.c:29:
/home/ar-cpu-35/amazon-freertos/vendors/espressif/boards/ports/ble/nimble/iot_ble_hal_common_gap.c: In function 'prvGAPeventHandler':
/home/ar-cpu-35/amazon-freertos/vendors/espressif/boards/ports/ble/nimble/iot_ble_hal_common_gap.c:256:14: error: 'BLE_GAP_EVENT_PAIRING_REQUEST' undeclared (first use in this function); did you mean 'BLE_GAP_REPEAT_PAIRING_RETRY'?
         case BLE_GAP_EVENT_PAIRING_REQUEST:
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
              BLE_GAP_REPEAT_PAIRING_RETRY
/home/ar-cpu-35/amazon-freertos/vendors/espressif/boards/ports/ble/nimble/iot_ble_hal_common_gap.c:256:14: note: each undeclared identifier is reported only once for each function it appears in
/home/ar-cpu-35/amazon-freertos/vendors/espressif/boards/ports/ble/nimble/iot_ble_hal_common_gap.c:260:46: error: 'struct ble_gap_event' has no member named 'pairing_req'
                 rc = ble_gap_conn_find( event->pairing_req.conn_handle, &desc );
                                              ^~
/home/ar-cpu-35/amazon-freertos/vendors/espressif/boards/ports/ble/nimble/iot_ble_hal_common_gap.c:268:44: error: 'struct ble_gap_event' has no member named 'pairing_req'
             if( ble_hs_cfg.sm_sc && ( event->pairing_req.authreq & BLE_SM_PAIR_AUTHREQ_SC ) )
                                            ^~
/home/ar-cpu-35/amazon-freertos/vendors/espressif/boards/ports/ble/nimble/iot_ble_hal_common_gap.c:270:65: error: 'struct ble_gap_event' has no member named 'pairing_req'
                 if( ble_sm_sc_ioa[ ble_hs_cfg.sm_io_cap ][ event->pairing_req.io_cap ] == BLE_SM_IOACT_NONE )
                                                                 ^~
/home/ar-cpu-35/amazon-freertos/vendors/espressif/boards/ports/ble/nimble/iot_ble_hal_common_gap.c:284:31: error: 'struct ble_gap_event' has no member named 'pairing_req'
                 else if( event->pairing_req.max_enc_key_size < BLE_SM_PAIR_KEY_SZ_MAX )
                               ^~
CMakeFiles/afr_ble_hal.dir/build.make:81: recipe for target 'CMakeFiles/afr_ble_hal.dir/vendors/espressif/boards/ports/ble/iot_ble_hal_common_gap.c.obj' failed
make[2]: *** [CMakeFiles/afr_ble_hal.dir/vendors/espressif/boards/ports/ble/iot_ble_hal_common_gap.c.obj] Error 1
make[2]: *** Waiting for unfinished jobs....

When i build my build directory using make all -j4 command , i am getting error how t osolve this error?

mahavirj, 2021-01-28:
@Romit188 Please file issues related to Amazon FreeRTOS [here](https://github.com/aws/amazon-freertos/issues) along with information per issue template there. I will close the issue from here, as it is unrelated to ESP-IDF project.

closed 2021-01-28 by mahavirj

Romit188, 2021-01-28:
Okay mahavirji, Sorry about it.