espressif/esp-idf #6473: Enabling WIFI_PROMIS_FILTER_MASK_FCSFAIL does not work

doragasu, 2021-01-28:
## Environment

    Development Kit: ESP-WROOM-32 NodeMCU
    Module or chip used: ESP32-WROOM-32
    IDF version: master branch, commit: 526f682397a8cfb74698c601fd2c5b30e1433837
    Build System: idf.py
    Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
    Operating System: Linux
    Using an IDE?: No
    Power Supply: USB

## Problem Description

When using the sniffer API, enabling the filter `WIFI_PROMIS_FILTER_MASK_FCSFAIL` should allow receiving frames with failed FCS check. But it seems this filter is not working, only packets with correct FCS are received.

### Expected Behavior

When enabling `WIFI_PROMIS_FILTER_MASK_FCSFAIL` and `WIFI_PROMIS_FILTER_MASK_DATA` filter masks, the promiscuous RX callback is invoked when data packets with failed FCS check are received.

### Actual Behavior

After enabling both `WIFI_PROMIS_FILTER_MASK_FCSFAIL` and `WIFI_PROMIS_FILTER_MASK_DATA` filter masks, the promiscuous RX callback is only invoked for data packets with correct FCS. This callback is not invoked for frames with failed FCS check.

### Steps to reproduce

You need the following setup:

a) ESP32 device flashed with the code pasted below, to capture frames with failed FCS on channel 7.
b) Device that can transmit frames with invalid FCS on channel 7.
c) Device able to capture packets with failed FCS on channel 7.

For (c), we use a computer with a wifi radio supporting monitor mode and FCS failed data reception (for example a Ralink 5370 based USB dongle). In this computer we enable monitor mode and FCS frames data reception as follows (this assumes the device interface is `wlan0`, change as needed):

```bash
# ip link set wlan0 down
# iwconfig mon0 channel 7
# iw dev mon0 set monitor fcsfail
```
Then on Wireshark, we start capturing on channel 7, and use the following filter to display only frames with failed FCS:

```
wlan.fcs.status == "Bad"
```

Once the setup is ready, we test the feature as follows:

1. Start the ESP32 sniffer (a).
2. From the device (b), send frames with incorrect FCS.
3. Check the output of the ESP32 sniffer program (a) and the computer capturing frames (c). Both should show the frames with failed FCS.

### Code to reproduce this issue

```c
#include <string.h>
#include "esp_wifi.h"
#include "esp_log.h"
#include "esp_event.h"
#include "nvs_flash.h"

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"

#define TAG "sniffer"

#define SNIFFER_DATA_LEN 256

static EventGroupHandle_t wifi_event_group;

static const int START_BIT = BIT0;

static char printbuf[100];

static void sniffer_cb(void* buf, wifi_promiscuous_pkt_type_t type)
{
	wifi_promiscuous_pkt_t* rx = (wifi_promiscuous_pkt_t*)buf;
	int len = rx->rx_ctrl.sig_len;
	int i;

	switch (type) {
		case WIFI_PKT_DATA:
			ESP_LOGI(TAG, "Rx data pkt len:%d", len);
			break;

		default:
			ESP_LOGI(TAG, "Not parsing type %d", type);
			return;
	}

	ESP_LOGI(TAG, "Data:");

	len = len > SNIFFER_DATA_LEN ? SNIFFER_DATA_LEN : len;

	for (i = 0; i < len; i++) {
		sprintf(printbuf + (i % 16) * 3, "%02x ", rx->payload[i]);

		if ((i + 1) % 16 == 0) {
			ESP_LOGI(TAG, "  - %s", printbuf);
		}
	}

	if ((i % 16) != 0) {
		printbuf[((i) % 16) * 3 - 1] = 0;
		ESP_LOGI(TAG, "  - %s", printbuf);
	}
}

static void sniffer_task(void* pvParameters)
{
	wifi_promiscuous_filter_t sniffer_filter = {0};

	sniffer_filter.filter_mask |= WIFI_PROMIS_FILTER_MASK_DATA;
	sniffer_filter.filter_mask |= WIFI_PROMIS_FILTER_MASK_FCSFAIL;

	xEventGroupWaitBits(wifi_event_group, START_BIT,
			false, true, portMAX_DELAY);
	// Use channel 7
	ESP_ERROR_CHECK(esp_wifi_set_channel(7, 0));
	ESP_ERROR_CHECK(esp_wifi_set_promiscuous_rx_cb(sniffer_cb));
	ESP_ERROR_CHECK(esp_wifi_set_promiscuous_filter(&sniffer_filter));
	ESP_ERROR_CHECK(esp_wifi_set_promiscuous(true));
	vTaskDelete(NULL);
}

static void event_handler(void* arg, esp_event_base_t event_base,
		int32_t event_id, void* event_data)
{
	switch (event_id) {
		case WIFI_EVENT_STA_START:
			xEventGroupSetBits(wifi_event_group, START_BIT);
			break;

		default:
			break;
	}
}

static void initialise_wifi(void)
{
	esp_event_handler_instance_t instance_any_id;

	wifi_event_group = xEventGroupCreate();
	ESP_ERROR_CHECK(esp_netif_init());
	ESP_ERROR_CHECK(esp_event_loop_create_default());
	esp_netif_create_default_wifi_sta();

	ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT,
				ESP_EVENT_ANY_ID,
				&event_handler,
				NULL, &instance_any_id));
	wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	ESP_ERROR_CHECK(esp_wifi_init(&cfg));
	ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
	ESP_ERROR_CHECK(esp_wifi_start());
}

void app_main()
{
	ESP_ERROR_CHECK(nvs_flash_init());
	initialise_wifi();
	xTaskCreate(&sniffer_task, "sniffer_task", 2048, NULL, 10, NULL);
}
```

zztiger123, 2021-01-29:
The corresponding status code of RX_FCS_ERR is 0X41. You can try to retrieve and mark the wrong FCS frame at the application layer, such as printing FCS ERR in the sample code :
`
static void wifi_sniffer_cb(void *recv_buf, wifi_promiscuous_pkt_type_t type)
{
    sniffer_packet_info_t packet_info;
    wifi_promiscuous_pkt_t *sniffer = (wifi_promiscuous_pkt_t *)recv_buf;
    /* prepare packet_info */
    packet_info.seconds = sniffer->rx_ctrl.timestamp / 1000000U;
    packet_info.microseconds = sniffer->rx_ctrl.timestamp % 1000000U;
    packet_info.length = sniffer->rx_ctrl.sig_len;

 

    **if (sniffer->rx_ctrl.rx_state == 0x41)
        printf("FCS err \n");**

 

    /* For now, the sniffer only dumps the length of the MISC type frame */
 
`

doragasu, 2021-01-29:
Hello @zztiger123, thanks a lot for help.

I have made a small modification to the program to take the `rx_state` into account as you suggested. Now the program only logs the contents of frames with `rx_state == 0x41`, and prints a log message otherwise:

```c
	const uint32_t state = rx->rx_ctrl.rx_state;

	if (state != 0x41) {
		ESP_LOGI(TAG, "Discarding frame with rx_state = 0x%x", state);
		return;
	}
```

Using this program, log only shows the "Discarding frame..." message:
```
[...]
I (370031) sniffer: Discarding frame with rx_state = 0x0
I (370051) sniffer: Discarding frame with rx_state = 0x0
I (370071) sniffer: Discarding frame with rx_state = 0x0
I (370471) sniffer: Discarding frame with rx_state = 0x0
I (370741) sniffer: Discarding frame with rx_state = 0x0
I (370831) sniffer: Discarding frame with rx_state = 0x0
I (370921) sniffer: Discarding frame with rx_state = 0x0
I (370961) sniffer: Discarding frame with rx_state = 0x0
I (371061) sniffer: Discarding frame with rx_state = 0x0
I (371071) sniffer: Discarding frame with rx_state = 0x0
I (371141) sniffer: Discarding frame with rx_state = 0x0
I (371461) sniffer: Discarding frame with rx_state = 0x0
I (371621) sniffer: Discarding frame with rx_state = 0x0
I (371651) sniffer: Discarding frame with rx_state = 0x0
I (371701) sniffer: Discarding frame with rx_state = 0x0
I (371711) sniffer: Discarding frame with rx_state = 0x0
I (371821) sniffer: Discarding frame with rx_state = 0x0
I (372141) sniffer: Discarding frame with rx_state = 0x0
I (372451) sniffer: Discarding frame with rx_state = 0x0
I (372451) sniffer: Discarding frame with rx_state = 0x0
I (372621) sniffer: Discarding frame with rx_state = 0x0
I (372641) sniffer: Discarding frame with rx_state = 0x0
[...]
```

We never get frames with wrong FCS logged, even though they are shown in the computer running Wireshark and capturing frames at the same time.

For reference, the modified program we are using now is:

```C
#include <string.h>
#include "esp_wifi.h"
#include "esp_log.h"
#include "esp_event.h"
#include "nvs_flash.h"

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"

#define TAG "sniffer"

#define SNIFFER_DATA_LEN 256

static EventGroupHandle_t wifi_event_group;

static const int START_BIT = BIT0;

static char printbuf[100];

static void sniffer_cb(void* buf, wifi_promiscuous_pkt_type_t type)
{
	wifi_promiscuous_pkt_t* rx = (wifi_promiscuous_pkt_t*)buf;
	int len = rx->rx_ctrl.sig_len;
	const uint32_t state = rx->rx_ctrl.rx_state;
	int i;

	if (state != 0x41) {
		ESP_LOGI(TAG, "Discarding frame with rx_state = 0x%x", state);
		return;
	}

	switch (type) {
		case WIFI_PKT_DATA:
			ESP_LOGI(TAG, "Rx data pkt len:%d", len);
			break;

		default:
			ESP_LOGI(TAG, "Not parsing type %d", type);
			return;
	}

	ESP_LOGI(TAG, "Data:");

	len = len > SNIFFER_DATA_LEN ? SNIFFER_DATA_LEN : len;

	for (i = 0; i < len; i++) {
		sprintf(printbuf + (i % 16) * 3, "%02x ", rx->payload[i]);

		if ((i + 1) % 16 == 0) {
			ESP_LOGI(TAG, "  - %s", printbuf);
		}
	}

	if ((i % 16) != 0) {
		printbuf[((i) % 16) * 3 - 1] = 0;
		ESP_LOGI(TAG, "  - %s", printbuf);
	}
}

static void sniffer_task(void* pvParameters)
{
	wifi_promiscuous_filter_t sniffer_filter = {0};

	sniffer_filter.filter_mask |= WIFI_PROMIS_FILTER_MASK_DATA;
	sniffer_filter.filter_mask |= WIFI_PROMIS_FILTER_MASK_FCSFAIL;

	xEventGroupWaitBits(wifi_event_group, START_BIT,
			false, true, portMAX_DELAY);
	// Use channel 7
	ESP_ERROR_CHECK(esp_wifi_set_channel(7, 0));
	ESP_ERROR_CHECK(esp_wifi_set_promiscuous_rx_cb(sniffer_cb));
	ESP_ERROR_CHECK(esp_wifi_set_promiscuous_filter(&sniffer_filter));
	ESP_ERROR_CHECK(esp_wifi_set_promiscuous(true));
	vTaskDelete(NULL);
}

static void event_handler(void* arg, esp_event_base_t event_base,
		int32_t event_id, void* event_data)
{
	switch (event_id) {
		case WIFI_EVENT_STA_START:
			xEventGroupSetBits(wifi_event_group, START_BIT);
			break;

		default:
			break;
	}
}

static void initialise_wifi(void)
{
	esp_event_handler_instance_t instance_any_id;

	wifi_event_group = xEventGroupCreate();
	ESP_ERROR_CHECK(esp_netif_init());
	ESP_ERROR_CHECK(esp_event_loop_create_default());
	esp_netif_create_default_wifi_sta();

	ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT,
				ESP_EVENT_ANY_ID,
				&event_handler,
				NULL, &instance_any_id));
	wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	ESP_ERROR_CHECK(esp_wifi_init(&cfg));
	ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
	ESP_ERROR_CHECK(esp_wifi_start());
}

void app_main()
{
	ESP_ERROR_CHECK(nvs_flash_init());
	initialise_wifi();
	xTaskCreate(&sniffer_task, "sniffer_task", 2048, NULL, 10, NULL);
}
```

So we are still not able to capture frames with failed FCS check.

zztiger123, 2021-01-29:
Hi doragasu,
In your case, the captured packet results only contain FCS error packets (but if you configure to obtain the wrong FCS packet, all FCS error packets will be considered by the ESP32 hardware as normal packets). At the same time, since the packets captured by ESP32 have no option to show whether the FCS is correct, you cannot see the FCS information of the packets captured by ESP32 on wireshark.

doragasu, 2021-02-01:
Hi @zztiger123,

So if I understand correctly, enabling `WIFI_PROMIS_FILTER_MASK_FCSFAIL`, should cause that __ONLY__ the frames with wrong FCS are uploaded to the wifi promiscuous callback, with `rx_state` set to 0.

But that is not what I am seeing. If I remove the `rx_state` filter (and use the code I put in the first message), I can clearly see that the wifi driver is calling the function for all frames with valid FCS, and is not showing the frames with wrong FCS.

In fact I can see no difference at all in the results when changing the `WIFI_PROMIS_FILTER_MASK_FCSFAIL` flag.

This is the setup I am using for the test:

[image]

When the `Tx PC` sends frames with wrong FCS on channel 7, I can see them on the `Rx PC`, but these frames are not captured by the `Rx ESP32` using the program I wrote in the first post. The rx promiscuous callback is not invoked for these frames, it is only called for frames with correct FCS. So either I am not setting it up correctly, or this filter is not working.

Do you have any code example and/or procedure allowing to capture and print packets with failed FCS?

zztiger123, 2021-02-02:
Hi @doragasu,
When WIFI_PROMIS_FILTER_MASK_FCSFAIL is enabled, esp32 will capture all packets (including fcs error packets). 

You can add a variable to the rx promiscuous callback to count the number. The initial value of count is set to 0. Every time a packet is received, count +1. And if (state == 0x41), print the value of count, so that if the number in the packet capture result corresponds to the printed count value, then this packet is the packet with the fcs error (for example, if count is 24 printed in log, Then the package number 24 in the ESP32 packets result is the fcs wrong packet), In this way, you can compare whether the fcs error packet of wireshark and esp32 are the same

doragasu, 2021-02-02:
Hi @zztiger123,

In all the packets captured by ESP32, I have never seen a single case with `rx_state == 0x41`. Driver is not  uploading packets with failed FCS. In addition to the `rx_state` I am also monitoring the packet payload and I can confirm no packet with failed FCS is uploaded.

zztiger123, 2021-02-02:
Hi @doragasu,
Let me summarize the main points:

1. When ESP32 captures a packet, the FCS check digit at the end of the packet will be filtered by hardware.
2. Based on the first point, when analyzing the packet captured by ESP32 using packet capture software (such as wireshark), it is impossible to see whether it is an FCS wrong packet
3. According to the code you provided in the issue(with enabling WIFI_PROMIS_FILTER_MASK_FCSFAIL), 
`
if (state != 0x41) {
    ESP_LOGI(TAG, "Discarding frame with rx_state = 0x%x", state);
    return;
}
`
The packages captured by the above code are all FCS wrong packages. For packages that are not FCS errors, the program will trigger return immediately, so it will not be saved

doragasu, 2021-02-02:
Hi @zztiger123 

I think there is a misunderstanding here:

1. I am __NOT__ opening the capture from ESP32 with Wireshark. Wireshark is used to capture packets from the air with a PC wireless dongle, just double checking.
2. What my example code does, is logging (using LOGI) the packets to console, I am __NOT__ saving the packets to an SD for further analysis.
3. I have removed the code discarding frames with `state != 0x41`, so basically I am using the code in the first post as is.
4. In my tests I am generating packets with a specific payload and invalid FCS. None of them are captured by the ESP32.

To simplify, __let's forget about Wireshark completely, and consider the following setup__:

1. From a PC, I transmit packets with a specific payload and an invalid FCS checksum on channel 7.
2. I am trying to capture on channel 7 these packets with invalid FCS on an ESP32, and log them (using LOGI). For this I am using the example program in the first post of this issue.
3. In the logged data, I cannot see any of these packets I send with the PC.

zztiger123, 2021-02-02:
Hi @doragasu,
"I am trying to capture on channel 7 these packets with invalid FCS on an ESP32, and log them (using LOGI)"
Just enabling `WIFI_PROMIS_FILTER_MASK_FCSFAIL`, and add some code to print the packet info. As shown below :
```c
static void wifi_sniffer_cb(void *recv_buf, wifi_promiscuous_pkt_type_t type)
{
    sniffer_packet_info_t packet_info;
    wifi_promiscuous_pkt_t *sniffer = (wifi_promiscuous_pkt_t *)recv_buf;
    /* prepare packet_info */
    packet_info.seconds = sniffer->rx_ctrl.timestamp / 1000000U;
    packet_info.microseconds = sniffer->rx_ctrl.timestamp % 1000000U;
    packet_info.length = sniffer->rx_ctrl.sig_len;
    /* For now, the sniffer only dumps the length of the MISC type frame */
    
    if (sniffer->rx_ctrl.rx_state == 0x41) {
        printf("FCS err packet payload = \n");

        for(int i = 0; i < packet_info.length; i++) {
            if(i % 16  == 0)
                printf("\n");
            printf("0x%02x ", sniffer->payload[i]);
        }
        printf("\n");
    }
```
And the result log is shown below :
```shell
FCS err packet payload = 

0x88 0x41 0x2c 0x00 0x70 0x3a 0x73 0x0c 0x6c 0xfb 0x40 0xf5 0x20 0x47 0x0f 0x5c 
0xf8 0x98 0xef 0xc7 0x3e 0xd1 0xd0 0x63 0x00 0x00 0x75 0x36 0x00 0x20 0x02 0x00 
0x00 0x00 0x71 0x02 0xc7 0xbe 0x81 0xdb 0xf8 0x6f 0x74 0x59 0x36 0x55 0x7b 0x39 
0xf5 0x22 0x81 0x69 0x59 0x52 0x8b 0x4a 0x67 0x62 0x81 0x22 0x45 0x8f 0x6e 0xaa 
0x40 0x4c 0x7d 0xdb 0x93 0x0f 0x13 0x3a 0xde 0x72 0xc3 0xca 0xb2 0x39 0x4b 0x00 
0xa6 0x43 0x84 
FCS err packet payload = 

0x88 0x42 0x2c 0x00 0x40 0xf5 0x20 0x47 0x0f 0x5c 0x70 0x3a 0x73 0x0c 0x6c 0xfb 
0xf8 0x98 0xef 0xc7 0x3e 0xd1 0x70 0x4a 0x00 0x00 0xad 0x2d 0x84 0x19 0x51 0xd9 
0x68 0xfd 0x7f 0x43 0x80 0xd8 0xf8 0x2d 0x92 0xcb 0xa4 0xae 0xb3 0x1f 0xae 0x31 
0xca 0xbf 0x30 0x32 0x8a 0x64 0xbd 0x50 0xb4 0x8c 0xb4 0xb9 0x86 0x55 0x0f 0x34 
0xaa 0xb6 0x4d 0x43 0xfe 0x4d 0xd4 0x6b 0x04 0x62 0xc5 0x68 0x58 0x72 0x4e 0xb2 
0xe9 0x2a 0xe3 0x28 0x5d 0x99 0xdb 0xa7 0x07 0xf0 0xb5 0x71 0x5f 0x45 0x36 0xbe 
0x0d 0x31 0x3d 0x0a 
```

The packet information in ESP32 log is similar to the packet information in wireshark, but the packet captured by ESP32 lacks the FCS field, as follows :

[image]
The left side of the above figure is the packet captured by the packet capture card, which has the FCS field. On the right side is the packet captured by ESP32, there is no FCS field.

doragasu, 2021-02-03:
Hi @zztiger123

Now that it is clear in your example that you are receiving the 0x41 state, I started doing all kind of tests and __got it to work__ when I tried enabling every single bit in the filter mask:

```c
	wifi_promiscuous_filter_t sniffer_filter = {0};

	sniffer_filter.filter_mask = WIFI_PROMIS_FILTER_MASK_ALL;

	ESP_ERROR_CHECK(esp_wifi_set_promiscuous_rx_cb(sniffer_cb));
	ESP_ERROR_CHECK(esp_wifi_set_promiscuous_filter(&sniffer_filter));
```

I was only enabling FCSFAIL and DATA flags and that did not work. As I did not want to enable capturing everything, I have been investigating which flags are the minimum required to make it work, and I have found that it works when I enable `WIFI_PROMIS_FILTER_MASK_MISC | WIFI_PROMIS_FILTER_MASK_DATA_MPDU | WIFI_PROMIS_FILTER_MASK_FCSFAIL`. If I remove the MISC or the DATA_MPDU flags, I get no FCS failed packets.

Thanks for support, now it is working as I expected, but maybe the flags needed should be written in the documentation to avoid confusion.

closed 2021-02-03 by doragasu

espxiehang, 2021-02-03:
Hi @doragasu ,Yes,  WIFI_PROMIS_FILTER_MASK_FCSFAIL is just a switch. When it is turned on, it will let the hardware receive the FCS fail package.