espressif/esp-idf #6474: System crashes when connecting to TLS-enabled httpd right after successful (universal) provisioning

dbahrdt, 2021-01-28:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32D
- IDF version: v4.2-238-g8cd16b60f
- Build System:        idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Arch Linux
- Using an IDE?: No
- Power Supply:    USB

## Problem Description
Using the httpd-component with WolfSSL results in a crash if one connects to the server after a successful provisioning of the device.
After a reboot initiated by an external reset one can contact the https-server without crashing the system.
I have also tested upstream WolfSSL (with modifications to use it with esp-tls) to debug this.
I tried versions 4.{3,4,5,6} and all exhibit the same problem.
The crash does not happen if I build WolfSSL with `NO_WOLFSSL_ESP32WROOM32_CRYPT_RSA_PRI`.

### Expected Behavior
System should not crash.

### Actual Behavior
System crashes.

### Steps to reproduce
You need an ESP32-WROOM-32D potentially with a MFI-Chip.
Use WolfSSL as TLS-Library, provision the device with universal provisioning, start a https-server after a successful provisioning and connect to it using curl/openssl.

### Code to reproduce this issue
See https://github.com/dbahrdt/esp-wolfssl-provisioning-crash.
Simply edit `tools/debuggin/cfg` to your needs and rename it to `.cfg`.
The script `energize` will:
1. Create temporary factory_nvs data
2. Temporary CA+server-certificate+sign it
3. Build the project
4. Flash the project and the factory_nvs data
5. Start the app using idf.py monitor
6. Provision the device using universal provisioning (connection to the wifi network "mwe-wolfssl-crash" is not done automatically)

After a successful provisioning you may use curl to connect to the device:
```bash
curl --cacert components/web/certs/ca.pem https://mwe.lan:4443/test
```

## Debug Logs
Crash log: https://gist.github.com/dbahrdt/706a8e6fff4118e6b728f28001979a9f
```
I (80587) esp_https_server: performing session handshake
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x401aafcf  PS      : 0x00060c30  A0      : 0x80178726  A1      : 0x3ffd57c0  
0x401aafcf: fp_lshd at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/tfm.c:3214 (discriminator 2)

A2      : 0x3ffdbdd8  A3      : 0xffffffbf  A4      : 0x3ffdbdd8  A5      : 0x00000040  
A6      : 0x00000001  A7      : 0x00000022  A8      : 0x9cb0cdd0  A9      : 0x172cc3ff  
A10     : 0x172cc3fe  A11     : 0x00000087  A12     : 0x58326e78  A13     : 0x00000001  
A14     : 0x3ffdb6d4  A15     : 0x00000022  SAR     : 0x00000001  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x9cb0cdd8  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace:0x401aafcc:0x3ffd57c0 0x40178723:0x3ffd57e0 0x4017898a:0x3ffd5830 0x40178c63:0x3ffd5850 0x40178eda:0x3ffd5870 0x40179349:0x3ffd58c0 0x4017936a:0x3ffd58e0 0x4016bf9e:0x3ffd5900 0x4016c2ad:0x3ffd5940 0x4016c6de:0x3ffd5980 0x4016c7d5:0x3ffd59d0 0x4016c7fa:0x3ffd5a10 0x4017b7ed:0x3ffd5a40 0x40182567:0x3ffd5a80 0x4015906c:0x3ffd5b00 0x4011fe3d:0x3ffd5b20 0x4011f6f9:0x3ffd5b40 0x400dea04:0x3ffd5b60 0x401213a6:0x3ffd5b80 0x4011ffaa:0x3ffd5bb0 0x4012014b:0x3ffd5c00 0x40120187:0x3ffd5c40 0x40088c15:0x3ffd5c60
0x401aafcc: fp_lshd at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/tfm.c:3214 (discriminator 2)

0x40178723: fp_div at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/tfm.c:642

0x4017898a: fp_mod at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/tfm.c:834

0x40178c63: fp_mulmod at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/tfm.c:1173

0x40178eda: _fp_exptmod at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/tfm.c:1526

0x40179349: fp_exptmod at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/tfm.c:2267

0x4017936a: mp_exptmod at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/tfm.c:3531

0x4016bf9e: wc_RsaFunctionSync at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/rsa.c:2148 (discriminator 1)

0x4016c2ad: wc_RsaFunction at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/rsa.c:2656

0x4016c6de: RsaPublicEncryptEx at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/rsa.c:2776

0x4016c7d5: wc_RsaPSS_Sign_ex at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/rsa.c:3501

0x4016c7fa: wc_RsaPSS_Sign at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/rsa.c:3477

0x4017b7ed: RsaSign at /builds/esp-components/esp-wolfssl/wolfssl/src/internal.c:3640

0x40182567: SendServerKeyExchange at /builds/esp-components/esp-wolfssl/wolfssl/src/internal.c:24196

0x4015906c: wolfSSL_accept at /builds/esp-components/esp-wolfssl/wolfssl/src/ssl.c:11975

0x4011fe3d: esp_wolfssl_server_session_create at ${IDF_PATH}/components/esp-tls/esp_tls_wolfssl.c:444

0x4011f6f9: esp_tls_server_session_create at ${IDF_PATH}/components/esp-tls/esp_tls.c:451

0x400dea04: httpd_ssl_open at ${IDF_PATH}/components/esp_https_server/src/https_server.c:104 (discriminator 13)

0x401213a6: httpd_sess_new at ${IDF_PATH}/components/esp_http_server/src/httpd_sess.c:79

0x4011ffaa: httpd_accept_conn at ${IDF_PATH}/components/esp_http_server/src/httpd_main.c:65 (discriminator 15)

0x4012014b: httpd_server at ${IDF_PATH}/components/esp_http_server/src/httpd_main.c:213 (discriminator 15)

0x40120187: httpd_thread at ${IDF_PATH}/components/esp_http_server/src/httpd_main.c:229 (discriminator 15)

0x40088c15: vPortTaskWrapper at ${IDF_PATH}/components/freertos/xtensa/port.c:143

```

AdityaHPatwardhan, 2021-03-15:
Hi @dbahrdt sort for the delay, Thank you for the issue and your analysis.
Unfortunately, I do not have access to the `MFI` folder which is requires to reproduce this (https://github.com/dbahrdt/esp-wolfssl-provisioning-crash), Can you please provide something with which I can reproduce this issue without the `MFI` folder.

dbahrdt, 2021-03-17:
Hi @AdityaHPatwardhan,
I've update the the mwe to support https://github.com/espressif/esp-homekit-sdk as well. You may have to either delete your `sdkconfig` file or disable MFI support in `menuconfig`.
Unfortunately the crash is slightly different now. Instead of crashing the respective task simply hangs and triggers the watchdog.

```bash
I (264747) MWE_WOLFSSL_CRASH: MWE still running
E (264867) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (264867) task_wdt:  - IDLE1 (CPU 1)
E (264867) task_wdt: Tasks currently running:
E (264867) task_wdt: CPU 0: IDLE0
E (264867) task_wdt: CPU 1: httpd
E (264867) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x400D49D4:0x3FFB09A0 0x4008291D:0x3FFB09C0 0x401A653F:0x3FFBC890 0x400D4CD2:0x3FFBC8B0 0x40088265:0x3FFBC8D0 0x400878D9:0x3FFBC8F0 
0x400d49d4: task_wdt_isr at esp-idf/components/esp_common/src/task_wdt.c:187

0x4008291d: _xt_lowint1 at esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105

0x401a653f: esp_pm_impl_waiti at esp-idf/components/esp32/pm_esp32.c:484

0x400d4cd2: esp_vApplicationIdleHook at esp-idf/components/esp_common/src/freertos_hooks.c:63

0x40088265: prvIdleTask at esp-idf/components/freertos/tasks.c:3386 (discriminator 1)

0x400878d9: vPortTaskWrapper at esp-idf/components/freertos/xtensa/port.c:143

E (264867) task_wdt: Print CPU 1 backtrace

Backtrace:0x40081B9F:0x3FFB11E0 0x4008291D:0x3FFB1200 0x40175E41:0x3FFD6040 0x40176086:0x3FFD6090 0x4017635F:0x3FFD60B0 0x401765D6:0x3FFD60D0 0x40176A45:0x3FFD6120 0x40176A66:0x3FFD6140 0x4016969A:0x3FFD6160 0x401699A9:0x3FFD61A0 0x40169DDA:0x3FFD61E0 0x40169ED1:0x3FFD6230 0x40169EF6:0x3FFD6270 0x40178EE9:0x3FFD62A0 0x4017FC63:0x3FFD62E0 0x40156768:0x3FFD6360 0x4011D515:0x3FFD6380 0x4011CDD9:0x3FFD63A0 0x400DEB30:0x3FFD63C0 0x4011EA6A:0x3FFD63E0 0x4011D67F:0x3FFD6410 0x4011D80B:0x3FFD6460 0x4011D847:0x3FFD64A0 0x400878D9:0x3FFD64C0 
0x40081b9f: esp_crosscore_isr at esp-idf/components/esp32/crosscore_int.c:80

0x4008291d: _xt_lowint1 at esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105

0x40175e41: fp_div at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/tfm.c:644

0x40176086: fp_mod at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/tfm.c:834

0x4017635f: fp_mulmod at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/tfm.c:1173

0x401765d6: _fp_exptmod at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/tfm.c:1526

0x40176a45: fp_exptmod at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/tfm.c:2267

0x40176a66: mp_exptmod at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/tfm.c:3531

0x4016969a: wc_RsaFunctionSync at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/rsa.c:2148 (discriminator 1)

0x401699a9: wc_RsaFunction at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/rsa.c:2656

0x40169dda: RsaPublicEncryptEx at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/rsa.c:2776

0x40169ed1: wc_RsaPSS_Sign_ex at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/rsa.c:3501

0x40169ef6: wc_RsaPSS_Sign at /builds/esp-components/esp-wolfssl/wolfssl/wolfcrypt/src/rsa.c:3477

0x40178ee9: RsaSign at /builds/esp-components/esp-wolfssl/wolfssl/src/internal.c:3640

0x4017fc63: SendServerKeyExchange at /builds/esp-components/esp-wolfssl/wolfssl/src/internal.c:24196

0x40156768: wolfSSL_accept at /builds/esp-components/esp-wolfssl/wolfssl/src/ssl.c:11975

0x4011d515: esp_wolfssl_server_session_create at esp-idf/components/esp-tls/esp_tls_wolfssl.c:444

0x4011cdd9: esp_tls_server_session_create at esp-idf/components/esp-tls/esp_tls.c:485

0x400deb30: httpd_ssl_open at esp-idf/components/esp_https_server/src/https_server.c:104 (discriminator 13)

0x4011ea6a: httpd_sess_new at esp-idf/components/esp_http_server/src/httpd_sess.c:79

0x4011d67f: httpd_accept_conn at esp-idf/components/esp_http_server/src/httpd_main.c:65 (discriminator 15)

0x4011d80b: httpd_server at esp-idf/components/esp_http_server/src/httpd_main.c:211 (discriminator 15)

0x4011d847: httpd_thread at esp-idf/components/esp_http_server/src/httpd_main.c:227 (discriminator 15)

0x400878d9: vPortTaskWrapper at esp-idf/components/freertos/xtensa/port.c:143
```

AdityaHPatwardhan, 2021-03-18:
@dbahrdt Thanks, I will try to find out the cause and hopefully a solution and let you know.

AdityaHPatwardhan, 2021-03-25:
Hi @dbahrdt
In above mentioned case as the watchdog is getting triggered which can be avoided by increasing the watchdog timeout.
When I try the example that you have given, there is no crash
I tried to reproduce the issue by softAP provisioning + https_server on release/v4.2 . It is working fine and not able to reproduce that crash that you have mentioned.

dbahrdt, 2021-03-29:
That is strange. I increased all watchdog timers and the watchdog is still getting triggered.
I'm using commit `c9cf7bcb0ef29c8f10df0906bf1fbac751fc1299`.
I tested two different chip revisions (0 and 3). The revision 0 chip is part of our product. The revision 3 chip ist used with a programming socket (like this one https://www.amazon.de/-/en/VAILANG-Programming-Socket-Adaptor-ESP-WROOM-32/dp/B07ZCYWX4D).

sdkconfig: https://gist.github.com/dbahrdt/802c9f25d5a715aede91a2f1e0a10a0e
crash.log: https://gist.github.com/dbahrdt/b5d32dec93663d5bf851efdacf18cee0

Interestingly I also got the crash that I mentioned before with the chip with revision 3 iff the minimum supported revision is set to 0:

sdkconfig: https://gist.github.com/dbahrdt/a427ade046aa610daa8b89f739d2ffb9
crash.log: https://gist.github.com/dbahrdt/ea2cdb6501ae7819f4485af04d28180c

I will do some more tests with the latest master branch and the release 4.3 branch. I hope one of these is working so I can pin down the commit that fixed the issue.

AdityaHPatwardhan, 2021-03-30:
Hi @dbahrdt, thanks for the update.
I was able to reproduce the crash in idf-master as you have mentioned in https://gist.github.com/dbahrdt/1f3e9890299f76fe0843a8813140d061 ( the crash log contains this line- `assertion "periph < PERIPH_MODULE_MAX" failed: file "IDF/components/driver/periph_ctrl.c", line 24, function: periph_module_enable ` .  When I use the wolfSSL source to build the example then the crash is fixed. So I think updating the binaries should fix the issue.

However, I was able to reproduce this only on the master and release/v4.3 branch. the original issue on release/v4.2 is a bit different and the crash log is also different. I was not able to reproduce the first crash ( without the line mentioned earlier in my message) you have reproduced in release/v4.2.

dbahrdt, 2021-03-31:
Hi @AdityaHPatwardhan,
In order to debug this I have also built WolfSSL from source. Did you make any specific changes?
You can find my variant at https://github.com/dbahrdt/esp-wolfssl.

Using this repo instead of the official one I can still trigger the watchdog timer.
However I'm not experiencing a crash anymore (latest master branch of esp-idf)
```
I (29970) WEBSERVER: Delaying restart by 5000 ms
I (33500) MWE_WOLFSSL_CRASH: MWE still running
I (35000) WEBSERVER: Starting web server
D (35060) httpd: httpd_thread: web server started
D (35060) httpd: httpd_server: doing select maxfd+1 = 53
D (35070) httpd_uri: httpd_register_uri_handler: [0] installed /test
I (43500) MWE_WOLFSSL_CRASH: MWE still running
I (53500) MWE_WOLFSSL_CRASH: MWE still running
I (63500) MWE_WOLFSSL_CRASH: MWE still running
I (73500) MWE_WOLFSSL_CRASH: MWE still running
I (83500) MWE_WOLFSSL_CRASH: MWE still running
I (93500) MWE_WOLFSSL_CRASH: MWE still running
I (103500) MWE_WOLFSSL_CRASH: MWE still running
I (113500) MWE_WOLFSSL_CRASH: MWE still running
I (123500) MWE_WOLFSSL_CRASH: MWE still running
I (133500) MWE_WOLFSSL_CRASH: MWE still running
I (143500) MWE_WOLFSSL_CRASH: MWE still running
I (153500) MWE_WOLFSSL_CRASH: MWE still running
I (163500) MWE_WOLFSSL_CRASH: MWE still running
I (173500) MWE_WOLFSSL_CRASH: MWE still running
I (183500) MWE_WOLFSSL_CRASH: MWE still running
I (193500) MWE_WOLFSSL_CRASH: MWE still running
I (203500) MWE_WOLFSSL_CRASH: MWE still running
I (213500) MWE_WOLFSSL_CRASH: MWE still running
W (221500) wifi:<ba-add>idx:1 (ifx:0, 3c:a6:2f:dd:d8:dc), tid:0, ssn:1, winSize:64
D (221500) httpd: httpd_server: processing listen socket 51
D (221500) httpd: httpd_accept_conn: newfd = 54
D (221500) httpd_sess: httpd_sess_new: fd = 54
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6704
load:0x40078000,len:14256
load:0x40080400,len:3672
0x40080400: _init at ??:?

```

dbahrdt, 2021-03-31:
Hi @AdityaHPatwardhan,

unfortunately I have to revise my former statement. I'm still getting the crash same crash.
I'm using the docker image for testing. To ease debugging one can now select the idf version to run (each version is a service):
`docker-compose run --rm mwe-wolfssl-latest ./energize`

To get a specific version (4.2 or 4.3):
`docker-compose run --rm mwe-wolfssl-4.2 ./energize`.
The branch https://github.com/dbahrdt/esp-wolfssl-provisioning-crash/tree/wolfssl-src uses the source based WolfSSL.

The crash.logs look like the ones I have already posted. If you want I can also post these.

Could this be related to my network setup?
My setup is as follows:

- Router: 192.168.1.0/24
- Laptop is connected using an ethernet cable
- Connection to ESP32 is done using the wifi card of my laptop
- I have selected the ESP32's network as sole option for wpa_supplicant (set_network command)

dbahrdt, 2021-03-31:
Hi @AdityaHPatwardhan,

I've setup a completely isolated network using an old FritzBox 7390.
I used a fresh clone of the repository, edited the `tools/debugging/cfg` file accordingly and executed the `energize` script using the docker image.

I'm getting the crash using idf-master and wolfssl from source. I'm also getting the crash using the original esp-wolfssl repository.
Log from docker-compose until reset: https://gist.github.com/dbahrdt/f831d7e859b3f268e1c6fa9b64a537bb

Enabling `NO_WOLFSSL_ESP32WROOM32_CRYPT_RSA_PRI` still helps and everything is working fine.
This disables the usage of the following functions:
- `fp_count_bits`
- `esp_mp_mulmod`
- `esp_mp_mul`
- `esp_mp_exptmod`

Do you have any idea why I'm getting a crash and you are not? Shoul I report this at the wolfssl repository?

AdityaHPatwardhan, 2021-04-01:
Hi @dbahrdt , I was able to reproduce the issue at me end. I was doing it wrong at first.
I think the crash is due to `DPORT_RSA_PD` bit in the `DPORT_RSA_PD_CTRL_REG` was not getting cleared.
Please check the patch I have provided. 
[crash_issue_fix.patch.zip](https://github.com/espressif/esp-idf/files/6242399/crash_issue_fix.patch.zip)

dbahrdt, 2021-04-01:
Hi @AdityaHPatwardhan ,

a quick test resulted in a hang in a different function. I'll try some more tests.

As I understand it your patch waits for the DPORT_RSA_PD_CTRL_REG to be cleared as stated in the [esp32 technical reference, section 24.3.1](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf).
However this is also done in [`esp_mp_hw_wait_clean`](https://github.com/dbahrdt/esp-wolfssl/blob/32da22f18b568155213be3488f1a281ec91906ed/wolfcrypt/src/port/Espressif/esp32_mp.c#L60)
All usages of the function [`esp_mp_hw_lock`](https://github.com/dbahrdt/esp-wolfssl/blob/32da22f18b568155213be3488f1a281ec91906ed/wolfcrypt/src/port/Espressif/esp32_mp.c#L75) seem to be accompanied by `esp_mp_hw_wait_clean`.
Sometimes there's another function used in-between - [`esp_calc_Mdash`](https://github.com/dbahrdt/esp-wolfssl/blob/32da22f18b568155213be3488f1a281ec91906ed/wolfcrypt/src/port/Espressif/esp32_mp.c#L112) - however that one does not look like it's using the esp rsa accelerator?

dbahrdt, 2021-04-01:
Hi @AdityaHPatwardhan,

I can confirm that I'm still getting a hang iff I set the minimum chip revision to 3 instead of 0. This also happens on release 4.2.

The following data is based on a docker container run using the `latest` tag:
sdkconfig: https://gist.github.com/dbahrdt/ce38b31dbdce4acbf0f4f5c2315349d0
sdkconfig chip rev3: https://gist.github.com/dbahrdt/1f8456de6114be9579ce1a2e2b58ed7a
crash.log: https://gist.github.com/dbahrdt/9f9a27467c993ab2bc6ef5b1e9111613

```
E (92636) task_wdt: Print CPU 1 backtrace

Backtrace:0x400821A5:0x3FFB1680 0x40082B9D:0x3FFB16A0 0x4014F68E:0x3FFD7500 0x401502B0:0x3FFD7520 0x4014E9F6:0x3FFD7780 0x4014EABE:0x3FFD77A0 0x401477A5:0x3FFD77C0 0x40147A2D:0x3FFD7800 0x40147E6A:0x3FFD7840 0x40147F65:0x3FFD7890 0x40147F8A:0x3FFD78D0 0x40151DC9:0x3FFD7900 0x40159B4A:0x3FFD7940 0x4012455F:0x3FFD79C0 0x401200C3:0x3FFD79E0 0x4011F8F1:0x3FFD7A00 0x400FD74A:0x3FFD7A20 0x40121862:0x3FFD7A40 0x40120273:0x3FFD7A70 0x401203CF:0x3FFD7AC0 0x40120413:0x3FFD7B10 0x4008BE59:0x3FFD7B30 
0x400821a5: esp_crosscore_isr at /opt/esp/idf/components/esp32/crosscore_int.c:77

0x40082b9d: _xt_lowint1 at /opt/esp/idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x4014f68e: wait_uitil_done at /project/build/../components/esp-wolfssl/wolfcrypt/src/port/Espressif/esp32_mp.c:153

0x401502b0: esp_mp_exptmod at /project/build/../components/esp-wolfssl/wolfcrypt/src/port/Espressif/esp32_mp.c:509
```

My understanding is that `wait_uitil_done` waits for `RSA_INTERRUPT_REG` as specified in the TRM in section 24.3.2, step 5:

> 5. Wait for the operation to be completed. Poll RSA_INTERRUPT_REG until it reads 1, or until the RSA_INTR
interrupt is generated.

AdityaHPatwardhan, 2021-04-05:
Hi @dbahrdt, you mean the example works correctly when minimum revision is `REV0` ? and fails only when you set minimum revision to `REV3` ? ( that's weird) 
Unfortunately, I do not currently have a `REV3` chip to reproduce this at the moment. Can you try reducing the `ESP_RSA_TIMEOUT` by defining it in the `user_settings.h' ?
the `wait_uitil_done` function should fail and return in that case.

dbahrdt, 2021-04-06:
Hi @AdityaHPatwardhan,
It does sound weird. I've tested this multiple times. Maybe I just had bad luck.
I'm currently on vacation (at home, if one wants to call that a vacation).
I'm getting back to you as soon as I'm back to work (next Monday). I hope that does not bother you too much.

AdityaHPatwardhan, 2021-04-06:
Sure, No problem. Enjoy your stay at home :)

dbahrdt, 2021-04-14:
Hi @AdityaHPatwardhan,

I did some more tests and in some everything worked and sometimes it didn't. However I think I have narrowed down the problem.
The function `wait_until_done` (https://github.com/wolfSSL/wolfssl/blob/21faeff4786a2445919465431ec62fe0339ca9e9/wolfcrypt/src/port/Espressif/esp32_mp.c#L146):
```C++
static int wait_uitil_done(word32 reg)
{
    int timeout = 0;
    /* wait until done && not timeout */
    while(1) {
        // If timeout >= ESP_RSA_TIMEOUT then we never leave the loop, since ++timeout < ESP_RSA_TIMEOUT is false
        // This may happen if  DPORT_REG_READ(reg) takes too long
        if(++timeout < ESP_RSA_TIMEOUT && DPORT_REG_READ(reg) == 1){
            break;
        }
    }
...
```
The next question is, why is it, that this does not happen if we set the minimum revision to 0?
The macro `DPORT_REG_READ` expands to the following (https://github.com/espressif/esp-idf/blob/1cb31e50943bb757966ca91ed7f4852692a5b0ed/components/soc/esp32/include/soc/dport_access.h#L74):
```C++
static inline uint32_t IRAM_ATTR DPORT_REG_READ(uint32_t reg)
{
#if defined(BOOTLOADER_BUILD) || !defined(CONFIG_ESP32_DPORT_WORKAROUND) || !defined(ESP_PLATFORM)
    return _DPORT_REG_READ(reg);
#else
    return esp_dport_access_reg_read(reg);
#endif
}
```
`CONFIG_ESP32_DPORT_WORKAROUND` is defined for min rev < 3.
On min rev < 3 the function `esp_dport_access_reg_read` is used instead which does a lot more things (https://github.com/espressif/esp-idf/blob/1cb31e50943bb757966ca91ed7f4852692a5b0ed/components/esp32/dport_access.c#L246).

I guess that's the main reason that the bug in the `wait_until_done` function is not triggered if we set min rev < 3.
Increasing the timeout solved the issue in such a way, that I did not get the hang anymore.
Also fixing the loop condition solved the problem.

AdityaHPatwardhan, 2021-04-15:
Hi, @dbahrdt Thanks for the reply and I am glad that the issue is fixed for your case. I will update the libraries.
I would also like to bring your attention to the license for `esp-wolfssl`.
As the [README](https://github.com/espressif/esp-wolfssl#readme) already mentions that the license for Espressif to distribute binaries for production use-case is only until March 2021 . After that  you may contact wolfSSL for further licensing information. If you are using the source code in your project, please note that it has GPL license and cannot be used for a closed source project.

Alternatively, if you are using wolfSSL instead of mbedTLS for its lower memory consumption.
This is the list of options with which the mbedTLS memory consumption can be reduced.

* First ensure that mbedTLS asymmetric content length option is on, https://github.com/espressif/esp-idf/blob/master/components/mbedtls/Kconfig#L69
* Second with above, TLS out length can be further reduced to 2K (or even smaller per application requirement), https://github.com/espressif/esp-idf/blob/master/components/mbedtls/Kconfig#L86
* With above TLS in/out fragment takes 16K+2K = 18K memory, next it is possible to redirect these 18K allocations to IRAM (instruction memory) using https://github.com/espressif/esp-idf/blob/master/components/mbedtls/Kconfig#L36. Note that this will need single core configuration of FreeRTOS. This allows to free up DRAM (data memory) for generic application usage purpose.
* There is also one option called mbedtls dynamic buffers which helps reduce heap consumption, can be enabled in `menuconfig->Component Config->mbedTLS-> Using dynamic TX/RX buffers.

dbahrdt, 2021-04-23:
Hi @AdityaHPatwardhan,

thanks for the info. I've tested your suggestions and results are looking good.
I have also implemented support for rfc 5077 session tickets for which I'll open a PR next week.

Regarding the bug, are you going to report this at WolfSSL? Since this is entirely in upstreams code they should be the one to fix it.

AdityaHPatwardhan, 2021-04-25:
@dbahrdt Yeah, I have conveyed the issue to the wolfSSL team, they are reviewing the changes and then probably they will merge it upstream.
Thanks for finding this issue and helping to find a fix for it.

AdityaHPatwardhan, 2021-05-17:
Hi @dbahrdt the issue has been fixed in wolfSSL with this PR https://github.com/wolfSSL/wolfssl/pull/3994 , will update the upstream version when the fix is incorporated in the latest release. 
Can you close down this issue ?

dbahrdt, 2021-05-19:
Hi @AdityaHPatwardhan ,
thank you for your help. I've opened a PR (https://github.com/espressif/esp-idf/pull/7048).

closed 2021-05-19 by dbahrdt